openwrt/packages #14122: [lighttpd] -mod-wolfssl fails (likely due to missing libwolfssl functionality)

n8v8R, 2020-12-03:
lighttpd: 1.4.56-2
lighttpd-mod-wolfssl: 1.4.56-2
Maintainer: @flyn-org

libwolfssl24: 4.5.0-stable-2
Maintainer: @cotequeiroz

Description:

```
# lighttpd -tt -f /etc/lighttpd/lighttpd.conf

(../src/plugin.c.195) dlopen() failed for: /usr/lib/lighttpd/mod_wolfssl.so Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_EVP_sha256: symbol not found
(../src/server.c.1225) loading plugins finally failed
```
given the output it would seem being caused by some missing functionality from libwolfssl.

Subsequent TLS connectivity with lighttpd is not possible (least not with wolfssl backend)
```
# opkg files lighttpd-mod-wolfssl

Package lighttpd-mod-wolfssl (1.4.56-2) is installed on root and has the following files:
/etc/lighttpd/conf.d/30-wolfssl.conf
/usr/lib/lighttpd/mod_wolfssl.so
```

n8v8R, 2020-12-03:
seems not being an issue with _libwolfssl24_

```
# dltest /usr/lib/libwolfssl.so.24.2.0 wolfSSL_EVP_sha256

SUCCESS: Loaded /usr/lib/libwolfssl.so.24.2.0
SUCCESS: Found wolfSSL_EVP_sha256
```

n8v8R, 2020-12-03:
```
# ldd /usr/lib/lighttpd/mod_wolfssl.so

        ldd (0xb55a3000)
        libc.so => ldd (0xb55a3000)
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_EVP_sha256: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_eq_icase_slen: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_ERR_GET_REASON: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wc_DerToPemEx: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_get_verify_result: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_X509_get_der: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_init: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CTX_set_session_id_context: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: config_get_config_cond_info: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: li_tohex_uc: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: chunkqueue_mark_written: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: fdevent_load_file: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_string_set_length: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_append_base64_decode: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_EVP_DecryptInit_ex: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_free: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_string_prepare_copy: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_X509_STORE_new: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_shutdown: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_DH_free: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CTX_set_info_callback: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_EVP_EncryptInit_ex: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_X509_NAME_ENTRY_get_data: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSLv23_server_method: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_OBJ_obj2nid: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: array_get_element_klen: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: array_free: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: chunkqueue_get_memory: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_Init: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_RAND_status: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_HMAC_Init_ex: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: log_error: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_X509_free: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: http_header_env_set: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: config_plugin_values_init: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: chunkqueue_use_memory: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: chunkqueue_peek_data: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CTX_clear_options: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_eq_icase_ssn: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_BN_bin2bn: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CTX_use_PrivateKey_buffer: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_BN_free: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: http_header_env_get: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_Cleanup: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_EVP_aes_256_cbc: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_get_error: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: buffer_append_string: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CTX_set_tlsext_ticket_key_cb: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: safe_memclear: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_OBJ_sn2nid: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CIPHER_get_bits: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: chunkqueue_remove_finished_chunks: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_pending: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_GetVersion: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_CTX_set_tmp_dh: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_RAND_bytes: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: array_insert_value: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: li_itostrn: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: array_init: symbol not found
[... cut ...]
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_get_app_data: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: config_plugin_value_tobool: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_X509_STORE_add_cert: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_get_current_cipher: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: wolfSSL_set_fd: symbol not found
Error relocating /usr/lib/lighttpd/mod_wolfssl.so: log_epoch_secs: symbol not found

```

gstrauss, 2020-12-03:
I submitted PR #14128 to fix this issue.

gstrauss, 2020-12-04:
#14128 has been merged.  I believe that this issue is fixed and can be closed.

neheb, 2020-12-04:
Fixed.

closed 2020-12-04 by neheb