openwrt/packages #14123: mwan3 missing routing table on wan interface after restart

thomaslunze, 2020-12-03:
I am running mwan3 with 2 wan interfaces (wan and wan1). With the latest build firmware I am having a routing issue.

see my troubleshooting output here:
Software-Version
-------------------------------------------------
OpenWrt - OpenWrt SNAPSHOT r14966-7330348f2d
LuCI - git-20.311.85590-5c5b134

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1508 qdisc mq state UP group default qlen 1024
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
    inet6 fe80::1691:82ff:fec2:82ca/64 scope link 
       valid_lft forever preferred_lft forever
3: lan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 1000
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
4: lan3@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-lan state LOWERLAYERDOWN group default qlen 1000
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
5: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
6: lan1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.3.2/24 brd 192.168.3.255 scope global lan1
       valid_lft forever preferred_lft forever
    inet6 fe80::1691:82ff:fec2:82ca/64 scope link 
       valid_lft forever preferred_lft forever
7: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.2/24 brd 192.168.2.255 scope global wan
       valid_lft forever preferred_lft forever
    inet6 fe80::1691:82ff:fec2:82ca/64 scope link 
       valid_lft forever preferred_lft forever
8: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:25:9c:14:03:46 brd ff:ff:ff:ff:ff:ff
9: wlan1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 00:25:9c:14:03:45 brd ff:ff:ff:ff:ff:ff
10: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
11: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 14:91:82:c2:82:ca brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fe80::1691:82ff:fec2:82ca/64 scope link 
       valid_lft forever preferred_lft forever

Output of "ip route show"
-------------------------------------------------
default via 192.168.2.1 dev wan proto static metric 10 
default via 192.168.3.1 dev lan1 proto static metric 20 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev wan proto static scope link metric 10 
192.168.3.0/24 dev lan1 proto static scope link metric 20

Output of "ip rule show"
-------------------------------------------------
0:	from all lookup local
1001:	from all iif wan lookup 1
1002:	from all iif lan1 lookup 2
2001:	from all fwmark 0x100/0xff00 lookup 1
2002:	from all fwmark 0x200/0xff00 lookup 2
2253:	from all fwmark 0xfd00/0xff00 blackhole
2254:	from all fwmark 0xfe00/0xff00 unreachable
3001:	from all fwmark 0x100/0xff00 unreachable
3002:	from all fwmark 0x200/0xff00 unreachable
32766:	from all lookup main
32767:	from all lookup default

Output of "ip route list table 1-250"
-------------------------------------------------
Table 2: default via 192.168.3.1 dev lan1 proto static metric 20 
192.168.3.0/24 dev lan1 proto static scope link metric 20

as you can see after a restart the routing table1 is missing. after running a "hotplug ifup" on interface: wan the routing table shows up. 

Output of "ip route list table 1-250"
-------------------------------------------------
Table 1: default via 192.168.2.1 dev wan proto static metric 10 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 
192.168.2.0/24 dev wan proto static scope link metric 10
Table 2: default via 192.168.3.1 dev lan1 proto static metric 20 
192.168.3.0/24 dev lan1 proto static scope link metric 20

With an older built of openwrt (OpenWrt SNAPSHOT r14154-2d9db93397) everything is working fine after a reboot. so whats wrong with my config that it does prevent to be working as expected after a reboot of the wrt1200ac

Any help is highly appreciated

feckert, 2020-12-03:
I know this issue could you try https://github.com/openwrt/packages/pull/13889

feckert, 2020-12-03:
We need also the network and mwa3 config. And more information what your setup is.
You can also enable the debug https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/common.sh#L26
output her. Have a look at the start sequence in the log, maybe you can see something

thomaslunze, 2020-12-03:
checking the syslog I have found this for mwan:

487: Thu Dec  3 11:45:13 2020 user.notice mwan3track[2843]: Interface wan (wan) is online
488: Thu Dec  3 11:45:13 2020 user.notice mwan3track[2844]: Interface wan1 (lan1) is online
489: Thu Dec  3 11:45:13 2020 user.notice mwan3-hotplug[2604]: mwan3 hotplug on lan not called because interface disabled
527: Thu Dec  3 11:45:15 2020 user.notice mwan3-hotplug[3917]: mwan3 hotplug on loopback not called because interface disabled
564: Thu Dec  3 11:45:16 2020 user.notice mwan3-hotplug[4483]: Execute ifup event on interface wan (wan)
569: Thu Dec  3 11:45:16 2020 user.warn mwan3-hotplug[4483]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 1
574: Thu Dec  3 11:45:17 2020 user.info mwan3track[2843]: Detect ifup event on interface wan (wan)
575: Thu Dec  3 11:45:17 2020 user.info mwan3track[2843]: Check (ping: latency=999999ms loss=100%) failed for target "8.8.8.8" on interface wan (wan). Current score: 6
584: Thu Dec  3 11:45:18 2020 user.notice mwan3-hotplug[5421]: Execute ifup event on interface wan1 (lan1)
590: Thu Dec  3 11:45:18 2020 user.warn mwan3-hotplug[5421]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 2
591: Thu Dec  3 11:45:18 2020 user.notice mwan3track[2843]: Interface wan (wan) is online
592: Thu Dec  3 11:45:19 2020 user.info mwan3track[2844]: Detect ifup event on interface wan1 (lan1)
593: Thu Dec  3 11:45:19 2020 user.notice mwan3track[2844]: Interface wan1 (lan1) is online

I am helping myself now with a delayed restart (10 secs) of mwan3 in rc.local as this seems to help the routing.

will add my configs in a couple of mins

thomaslunze, 2020-12-03:
[mwan3.txt](https://github.com/openwrt/packages/files/5635639/mwan3.txt)
[network.txt](https://github.com/openwrt/packages/files/5635641/network.txt)

hope this helps

pgaufillet, 2020-12-03:
Similar to #14003.

feckert, 2020-12-03:
@thomaslunze could you please switch of the uci option 'check_quality' in your config. And only use ping?
Because I think this message is strange.

> 575: Thu Dec 3 11:45:17 2020 user.info mwan3track[2843]: Check (ping: latency=999999ms loss=100%) failed for target "8.8.8.8" on interface wan (wan). Current score: 6

And then a view lines later I see the following.

> 591: Thu Dec 3 11:45:18 2020 user.notice mwan3track[2843]: Interface wan (wan) is online

I miss some message.

thomaslunze, 2020-12-03:
removed the line quality check and rebooted withour my rc.local mwan3 restart -->
these are all loglines with mwan

493: Fri Dec  4 00:01:21 2020 user.notice mwan3track[2844]: Interface wan1 (lan1) is online
494: Fri Dec  4 00:01:21 2020 user.notice mwan3track[2843]: Interface wan (wan) is online
495: Fri Dec  4 00:01:21 2020 user.notice mwan3-hotplug[2597]: mwan3 hotplug on lan not called because interface disabled
524: Fri Dec  4 00:01:23 2020 user.notice mwan3-hotplug[3993]: mwan3 hotplug on loopback not called because interface disabled
569: Fri Dec  4 00:01:24 2020 user.notice mwan3-hotplug[4510]: Execute ifup event on interface wan (wan)
571: Fri Dec  4 00:01:25 2020 user.warn mwan3-hotplug[4510]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 1
576: Fri Dec  4 00:01:25 2020 user.info mwan3track[2843]: Detect ifup event on interface wan (wan)
577: Fri Dec  4 00:01:25 2020 user.notice mwan3track[2843]: Interface wan (wan) is online
588: Fri Dec  4 00:01:26 2020 user.notice mwan3-hotplug[5558]: Execute ifup event on interface wan1 (lan1)
591: Fri Dec  4 00:01:26 2020 user.warn mwan3-hotplug[5558]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 2
592: Fri Dec  4 00:01:27 2020 user.info mwan3track[2844]: Detect ifup event on interface wan1 (lan1)
593: Fri Dec  4 00:01:27 2020 user.notice mwan3track[2844]: Interface wan1 (lan1) is online

thomaslunze, 2020-12-03:
and here the same log including mwan restart after 10 secs

494: Fri Dec  4 00:03:55 2020 user.notice mwan3track[2845]: Interface wan1 (lan1) is online
495: Fri Dec  4 00:03:55 2020 user.notice mwan3-hotplug[2602]: mwan3 hotplug on lan not called because interface disabled
527: Fri Dec  4 00:03:57 2020 user.notice mwan3-hotplug[3994]: mwan3 hotplug on loopback not called because interface disabled
567: Fri Dec  4 00:03:58 2020 user.notice mwan3-hotplug[4464]: Execute ifup event on interface wan (wan)
573: Fri Dec  4 00:03:59 2020 user.warn mwan3-hotplug[4464]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 1
578: Fri Dec  4 00:03:59 2020 user.info mwan3track[2844]: Detect ifup event on interface wan (wan)
579: Fri Dec  4 00:03:59 2020 user.info mwan3track[2844]: Check (ping) failed for target "8.8.4.4" on interface wan (wan). Current score: 6
580: Fri Dec  4 00:03:59 2020 user.notice mwan3track[2844]: Interface wan (wan) is online
583: Fri Dec  4 00:04:00 2020 user.notice mwan3-hotplug[5324]: Execute ifup event on interface wan1 (lan1)
586: Fri Dec  4 00:04:00 2020 user.warn mwan3-hotplug[5324]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 2
587: Fri Dec  4 00:04:00 2020 user.info mwan3track[2845]: Detect ifup event on interface wan1 (lan1)
588: Fri Dec  4 00:04:00 2020 user.notice mwan3track[2845]: Interface wan1 (lan1) is online
610: Fri Dec  4 00:04:10 2020 user.notice mwan3track[2845]: Stopping mwan3track for interface "wan1". Status was "online"
611: Fri Dec  4 00:04:10 2020 user.notice mwan3track[2844]: Stopping mwan3track for interface "wan". Status was "online"
613: Fri Dec  4 00:04:11 2020 user.notice mwan3-hotplug[6814]: Execute ifup event on interface wan (wan)
614: Fri Dec  4 00:04:11 2020 user.notice mwan3-hotplug[6826]: Execute ifup event on interface wan1 (lan1)
616: Fri Dec  4 00:04:11 2020 user.notice mwan3track[7113]: Interface wan (wan) is online
620: Fri Dec  4 00:04:11 2020 user.notice mwan3track[7114]: Interface wan1 (lan1) is online

feckert, 2020-12-04:
I am not sure and I can not reproduce your issue on my setup.
There have been a few changes.
I can't see which master version you are using.
But I have found some problems, which I have staged [here](https://github.com/TDT-AG/packages/tree/pr/20201204-mwan3).
These were overlooked in the last feature change 
Especially this [commit](https://github.com/TDT-AG/packages/commit/bb714b9883fba3b8bf1ca471604de036ae43eaee) could fix your problem.
@thomaslunze Can give them a try?

thomaslunze, 2020-12-04:
I have changed the file etc/hotplug.d/iface/15-mwan3 and rebooted. Routing not working still.
Find the syslog below

493: Fri Dec  4 14:33:18 2020 user.notice mwan3track[2844]: Interface wan (wan) is online
494: Fri Dec  4 14:33:18 2020 user.notice mwan3track[2845]: Interface wan1 (lan1) is online
495: Fri Dec  4 14:33:18 2020 user.notice mwan3-hotplug[2607]: mwan3 hotplug on lan not called because interface disabled
527: Fri Dec  4 14:33:20 2020 user.notice mwan3-hotplug[3945]: mwan3 hotplug on loopback not called because interface disabled
566: Fri Dec  4 14:33:21 2020 user.notice mwan3-hotplug[4419]: Execute ifup event on interface wan (wan)
572: Fri Dec  4 14:33:22 2020 user.warn mwan3-hotplug[4419]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 1
578: Fri Dec  4 14:33:22 2020 user.info mwan3track[2844]: Detect ifup event on interface wan (wan)
579: Fri Dec  4 14:33:22 2020 user.notice mwan3track[2844]: Interface wan (wan) is online
584: Fri Dec  4 14:33:23 2020 user.notice mwan3-hotplug[5334]: Execute ifup event on interface wan1 (lan1)
585: Fri Dec  4 14:33:23 2020 user.warn mwan3-hotplug[5334]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 2
587: Fri Dec  4 14:33:23 2020 user.info mwan3track[2845]: Detect ifup event on interface wan1 (lan1)
588: Fri Dec  4 14:33:23 2020 user.notice mwan3track[2845]: Interface wan1 (lan1) is online

feckert, 2020-12-04:
I can now also see that table 1 is sometimes incomplete. I am missing routes. After a /etc/init.d/mwan3 restart they are back. Very funny. I have to take a closer look at it now. Can you look if you see a mwa3rtmon in the log when booting?
```
Fri Dec  4 15:58:28 2020 user.warn mwan3rtmon[6891]: failed: 'ip -4 route replace table 1 172.18.0.0/24 dev docker0 proto kernel scope link src 172.18.0.1 linkdown '
Fri Dec  4 15:58:29 2020 user.warn mwan3rtmon[6891]: failed: 'ip -4 route replace table 1 192.168.100.0/24 dev dockertig0 proto kernel scope link src 192.168.100.1 linkdown '
```

thomaslunze, 2020-12-04:
yeah you are right ... same errors here :-)

503: Fri Dec  4 14:36:08 2020 user.notice mwan3-hotplug[3425]: mwan3 hotplug on lan not called because interface disabled
504: Fri Dec  4 14:36:08 2020 user.warn mwan3rtmon[2838]: failed: 'ip -4 route replace table 1 192.168.2.0/24 dev wan proto kernel scope link src 192.168.2.2 linkdown '
513: Fri Dec  4 14:36:09 2020 user.warn mwan3rtmon[2838]: failed: 'ip -4 route replace table 1 192.168.2.0/24 dev wan proto static scope link metric 10 linkdown '
542: Fri Dec  4 14:36:10 2020 user.notice mwan3-hotplug[4286]: mwan3 hotplug on loopback not called because interface disabled
558: Fri Dec  4 14:36:11 2020 user.notice mwan3-hotplug[4804]: Execute ifup event on interface wan (wan)
559: Fri Dec  4 14:36:11 2020 user.warn mwan3-hotplug[4804]: failed to add default via 192.168.2.1 dev wan proto static metric 10 linkdown to table 1
560: Fri Dec  4 14:36:11 2020 user.warn mwan3-hotplug[4804]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 1
561: Fri Dec  4 14:36:11 2020 user.warn mwan3-hotplug[4804]: failed to add 192.168.2.0/24 dev wan proto static scope link metric 10 linkdown to table 1
562: Fri Dec  4 14:36:11 2020 user.info mwan3track[2836]: Detect ifup event on interface wan ()
572: Fri Dec  4 14:36:12 2020 user.notice mwan3-hotplug[5292]: Execute ifup event on interface wan1 (lan1)
575: Fri Dec  4 14:36:12 2020 user.warn mwan3-hotplug[5292]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 2
578: Fri Dec  4 14:36:12 2020 user.info mwan3track[2837]: Detect ifup event on interface wan1 ()
585: Fri Dec  4 14:36:13 2020 user.notice mwan3track[2836]: Interface wan (wan) is online
589: Fri Dec  4 14:36:14 2020 user.warn mwan3rtmon[2838]: failed: 'ip -4 route replace table 1 default via 192.168.2.1 dev wan proto static metric 10 linkdown '
590: Fri Dec  4 14:36:14 2020 user.warn mwan3rtmon[2838]: failed: 'ip -4 route replace table 2 192.168.3.0/24 dev lan1 proto kernel scope link src 192.168.3.2 linkdown '
591: Fri Dec  4 14:36:14 2020 user.warn mwan3rtmon[2838]: failed: 'ip -4 route replace table 2 192.168.3.0/24 dev lan1 proto static scope link metric 20 linkdown '
592: Fri Dec  4 14:36:14 2020 user.notice mwan3track[2837]: Interface wan1 (lan1) is online
603: Fri Dec  4 14:36:19 2020 user.warn mwan3rtmon[2838]: failed: 'ip -4 route replace table 2 default via 192.168.3.1 dev lan1 proto static metric 20 linkdown '
618: Fri Dec  4 14:36:22 2020 user.notice mwan3track[2837]: Stopping mwan3track for interface "wan1". Status was "online"
619: Fri Dec  4 14:36:22 2020 user.notice mwan3track[2836]: Stopping mwan3track for interface "wan". Status was "online"
620: Fri Dec  4 14:36:22 2020 user.notice mwan3-hotplug[6959]: Execute ifup event on interface wan (wan)
621: Fri Dec  4 14:36:22 2020 user.notice mwan3-hotplug[6966]: Execute ifup event on interface wan1 (lan1)
629: Fri Dec  4 14:36:23 2020 user.notice mwan3track[7243]: Interface wan1 (lan1) is online
631: Fri Dec  4 14:36:23 2020 user.notice mwan3track[7242]: Interface wan (wan) is online

feckert, 2020-12-04:
I currently have no configurable logging in mwan3 for debugging. Can you comment out this [line](
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/common.sh#L30) and send me the logging at boot time? Maybe we get more information what exactly is going in mwan3tmon.

thomaslunze, 2020-12-05:
there you go :-)

`455: Fri Dec  4 14:36:29 2020 user.debug mwan3rtmon[2559]: handling route update ipv6 'unreachable fdc0:57d7:8653::/48 dev lo proto static metric 2147483647 pref medium'
467: Fri Dec  4 14:36:29 2020 user.debug mwan3rtmon[2558]: handling route update ipv4 '192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown '
501: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: interface wan is disabled - skipping '192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown '
503: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: interface wan1 is disabled - skipping '192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown '
505: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: handling route update ipv4 '192.168.2.0/24 dev wan proto kernel scope link src 192.168.2.2 linkdown '
506: Fri Dec  4 14:36:30 2020 user.notice mwan3-hotplug[3126]: mwan3 hotplug on lan not called because interface disabled
507: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: adjusting route : 'ip -4 route replace table 1 192.168.2.0/24 dev wan proto kernel scope link src 192.168.2.2 linkdown '
508: Fri Dec  4 14:36:30 2020 user.warn mwan3rtmon[2558]: failed: 'ip -4 route replace table 1 192.168.2.0/24 dev wan proto kernel scope link src 192.168.2.2 linkdown '
509: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: handling route update ipv4 'Deleted 192.168.2.0/24 dev wan proto kernel scope link src 192.168.2.2 linkdown '
512: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: skipping already deleted route table 1 - skipping '192.168.2.0/24 dev wan proto kernel scope link src 192.168.2.2 linkdown '
513: Fri Dec  4 14:36:30 2020 user.debug mwan3rtmon[2558]: handling route update ipv4 '192.168.2.0/24 dev wan proto static scope link metric 10 linkdown '
519: Fri Dec  4 14:36:31 2020 user.debug mwan3rtmon[2558]: adjusting route : 'ip -4 route replace table 1 192.168.2.0/24 dev wan proto static scope link metric 10 linkdown '
520: Fri Dec  4 14:36:31 2020 user.warn mwan3rtmon[2558]: failed: 'ip -4 route replace table 1 192.168.2.0/24 dev wan proto static scope link metric 10 linkdown '
521: Fri Dec  4 14:36:31 2020 user.debug mwan3rtmon[2558]: handling route update ipv4 'default via 192.168.2.1 dev wan proto static metric 10 linkdown '
550: Fri Dec  4 14:36:32 2020 user.notice mwan3-hotplug[4043]: mwan3 hotplug on loopback not called because interface disabled
559: Fri Dec  4 14:36:32 2020 user.notice mwan3-hotplug[4519]: Execute ifup event on interface wan (wan)
563: Fri Dec  4 14:36:32 2020 user.debug mwan3-hotplug[4519]: create_iface_iptables: mwan3_iface_in_wan not in iptables, adding
568: Fri Dec  4 14:36:32 2020 user.warn mwan3-hotplug[4519]: failed to add default via 192.168.2.1 dev wan proto static metric 10 linkdown to table 1
569: Fri Dec  4 14:36:32 2020 user.warn mwan3-hotplug[4519]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 1
570: Fri Dec  4 14:36:32 2020 user.warn mwan3-hotplug[4519]: failed to add 192.168.2.0/24 dev wan proto static scope link metric 10 linkdown to table 1
571: Fri Dec  4 14:36:33 2020 user.info mwan3track[2556]: Detect ifup event on interface wan ()
581: Fri Dec  4 14:36:33 2020 user.debug mwan3rtmon[2559]: handling route update ipv6 'fe80::/64 dev eth0 proto kernel metric 256 pref medium'
583: Fri Dec  4 14:36:33 2020 user.debug mwan3rtmon[2559]: handling route update ipv6 'fe80::/64 dev lan1 proto kernel metric 256 pref medium'
584: Fri Dec  4 14:36:33 2020 user.debug mwan3rtmon[2559]: handling route update ipv6 'fe80::/64 dev wan proto kernel metric 256 pref medium'
586: Fri Dec  4 14:36:34 2020 user.notice mwan3-hotplug[5032]: Execute ifup event on interface wan1 (lan1)
587: Fri Dec  4 14:36:34 2020 user.debug mwan3-hotplug[5032]: create_iface_iptables: mwan3_iface_in_wan1 not in iptables, adding
589: Fri Dec  4 14:36:34 2020 user.warn mwan3-hotplug[5032]: failed to add 192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1 linkdown to table 2
591: Fri Dec  4 14:36:34 2020 user.info mwan3track[2557]: Detect ifup event on interface wan1 ()
599: Fri Dec  4 14:36:35 2020 user.debug mwan3track[2556]: Register ifup event on interface wan ()
602: Fri Dec  4 14:36:35 2020 user.debug mwan3track[2556]: firstconnect: called on wan/wan (wan). Status is online. SRC_IP is 192.168.2.2
603: Fri Dec  4 14:36:35 2020 user.notice mwan3track[2556]: Interface wan (wan) is online
606: Fri Dec  4 14:36:36 2020 user.debug mwan3rtmon[2558]: adjusting route : 'ip -4 route replace table 1 default via 192.168.2.1 dev wan proto static metric 10 linkdown '
607: Fri Dec  4 14:36:36 2020 user.warn mwan3rtmon[2558]: failed: 'ip -4 route replace table 1 default via 192.168.2.1 dev wan proto static metric 10 linkdown '
608: Fri Dec  4 14:36:36 2020 user.debug mwan3rtmon[2558]: handling route update ipv4 '192.168.3.0/24 dev lan1 proto kernel scope link src 192.168.3.2 linkdown '
609: Fri Dec  4 14:36:36 2020 user.debug mwan3rtmon[2558]: adjusting route : 'ip -4 route replace table 2 192.168.3.0/24 dev lan1 proto kernel scope link src 192.168.3.2 linkdown '
610: Fri Dec  4 14:36:36 2020 user.warn mwan3rtmon[2558]: failed: 'ip -4 route replace table 2 192.168.3.0/24 dev lan1 proto kernel scope link src 192.168.3.2 linkdown '
[... cut ...]
642: Fri Dec  4 14:36:44 2020 user.debug mwan3-hotplug[6720]: create_iface_iptables: mwan3_iface_in_wan1 not in iptables, adding
648: Fri Dec  4 14:36:45 2020 user.debug mwan3track[6997]: firstconnect: called on wan/wan (wan). Status is online. SRC_IP is 192.168.2.2
649: Fri Dec  4 14:36:45 2020 user.notice mwan3track[6997]: Interface wan (wan) is online
650: Fri Dec  4 14:36:45 2020 user.debug mwan3track[6998]: firstconnect: called on wan1/wan1 (lan1). Status is online. SRC_IP is 192.168.3.2
651: Fri Dec  4 14:36:45 2020 user.notice mwan3track[6998]: Interface wan1 (lan1) is online
`

thomaslunze, 2020-12-08:
Unfortunately I had to go back to this previous built:

OpenWrt - OpenWrt SNAPSHOT r14154-2d9db93397
LuCI - git-20.219.57394-35e2735

as the router was not responding at all afger a couple of hours. So I stay with this earlier version now

aaronjg, 2020-12-08:
> Similar to #14003.

Yes, I think these are the same issues. Thank you for reporting and the detailed debug logs. I am looking into it.

feckert, 2020-12-16:
Fixed with commit https://github.com/openwrt/packages/commit/21ac1b2c3ebf4735644366e22dccba482a92d558

closed 2020-12-16 by feckert