openwrt/packages #14124: lvm2: compile error

zuzia-dev, 2020-12-03:
```
make package/lvm2/compile V=sc
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/config'
make[1]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1'
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/toolchain'
Makefile:762: WARNING: skipping libgomp -- package has no install section
echo "libc" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/toolchain'
time: package/libs/toolchain/compile#0.24#0.07#0.33
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/feeds/packages/libs/libaio'
mkdir -p /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl
SHELL= flock /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/tmp/.libaio-0.3.112.tar.gz.flock -c '  	/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/download.pl "/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl" "libaio-0.3.112.tar.gz" "ab0462f2c9d546683e5147b1ce9c195fe95d07fac5bf362f6c01637955c3b492" "" "https://releases.pagure.org/libaio"    '
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/feeds/packages/libs/libaio'
time: package/feeds/packages/libaio/compile#0.19#0.06#0.26
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/ncurses'
mkdir -p /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl
SHELL= flock /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/tmp/.ncurses-6.2.tar.gz.flock -c '  	/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/download.pl "/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl" "ncurses-6.2.tar.gz" "30306e0c76e0f9f1f0de987cf1c82a5c21e1ce6568b9227f7da5b71cbea86c9d" "" "@GNU/ncurses"    '
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/ncurses'
time: package/libs/ncurses/host-compile#0.30#0.08#0.37
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/zlib'
mkdir -p /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl
SHELL= flock /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/tmp/.zlib-1.2.11.tar.xz.flock -c '  	/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/download.pl "/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl" "zlib-1.2.11.tar.xz" "4ff941449631ace0d4d203e3483be9dbc9da454084111f97ea0a2114e19bf066" "" "@SF/libpng" "http://www.zlib.net"    '
echo "zlib" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/zlib'
time: package/libs/zlib/compile#0.29#0.10#0.40
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/ncurses'
mkdir -p /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl
SHELL= flock /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/tmp/.ncurses-6.2.tar.gz.flock -c '  	/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/download.pl "/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl" "ncurses-6.2.tar.gz" "30306e0c76e0f9f1f0de987cf1c82a5c21e1ce6568b9227f7da5b71cbea86c9d" "" "@GNU/ncurses"    '
echo "terminfo" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/ncurses.default.install
echo "libncurses" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/ncurses.default.install
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/ncurses'
time: package/libs/ncurses/compile#0.35#0.10#0.43
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/readline'
mkdir -p /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl
SHELL= flock /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/tmp/.readline-8.0.tar.gz.flock -c '  	/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/download.pl "/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl" "readline-8.0.tar.gz" "e339f51971478d369f8a053a330a190781acb9864cf4c541060f12078948e461" "" "@GNU/readline"    '
echo "libreadline" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/readline.default.install
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/libs/readline'
time: package/libs/readline/compile#0.29#0.09#0.38
make[2]: Entering directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/package/utils/util-linux'
mkdir -p /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl
SHELL= flock /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/tmp/.util-linux-2.36.tar.xz.flock -c '  	/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/scripts/download.pl "/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/dl" "util-linux-2.36.tar.xz" "9e4b1c67eb13b9b67feb32ae1dc0d50e08ce9e5d82e1cccd0ee771ad2fa9e0b1" "" "@KERNEL/linux/utils/util-linux/v2.36"    '
echo "libblkid" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
echo "libfdisk" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
echo "libmount" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
echo "libsmartcols" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
echo "libuuid" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
echo "blkid" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
echo "fdisk" >> /home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/staging_dir/target-mips_24kc_musl/pkginfo/util-linux.default.install
[... cut ...]
make[2]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/feeds/packages/utils/lvm2'
time: package/feeds/packages/lvm2/compile#22.70#9.25#30.92
    ERROR: package/feeds/packages/lvm2 failed to build.
make[1]: *** [package/Makefile:114: package/feeds/packages/lvm2/compile] Error 1
make[1]: Leaving directory '/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1'
make: *** [/home/zuzia/.openwrt/kompilacje/source-eb904-new-kernel-v12/openwrt-xrx200-1/include/toplevel.mk:242: package/lvm2/compile] Błąd 2
```

neheb, 2020-12-21:
Please test the updated version.

hnyman, 2023-01-14:
Apparently fixed by now.
(newer lvm2 version currently in use in all supported branches, in any case)

closed 2023-01-14 by hnyman