openwrt/packages #14126: php7-mod-iconv: does not support NextCloud's mail app

flyn-org, 2020-12-03:
Maintainer: @mhei 
Environment: x86_64

Description:

The php7-mod-iconv package does not seem sufficient to support NextCloud's mail app; see https://github.com/nextcloud/server/issues/24523. This seems to be a problem shared with Alpine, another minimalist distribution, as noted [here](https://github.com/nextcloud/docker/issues/1299).

Strangely, running ldd on /usr/lib/php/iconv.so prints this:
```
# ldd /usr/lib/php/iconv.so 
	ldd (0x7f95ad080000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f95ad05f000)
	libc.so => ldd (0x7f95ad080000)
```
I would have thought it would pull in libiconv.so too.

I am not yet sure what needs to change, but statements like `iconv("UTF-8", "UTF-8//IGNORE", $email)` do not seem to work in PHP.

Does php7-mod-iconv provide only a subset of `iconv`'s facilities? From what I can tell, PHP's iconv module supports compiling with and without libiconv(-full). Could we optionally compile against the full library?

mhei, 2020-12-03:
What is your setting for `CONFIG_BUILD_NLS`, i.e. "Compile with full language support" in "Global build settings"?

neheb, 2020-12-04:
The Turris people have nextcloud running on 18.06 and 19.07. No idea what their solution is.

mhei, 2020-12-04:
When you enable `CONFIG_BUILD_NLS`, then iconv.so is linked against libiconv and then it should work. However, this is enabled by default AFAIK, so the usage of the prebuild packages of the buildbots is limited (by design).

flyn-org, 2020-12-04:
Sounds promising. I should be able to perform a test this weekend.

flyn-org, 2020-12-06:
Building php7-mod-iconv with CONFIG_BUILD_NLS seems to fix this. Not a bug.

NextCloud and its Mail app now run fine. Thank you!

neheb, 2020-12-07:
So under musl iconv is built into the libc. Also, nls.mk uses libiconv-stub, which is a static library. That's why it doesn't show up with ldd.

flyn-org, 2020-12-07:
When I compile with CONFIG_BUILD_NLS, as suggested by @mhei, php7-mod-iconv does pull in libiconv.so:

```
# ldd /usr/lib/php/iconv.so 
	ldd (0x7f0f57b6f000)
	libiconv.so.2 => /usr/lib/libiconv.so.2 (0x7f0f57a83000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x7f0f57a6f000)
	libc.so => ldd (0x7f0f57b6f000)
	...
```

neheb, 2020-12-07:
Yesz because libiconv-full is a shared library. It's also bigger.

mhei, 2020-12-13:
Closing, since reported told us that it is not considered a bug.

closed 2020-12-13 by mhei