openwrt/packages #14139: [lighttpd] -mod-wolfssl fails (requires dependency on -mod-wolfssl?)

n8v8R, 2020-12-05:
lighttpd:1.4.56-3
lighttpd-mod-wolfssl: 1.4.56-3

Maintainer: @flyn-org
Tag: @gstrauss 
Environment: mvebu/cortexa9
___
```
# lighttpd -tt -f /etc/lighttpd/lighttpd.conf

(../src/configfile.c.278) Warning: please add "mod_openssl" to server.modules list in lighttpd.conf.  A future release of lighttpd 1.4.x *will not* automatically load mod_openssl and lighttpd *will not* use SSL/TLS where your lighttpd.conf contains ssl.* directives
(../src/mod_wolfssl.c.2052) SSL: WARNING: SNI callbacks *crippled* in wolfSSL library build
```

It seems odd/wrong that the WolfSSL module should be dependent on OpenSSL module (installed and enabled)

gstrauss, 2020-12-05:
Thanks for finding this.  That looks like a bug in attempted compatibility logic, which I missed when I split mod_wolfssl from mod_openssl.  (mod_wolfssl used to be part of mod_openssl, and you could build lighttpd against openssl *or* wolfssl.)  A workaround is to set `server.compat-module-load = "disable"` and to manually include one or more of `"mod_indexfile"`, `"mod_dirlisting"`, and `"mod_staticfile"` in `server.modules`.  I'll submit a patch shortly:
```
diff --git a/src/configfile.c b/src/configfile.c
index da444154..c7739c4f 100644
--- a/src/configfile.c
+++ b/src/configfile.c
@@ -345,6 +345,8 @@ static void config_compat_module_load (server *srv) {
             append_mod_openssl = 0;
         else if (buffer_eq_slen(m, CONST_STR_LEN("mod_openssl")))
             append_mod_openssl = 0;
+        else if (buffer_eq_slen(m, CONST_STR_LEN("mod_wolfssl")))
+            append_mod_openssl = 0;
         else if (buffer_eq_slen(m, CONST_STR_LEN("mod_authn_file")))
             append_mod_authn_file = 0;
         else if (buffer_eq_slen(m, CONST_STR_LEN("mod_authn_ldap")))
```

gstrauss, 2020-12-05:
BTW, that other warning `SSL: WARNING: SNI callbacks *crippled* in wolfSSL library build` means that wolfssl in openwrt was built without support for SNI, something that severely limits wolfssl use in a web server.  I'll look into that in a bit and will open a separate issue for discussion.

PR #14143 (2020-12-06): lighttpd: wolfssl TLS compat module detection
Maintainer: W. Michael Petullo mike@flyn.org
Compile tested: x86-64 OpenWrt master branch

Description:
lighttpd: wolfssl TLS compat module detection

x-ref:
  "[lighttpd] -mod-wolfssl fails (requires dependency on -mod-openssl?)"
  https://github.com/openwrt/packages/issues/14139

github: closes #14139
commit b575b915 (2020-12-06): lighttpd: wolfssl TLS compat module detection

x-ref:
  "[lighttpd] -mod-wolfssl fails (requires dependency on -mod-openssl?)"
  https://github.com/openwrt/packages/issues/14139

github: closes #14139
closed 2020-12-06 by neheb