openwrt/packages #14144: Transmission crash

Gingernut1978, 2020-12-05:
Transmission crashes after loading a torrent.

```
x86_64
Model: Micro-Star International Co., Ltd. Cubi N 8GL (MS-B171)
Architecture: Intel(R) Celeron(R) N4000 CPU @ 1.10GHz
Firmware Version:OpenWrt SNAPSHOT r15132-9c13513ce1 / LuCI Master git-20.339.47288-086f323
Kernel Version: 5.4.80
```

```
Sat Dec  5 17:35:44 2020 kern.info kernel: [27752.826230] traps: transmission-da[5705] general protection fault ip:7f68e5032deb sp:7f68e4f49320 error:0 in libwolfssl.so.24.2.0[7f68e4fbe000+7a000]
Sat Dec  5 17:35:49 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:35:58 2020 kern.info kernel: [27766.863157] traps: transmission-da[6901] general protection fault ip:7fca9a70cdeb sp:7fca9a623320 error:0 in libwolfssl.so.24.2.0[7fca9a698000+7a000]
Sat Dec  5 17:36:03 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:12 2020 kern.info kernel: [27780.895850] traps: transmission-da[6912] general protection fault ip:7f369c520deb sp:7f369c437320 error:0 in libwolfssl.so.24.2.0[7f369c4ac000+7a000]
Sat Dec  5 17:36:17 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:26 2020 kern.info kernel: [27794.925846] traps: transmission-da[6946] general protection fault ip:7f7498eb6deb sp:7f7498dcd320 error:0 in libwolfssl.so.24.2.0[7f7498e42000+7a000]
Sat Dec  5 17:36:31 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:41 2020 kern.info kernel: [27809.456663] traps: transmission-da[6951] general protection fault ip:7f9f65a56deb sp:7f9f6596d320 error:0 in libwolfssl.so.24.2.0[7f9f659e2000+7a000]
Sat Dec  5 17:36:46 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:55 2020 kern.info kernel: [27823.491442] traps: transmission-da[6960] general protection fault ip:7f405b383deb sp:7f405b29a320 error:0 in libwolfssl.so.24.2.0[7f405b30f000+7a000]
Sat Dec  5 17:37:00 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:37:09 2020 daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 9 seconds since last crash
Sat Dec  5 17:37:09 2020 kern.info kernel: [27837.522517] traps: transmission-da[6965] general protection fault ip:7f02f99c
```

neheb, 2020-12-06:
@dango found this same problem too: https://github.com/transmission/transmission/issues/1537

I don't know what the actual problem is. Wrong API usage? ping @cotequeiroz

dangowrt, 2020-12-26:
Not much has been happening there. Maybe we can just have the build variants back to people can choose transmission-mbedtls or transmission-openssl instead of transmission-wolfssl? (I anyway believe this is one of the greatest strength of OpenWrt as a distribution, to give users those choices!)

neheb, 2020-12-26:
I'll whip up a PR when I can.

Gingernut1978, 2023-11-04:
This was solved with an update

closed 2023-11-04 by Gingernut1978