openwrt/packages #14145: miniupnpd not creating redirects since version 2.2

danielvijge, 2020-12-05:
Maintainer: @neheb 
Environment: TP-Link TL-WDR4300 v1 / Atheros AR9344 rev 2 / OpenWrt 19.07.4 r11208-ce6496d796 / LuCI openwrt-19.07 branch git-20.334.47525-e9bfb12
Package version: miniupnpd 2.2.0-3

Description:

Miniupnpd fails to create redirects for UPnP devices. This happened since the upgrade to 2.2.0-x. Downgrading the package to 2.1.20200510-3 (the latests for which I could find a mirror) solves the problem.
My network setup consists of a modem/router, and an openwrt router behind it. Only the modem/router have a public IP address, the openwrt router only an internal IP address. The openwrt router is configured as the DMZ host in the modem/router. In upnp config I configured my public IP address.
I mainly performed tests with an Xbox, which on 2.1.x gets a redirect like "Teredo 10.0.0.39:3074->3074 UDP". On 2.2.x nothing happens.

**/etc/config/upnp**
```
config upnpd 'config'
	option download '1024'
	option upload '512'
	option internal_iface 'lan'
	option port '5000'
	option upnp_lease_file '/var/run/miniupnpd.leases'
	option igdv1 '1'
	option ext_ip_reserved_ignore '1'
	option enabled '1'
	option uuid '3a87f98a-ad33-4311-8055-4ed3532da4ed'
	option external_ip '62.x.y.z' # My public IP address

config perm_rule
	option action 'allow'
	option ext_ports '1024-65535'
	option int_addr '0.0.0.0/0'
	option int_ports '1024-65535'
	option comment 'Allow high ports'

config perm_rule
	option action 'deny'
	option ext_ports '0-65535'
	option int_addr '0.0.0.0/0'
	option int_ports '0-65535'
	option comment 'Default deny'
```

neheb, 2020-12-05:
These are the two relevant commits: https://github.com/openwrt/packages/commit/7077fd8c79eef5f044959bacfd6f66252850968a#diff-a1823a37eb35b2bdde74acb83321ae465da02b1e9b045509abbb2bde3650fcfd https://github.com/openwrt/packages/commit/b76aa9919489f49b472a8f939f6d46ca33d05f64#diff-a1823a37eb35b2bdde74acb83321ae465da02b1e9b045509abbb2bde3650fcfd

Since your issue seems to be an IPv6 one, I recommend undoing the changes on the first one and see what happens.

The file to edit is /etc/init.d/miniupnpd . You can do so with vi. Or opkg install nano

danielvijge, 2020-12-06:
Testing as well with upnpc command as `upnpc -a 10.0.0.100 2000 2000 UDP` from a laptop.

Indeed the change introduced in 7077fd8c79eef5f044959bacfd6f66252850968a breaks the possibility to create a redirect. With this undone, or with `option ipv6_disable '1'` `upnpc` can create a redirect. Otherwise `upnpc` exits with
```
connect: Connection refused
No valid UPNP Internet Gateway Device found.
```
It could be that my case is a bit strange, because I don't have native IPv6, but an IPv6-in-IPv4 tunnel through Hurricane Electic.

The case of the xbox is a still strange. With miniupnpd 2.1 it showed a redirect, usually for port 3074. With 2.2 this redirect is no longer created. Yet the xbox dashboard shows the NAT Type sometimes as moderate and sometimes as open. In both cases, there is nothing in the firewall rules.

neheb, 2020-12-06:
ping @dianlujitao

dianlujitao, 2020-12-07:
I'm able to reproduce the issue, `upnpc` started working after disabling ipv6, but transmission magically works in both cases thus I didn't notice the regression.

In theory https://github.com/openwrt/packages/commit/7077fd8c79eef5f044959bacfd6f66252850968a "fixed" `ipv6_listening_ip`, but this option doesn't work as we might expect.

1. When ipv6 enabled and `ipv6_listening_ip` not set (the case with 2.1.x), it defaults to [accept all inbound addrs](https://github.com/miniupnp/miniupnp/blob/ba130320f4180f8f9a54c48c0634e0ea0118fa38/miniupnpd/miniupnpd.c#L1191), which is equivalent to `::` if I don't get it wrong.
2. When ipv6 enabled and `ipv6_listening_ip` defined (the case after https://github.com/openwrt/packages/commit/7077fd8c79eef5f044959bacfd6f66252850968a), it [listens to the addr _only_](https://github.com/miniupnp/miniupnp/blob/ba130320f4180f8f9a54c48c0634e0ea0118fa38/miniupnpd/miniupnpd.c#L405) and _ignores_ `listening_ip`, which means __all requests from ipv4 addr will be rejected__, this explains the `Connection refused` error.
3. With `option ipv6_disable '1'`, `ipv6_listening_ip` becomes a no-op.

I'm not able to verify 2 because I didn't found how to make upnpc send the request through ipv6 lan addr, but I guess transmission is able to send such requests so I didn't notice the issue.

A simple yet effective fix is to remove `ipv6_listening_ip` to restore 2.1.x behavior.

closed 2020-12-07 by neheb

danielvijge, 2020-12-10:
For the changes introduced in b76aa9919489f49b472a8f939f6d46ca33d05f64, I cannot fully test if the work or not. But the patch was also submitted upstream, and there was some comment about it: https://github.com/miniupnp/miniupnp/pull/511#issuecomment-737561628 Do you see that comment?

neheb, 2020-12-11:
More info: https://github.com/openwrt/packages/issues/13773

I don't run such a setup so I can't comment on it.

danielvijge, 2020-12-11:
My setup is with a double NAT (192.168.100.x for modem; 10.0.0.x for router). As long as `ext_ip` is configured to the public IP address, I can create redirects. It does require of course that the router is configured as a DMZ in the modem, so all ports are forwarded from modem to router first.

professor-jonny, 2021-01-17:
I'm haveing the same issue with version 2.2.1 I dont get any rules created just the chains one on start of miniupnpd.

I have three NATS basically user provided router that I must use set up with dmz to my openwrt router and a carrier grade nat at the providers offices.

I use stun to get my public adress of the providers CGNAT and get this in my log:

`daemon.warn miniupnpd[4427]: STUN: ext interface eth1 with IP address 192.168.1.100 is now behind restrictive NAT with public IP address 115.189.81.247: Port forwarding is now impossible`

UPNP works with the providers router and I guess when the ports are accessed they are blindly opened and forwarded to me as I get open port tests with a port scan, I dont know what happens if two people request the same port behind the same nat but I have not had issues so far.

Neustradamus, 2021-08-31:
@ all: What is the status of this ticket?

professor-jonny, 2021-08-31:
There is a new miniupnp app that works  Windows based devices.
Windows and xbox does not support the v2 standard and it was the reason for it nor working for me.