openwrt/packages #14149: zerotier:

BKPepe, 2020-12-06:
Maintainer: @mwarning 
Environment: Turris MOX, cortexa53, OpenWrt 19.07

Description:

Since update of zerotier in OpenWrt 19.07, it fails to compile for Turris MOX, however it works for Turris Omnia (cortex-a9_vfpv3-d16) and Turris 1.x (power 8540)

Snip:
```
Applying ./patches/0001-find-miniupnpc.h-in-staging-directory.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0002-remove-pie.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0003-remove-arm32-conservative-CFLAGS.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0004-accept-external-linker-flags.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0005-link-natpmp.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0006-gcc10.patch using plaintext: 
patching file osdep/Binder.hpp

Applying ./patches/0007-add-cerrno-header-for-str-errno.patch using plaintext: 
patching file osdep/LinuxNetLink.cpp

Applying ./patches/0008-fix-compilation-for-arm_cortex-a7-neon.patch using plaintext: 
patching file node/Constants.hpp
make[3]: Entering directory '/home/mox-19.07//build/build_dir/target-aarch64_cortex-a53_musl/ZeroTierOne-1.6.2'
ccache_cxx -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/mox-19.07//build/build_dir/target-aarch64_cortex-a53_musl/ZeroTierOne-1.6.2:ZeroTierOne-1.6.2 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/mox-19.07//build/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/mox-19.07//build/staging_dir/target-aarch64_cortex-a53_musl/include -I/home/mox-19.07//build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/usr/include -I/home/mox-19.07//build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include/fortify -I/home/mox-19.07//build/staging_dir/toolchain-aarch64_cortex-a53_gcc-7.5.0_musl/include  -Wall -Wno-deprecated -std=c++11 -pthread  -DNDEBUG  -DZT_USE_MINIUPNPC -DZT_USE_SYSTEM_MINIUPNPC -DZT_USE_SYSTEM_NATPMP -DZT_NO_TYPE_PUNNING -DZT_ARCH_ARM_HAS_NEON -march=armv8-a+aes+crypto -mtune=generic -mstrict-align -DZT_BUILD_PLATFORM=1 -DZT_BUILD_ARCHITECTURE=4 -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -D_MT_ALLOCATOR_H -D_POOL_ALLOCATOR_H -D_EXTPTR_ALLOCATOR_H -D_DEBUG_ALLOCATOR_H   -c -o node/AES.o node/AES.cpp
cc1plus: error: invalid feature modifier in '-march=armv8-a+aes+crypto'
make[3]: *** [<builtin>: node/AES.o] Error 1
make[3]: Leaving directory '/home/mox-19.07//build/build_dir/target-aarch64_cortex-a53_musl/ZeroTierOne-1.6.2'
make[2]: *** [Makefile:81: /home/mox-19.07//build/build_dir/target-aarch64_cortex-a53_musl/ZeroTierOne-1.6.2/.built] Error 2
time: package/feeds/packages/zerotier/compile#0.96#0.57#1.26
```

Faillog: https://downloads.openwrt.org/releases/faillogs-19.07/aarch64_cortex-a53/packages/zerotier/compile.txt

RookieZoe, 2020-12-07:
Same issue, I guess the feature modifier is '-march=armv8-a+crypto' instead of '-march=armv8-a+aes+crypto'.

mwarning, 2020-12-08:
There has been some change in the ZeroTier package: https://github.com/openwrt/packages/pull/14155
Please try again and let us know. :-)

BKPepe, 2020-12-08:
Nope, didn't work.

Gist: https://gist.github.com/BKPepe/bbf995477b72e24f39f490d6b4084caa

mwarning, 2020-12-25:
Since the Turris MOX does not seem to be supported by OpenWrt - how can I reproduce the error?

BKPepe, 2020-12-25:
This bug is not specific to one device (Turris MOX), but it affects all the routers, which are using the ARMv8 processor, however, Turris MOX uses SoC Marvell Armada 3720 and I see there are routers supported by OpenWrt like ESPRESSObin. You might see from my OP that even it fails for aarch64_cortex-a53 (target which is using Turris MOX) according to buildbots for OpenWrt 19.07 release while it works for OpenWrt master.

You can reproduce this error by following steps:

1. Download SDK (mvebu-cortexa53), which is enough and we don't need to select specific router in later steps to reproduce this error.
```
https://downloads.openwrt.org/releases/19.07.5/targets/mvebu/cortexa53/openwrt-sdk-19.07.5-mvebu-cortexa53_gcc-7.5.0_musl.Linux-x86_64.tar.xz
```

2. Uncompress SDK
```
tar xf openwrt-sdk-19.07.5-mvebu-cortexa53_gcc-7.5.0_musl.Linux-x86_64.tar.xz 
```

3. Proceed to extracted folder and update & install all feeds (just to be sure)
```
./scripts/feeds update -a
./scripts/feeds install -a
```

4. Compile package zerotier
```
make package/zerotier/compile -jX V=s
```

And you willl be able reproduce bug present in the stable release.

BKPepe, 2021-02-14:
Fixed by this https://github.com/openwrt/packages/commit/db69f0b57c08c2df8d864095b9baa091a264d217

Closing.

closed 2021-02-14 by BKPepe