openwrt/packages #14179: miniupnpd: Daemon not starting during boot

urbenlegend, 2020-12-09:
# Issue template (remove lines from top till here)

Maintainer: @neheb
Environment: Netgear R7800, OpenWrt 19.07.5

Description: With the upgrade to 19.07.5, miniupnpd (2.2.0-3 as of this writing) is no longer starting at boot. In the System Logs, I see:

```
Tue Dec  8 17:45:22 2020 user.notice upnp daemon: external interface not found, not starting
Tue Dec  8 17:45:22 2020 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue Dec  8 17:45:22 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led USB 1
Tue Dec  8 17:45:22 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led USB 2
Tue Dec  8 17:45:22 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led WAN
Tue Dec  8 17:45:22 2020 daemon.notice procd: /etc/rc.d/S96led: setting up led eSATA
Tue Dec  8 17:45:22 2020 daemon.notice procd: /etc/rc.d/S96led: Skipping trigger 'ide-disk' for led 'eSATA' due to missing kernel module
Tue Dec  8 17:45:22 2020 daemon.warn miniupnpd[1066]: Cannot get IP address for ext interface . Network is down
Tue Dec  8 17:45:22 2020 daemon.notice netifd: Interface 'wg' is now up
Tue Dec  8 17:45:22 2020 daemon.notice netifd: Network device 'wg' link is up
Tue Dec  8 17:45:22 2020 daemon.err miniupnpd[1066]: bind(http): Address not available
Tue Dec  8 17:45:22 2020 daemon.err miniupnpd[1066]: Failed to open socket for HTTP. EXITING
```
and later on when the other interfaces are brought up, I see these:
```
Tue Dec  8 17:45:36 2020 daemon.notice miniupnpd[2146]: private/reserved address checking is ignore
Tue Dec  8 17:45:36 2020 daemon.err miniupnpd[2146]: bind(http): Address not available
Tue Dec  8 17:45:36 2020 daemon.err miniupnpd[2146]: Failed to open socket for HTTP. EXITING
```

If I go into System -> Startup and manually restart miniupnpd, the daemon starts up fine and ports get automatically opened. This issue only occurs whenever I reboot the router.

neheb, 2020-12-09:
Try changing https://github.com/openwrt/packages/blob/master/net/miniupnpd/files/miniupnpd.init#L4 in /etc/init.d/miniupnpd to 99.

I don't think it will work though. I believe miniupnpd is hotplugged.

neheb, 2020-12-09:
There's also a version 4. Run opkg update && opkg upgrade miniupnpd

urbenlegend, 2020-12-09:
I tried 99 and it did nothing as you correctly pointed out.

It seems like version 4 hasn't made it into the 19.07.5 repos just yet. I'll try it when they update it. Or is there an easy way to get a version 4 package?

neheb, 2020-12-09:
Seems I didn't backport it. Did just now.

Plan B: rename /etc/hotplug.d/iface/50-miniupnpd to /etc/hotplug.d/iface/90-miniupnpd and reboot.

urbenlegend, 2020-12-11:
It seems like the new package works! It started automatically upon reboot.

neheb, 2020-12-11:
I take it this can be closed?

urbenlegend, 2020-12-14:
Yes, I've tested a reboot twice and both times, miniupnpd was able to start up on its own. Thanks for the help. Closing.

closed 2020-12-14 by urbenlegend