openwrt/packages #14180: Subversion does not compile as APRUTIL can not be found

BKPepe, 2020-12-09:
Maintainer: @val-kulkov 

Environment: arm_cortex-a9_vfpv3-d16, OpenWrt master

Description:

Subversion does not compile.
Output:

```
configure: Apache Portable Runtime Utility (APRUTIL) library configuration
checking for APR-util... no
configure: WARNING: APRUTIL not found
The Apache Portable Runtime Utility (APRUTIL) library cannot be found.
Install APRUTIL on this system and configure Subversion with the
 appropriate --with-apr-util option.

configure: error: no suitable APRUTIL found
Makefile:121: recipe for target '/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/subversion-1.14.0/.configured_a17fb5ef857664f03cd0ce37cc5ea591' failed
make[3]: *** [/builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/subversion-1.14.0/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
time: package/feeds/packages/subversion/compile#22.69#3.09#32.36
```

More details: https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/subversion/compile.txt

Caused by this @neheb's commit: https://github.com/openwrt/packages/commit/465774e9b5543fedc7681e96caf95e0bfb173bf5

neheb, 2020-12-09:
Wait a day. It'll go away.

I didn't push the apr-util commit: https://github.com/openwrt/packages/commit/04b597a52d81f2f0f10ae94d40f530e2843314e7

BKPepe, 2020-12-09:
Ok! Thanks.

neheb, 2020-12-10:
Buildbots were updated.

closed 2020-12-10 by neheb