openwrt/packages #14197: modemmanager: How to disable tty interface lock

zokl, 2020-12-11:
Maintainer: @nickberry17 @aleksander0m
Environment: ModemManager 1.14.8

Is there any way how to disable the lock of ttyUSB interfaces by modemmanager? I use this interface for gpsd and cmd scripts. My modem uses /dev/cdc-wdm0.

I found o solution for udev with parameter ENV{ID_MM_PORT_IGNORE}=“1”, but I don't know the way in OpenWRT.

I have sucessfully tried delete the rule 25-modemmanager-tty or records in cache files but it is not a good way that i imagine.

zokl, 2020-12-11:
> Adding a custom "udev" rule setting `ENV{ID_MM_PORT_IGNORE}=“1”` also works in OpenWRT, with the built-in udev rules parser. Please try that and if it doesn't work post here the full rules file you're using to review it.

I am not sure I hotplug.d daemon is able to handle with udev rules. I have tried code bellow but without success. Is it correct?

```
#!/bin/sh

if [ ${SUBSYSTEM} = "tty" ]; then
    if [ "${ACTION}" = "add" ]; then
#       logger "Nastaven PORT ignore FLAG"
        ID_MM_PORT_IGNORE="1"
    fi
fi
```

Modification in the way of your rules works but i need to include your environment:

aleksander0m, 2020-12-11:
No, you don't need to do anything with hotplug. You need to create a udev rules file as you would in a system with udev. E.g. replace with your correct idVendor (instead of 0421) with your correct idProduct (instead of 060D) and with the correct USB interface to ignore (instead of 01):
```
# vim /lib/udev/rules.d/77-mm-test.rules
ACTION!="add|change|move|bind", GOTO="mm_test_end"
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0421", GOTO="mm_test_rules"
GOTO="mm_test_end"

LABEL="mm_test_rules"
SUBSYSTEMS=="usb", ATTRS{bInterfaceNumber}=="?*", ENV{.MM_USBIFNUM}="$attr{bInterfaceNumber}"
ATTRS{idVendor}=="0421", ATTRS{idProduct}=="060D", ENV{.MM_USBIFNUM}=="01", ENV{ID_MM_PORT_IGNORE}="1"
LABEL="mm_test_end"
```

zokl, 2020-12-11:
> No, you don't need to do anything with hotplug. You need to create a udev rules file as you would in a system with udev. E.g. replace with your correct idVendor (instead of 0421) with your correct [...]

Thank you for your help. It works in this way.

closed 2020-12-11 by zokl