openwrt/packages #14199: Python3 package misconfigured

iacchi, 2020-12-11:
Hello, I have a problem with a misconfiguration of the python3 package, which doesn't let me use numpy and other c-extensions packages. This is the third time I open this bug in 3 different projects, please see here: https://github.com/Entware/Entware/issues/568 what the Entware devs have to say (they told me to contact you about this) and here: https://github.com/numpy/numpy/issues/17975 the problem explained in detail.

ja-pa, 2020-12-12:
cc @jefferyto @commodo

commodo, 2020-12-12:
Oh.
This traveled a few projects to get here.

So, yes, for the Python SO suffix, we don't add the ABI suffix.
i.e. this patch: https://github.com/openwrt/packages/blob/master/lang/python/python3/patches/014-remove-platform-so-suffix.patch

I tried to remove it, but it is difficult.
It is a bit of work to match the ABI suffixes for all the targets; OpenWrt supports quite a few architectures, and I don't want to allow a wildcard match for the suffix [quite yet]. Though it may not be a problem [the wildcard matching].

I will try to think of a better solution.
But I see that you want numpy. So, why not just use the one that comes from OpenWrt?
https://github.com/openwrt/packages/blob/master/lang/python/numpy/Makefile

iacchi, 2020-12-12:
Because I guess I didn't realise it was there :P I'll try to compile it (I'm new to compiling with Entware/OpenWrt, although not with Linux in general) and see how it goes. Still, it's something worth figuring out, for a better integration with pip and the whole python ecosystem.

commodo, 2020-12-12:
> Because I guess I didn't realise it was there :P I'll try to compile it (I'm new to compiling with Entware/OpenWrt, although not with Linux in general) and see how it goes. Still, it's something worth figuring out, for a better integration with pip and the whole python ecosystem.

I may probably need to think of a solution for Python.
That’s where all this starts.
But some things on Openwrt are special/specific.
That patch (014) now is smaller than it used to be.
And I will try to remove it again.
But testing it means building everything; and it’s obvious that it would take some hours to validate.

jefferyto, 2020-12-12:
I think it's still necessary for us to remove the platform triplet from .so file names to be able to cross-compile Python packages (see #12754).

My preference would be to disallow target pip from installing platform-specific wheels, but this is probably too extreme a view for other people :joy:

jefferyto, 2021-06-21:
@iacchi Did you try installing numpy with opkg (`opkg install python3-numpy`)?

anguschang007, 2022-11-20:
> @iacchi Did you try installing numpy with opkg (`opkg install python3-numpy`)?

the opkg no the package python3-numpy! and other modules have the same issue when installing them by pip! Does anyone have a better solution?

closed 2026-04-05 by commodo