openwrt/packages #14200: Node-v14.15.1 compile error in master branch!

zxlhhyccc, 2020-12-11:
````
home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/node-v14.15.1/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o.d.raw   -c
In file included from /usr/lib/gcc/x86_64-linux-gnu/9/include/stdint.h:9,
                 from ../deps/v8/src/base/bits.h:8,
                 from ../deps/v8/src/base/bits.cc:5:
/usr/include/stdint.h:26:10: fatal error: bits/libc-header-start.h: No such file or directory
   26 | #include <bits/libc-header-start.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
tools/v8_gypfiles/v8_libbase.host.mk:157: recipe for target '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/node-v14.15.1/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o' failed
make[4]: *** [/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/node-v14.15.1/out/Release/obj.host/v8_libbase/deps/v8/src/base/bits.o] Error 1
Makefile:104: recipe for target 'node' failed
make[3]: *** [node] Error 2
make[3]: Leaving directory '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/node-v14.15.1'
Makefile:152: recipe for target '/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/node-v14.15.1/.built' failed
make[2]: *** [/home/lin/openwrt-luci-master/build_dir/target-arm_cortex-a9_musl_eabi/node-v14.15.1/.built] Error 2
make[2]: Leaving directory '/home/lin/openwrt-luci-master/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#1.08#0.70#4.96
    ERROR: package/feeds/packages/node failed to build.
package/Makefile:112: recipe for target 'package/feeds/packages/node/compile' failed
```````

neheb, 2020-12-11:
Install gcc-multilib

Habbie, 2021-04-22:
Posting this here in case somebdy else runs into an error around `#include <bits/libc-header-start.h>` - for me (working locally on some updates to the `net/dnsdist` package), installing `gcc-multilib` only brought me to the next problem, because the actual issue was that my target build had managed to included some headers from my host's /usr/include because https://github.com/tsuna/boost.m4 tries those if it can't find Boost in the right spaces.

So, tl;dr: this error may mean your host includes are leaking into your target builds, and in that case you should fix that instead.

GeorgeSapkin, 2025-11-08:
Closing as stale.