openwrt/packages #14244: qemu-ga: 4.0 and 5.0 can't shutdown OpenWRT virtual machine

RussianNeuroMancer, 2020-12-16:
Maintainer: @yousong
Environment: OpenWRT 19.07.5 armvirt64

Description:

Due to [issue on OpenWRT side](https://bugs.openwrt.org/index.php?do=details&task_id=3101&string=virtio&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=), acpid is not an option on armvirt64 target, so I had to resort to qemu-ga for shutdown/reboot, but it doesn't work either. As you can see in log below qemu-ga receive guest-shutdown message from host, but for some reason can not execute shutdown command inside virtual machine. This cause delay on host shutdown.

This issue was also [mentioned on OpenWRT forum](https://forum.openwrt.org/t/openwrt-proxmox-reboot-shutdown-timeout/59903) in context of Proxmox (which means it's also reproducible on x86_64, not only on armvirt64). [Here](https://gitlab.alpinelinux.org/alpine/aports/commit/76b81b486480fd9c3294cd420bcf2df01c27790d) you can take a look at qemu-ga patch for Alpine Linux.

```
root@router:~# qemu-ga --verbose
1608137479.770384: debug: received EOF
1608137487.692824: debug: read data, count: 59, data: {"execute":"guest-sync", "arguments":{"id":1608137487691}}

1608137487.693193: debug: process_event: called
1608137487.693309: debug: processing command
1608137487.693584: debug: sending data, count: 26
1608137487.696358: debug: read data, count: 62, data: {"execute":"guest-shutdown","arguments":{"mode":"powerdown"}}

1608137487.696671: debug: process_event: called
1608137487.696754: debug: processing command
1608137487.700017: debug: sending data, count: 85

```

RussianNeuroMancer, 2020-12-16:
@yousong what you think, maybe it's make sense to integrate similar [patch](https://gitlab.alpinelinux.org/alpine/aports/commit/76b81b486480fd9c3294cd420bcf2df01c27790d) into qemu-ga package in OpenWRT for both of 19.07 and master branches?

yousong, 2020-12-17:
Hi, @RussianNeuroMancer , I just picked the patch from alpine and imported it to master and openwrt-19.07 branch.  Only x86_64 was tested though, please see if it works for you on armvirt machines.  Thank you!

RussianNeuroMancer, 2020-12-17:
@yousong thanks! I will check new package as soon as it will be published in [repository](https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/).

RussianNeuroMancer, 2021-03-16:
@yousong 

Hello!

I can not find qemu-ga package here https://downloads.openwrt.org/releases/packages-21.02/aarch64_generic/ for some reason. Is it gone or I been missing something?

closed 2024-09-19 by RussianNeuroMancer