openwrt/packages #14249: vpn-policy-routing: remote domain with "-" in name is wrongly intepreted when ipset command is used for for remote policies

valkodev, 2020-12-17:
# Issue template (remove lines from top till here)

Maintainer: @stangri Stan Grishin (stangri@melmac.net)
Environment: (put here arch, model, OpenWrt version)
ASUS RT-AC57U
MediaTek MT7621 ver:1 eco:3
OpenWrt 19.07.3 r11063-85e04e9f46
 
Description:
I had been using policy routing successfully with DNSMASQ ipset function. As I started to use pihole and want to use DNS server outside openwrt, those policies stopped to work (as DNS request from client is not coming to openwrt). I changed the ipset option for remote policies to "use ipset command" instead. This was not working either and my investigation showed the issue with domain "3-e.cz". Instead adding this domain's IP address to the table, it is adding just e.cz domain. See below extract from system log showing wrong DNS query is send to get domain IP address.

````
daemon.err modprobe: xt_set is already loaded
daemon.err modprobe: ip_set is already loaded
daemon.err modprobe: ip_set_hash_ip is already loaded
user.notice vpn-policy-routing [13158]: Creating table 'wan/eth0.2/my public IP' [✓]
user.notice vpn-policy-routing [13158]: Creating table 'wgclient/My VPN tunnel GW' [✓]
daemon.info dnsmasq[13141]: 6 127.0.0.1/38952 query[A] e.cz from 127.0.0.1
daemon.info dnsmasq[13141]: 6 127.0.0.1/38952 forwarded e.cz to "My pihole DNS server"
daemon.info dnsmasq[13141]: 6 127.0.0.1/38952 reply e.cz is 62.109.151.58
user.notice vpn-policy-routing [13158]: Routing '3e.cz' via wgclient [✓]
````
doing the same for subdomain www.3-e.cz has another issues, trying to add domain www.3, as well as www.3-e.cz, see log:

````
user.notice vpn-policy-routing [15260]: Creating table 'wan/eth0.2/my public IP' [✓]
user.notice vpn-policy-routing [15260]: Creating table 'wgclient/My VPN tunnel GW' [✓]
daemon.info dnsmasq[15243]: 2 127.0.0.1/42068 query[A] www.3 from 127.0.0.1
daemon.info dnsmasq[15243]: 2 127.0.0.1/42068 forwarded www.3 to "My pihole DNS server"
daemon.info dnsmasq[15243]: 2 127.0.0.1/42068 reply www.3 is NXDOMAIN
daemon.info dnsmasq[15243]: 3 127.0.0.1/45215 query[A] www.3-e.cz from 127.0.0.1
daemon.info dnsmasq[15243]: 3 127.0.0.1/45215 forwarded www.3-e.cz to "My pihole DNS server"
daemon.info dnsmasq[15243]: 3 127.0.0.1/45215 ipset add wgclient 37.46.83.207 www.3-e.cz
daemon.info dnsmasq[15243]: 3 127.0.0.1/45215 reply www.3-e.cz is 37.46.83.207
daemon.info dnsmasq[15243]: 4 127.0.0.1/45215 query[AAAA] www.3-e.cz from 127.0.0.1
daemon.info dnsmasq[15243]: 4 127.0.0.1/45215 forwarded www.3-e.cz to "My pihole DNS server"
user.notice vpn-policy-routing [15260]: Routing '3e.cz' via wgclient [✓]
````

Thanks in advance for looking into this.

Radek

stangri, 2020-12-17:
Please check the [README](https://docs.openwrt.melmac.net/vpn-policy-routing/) for the information required for troubleshooting.

valkodev, 2020-12-17:
/etc/init.d/vpn-policy-routing reload
````
Creating table 'wan/eth0.2/my wan IP' [✓]
Creating table 'openvpnc/ovpnc0/my openvpvn gw ip' [✓]
Creating table 'wgclient/my wgclient gw ip' [✓]
Routing '3e.cz' via wgclient [✓]
vpn-policy-routing 0.2.1-13 started with gateways:
wan/eth0.2/my wan IP [✓]
openvpnc/ovpnc0/my openvpvn gw ip
wgclient/my wgclient gw ip
vpn-policy-routing 0.2.1-13 monitoring interfaces: wan openvpnc wgclient .
````

running:  /etc/init.d/vpn-policy-routing support -p
give me a error:
````
Pasting to paste.ee... [✗]
ERROR: curl, libopenssl or ca-bundle were not found!
Run 'opkg update; opkg install curl libopenssl ca-bundle' to install them.
````

and running opkg install curl libopenssl ca-bundle
prove I have them installed:

````
Package curl (7.66.0-3) installed in root is up to date.
Package libopenssl1.1 (1.1.1h-1) installed in root is up to date.
Package ca-bundle (20200601-1) installed in root is up to date.
````

Is there other way to share the support output not publicly?

Can you submit the issue for ipset or you want me to do? Not sure what commands you are running to create them. But I can see the table is messed up with many wrong IP addresses:

````
add wgclient 56.0.0.0/6 comment "3e.cz: 3-e.cz"
add wgclient 1.0.0.0/8 comment "3e.cz: 3-e.cz"
add wgclient 0.4.0.0/14 comment "3e.cz: 3-e.cz"
add wgclient 60.0.0.0/7 comment "3e.cz: 3-e.cz"
add wgclient 0.1.0.0/16 comment "3e.cz: 3-e.cz"
add wgclient 62.109.151.48/29 comment "3e.cz: 3-e.cz"
add wgclient 0.0.64.0/18 comment "3e.cz: 3-e.cz"
add wgclient 32.0.0.0/4 comment "3e.cz: 3-e.cz"
add wgclient 62.109.151.32/28 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.8/29 comment "3e.cz: 3-e.cz"
add wgclient 0.0.8.0/21 comment "3e.cz: 3-e.cz"
add wgclient 0.16.0.0/12 comment "3e.cz: 3-e.cz"
add wgclient 62.109.0.0/17 comment "3e.cz: 3-e.cz"
add wgclient 0.32.0.0/11 comment "3e.cz: 3-e.cz"
add wgclient 0.0.32.0/19 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.64/26 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.16/28 comment "3e.cz: 3-e.cz"
add wgclient 62.104.0.0/14 comment "3e.cz: 3-e.cz"
add wgclient 0.0.128.0/17 comment "3e.cz: 3-e.cz"
add wgclient 62.109.144.0/22 comment "3e.cz: 3-e.cz"
add wgclient 0.0.16.0/20 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.32/27 comment "3e.cz: 3-e.cz"
add wgclient 62.109.151.0/27 comment "3e.cz: 3-e.cz"
add wgclient 0.0.1.0/24 comment "3e.cz: 3-e.cz"
add wgclient 4.0.0.0/6 comment "3e.cz: 3-e.cz"
add wgclient 62.109.151.58 comment "3e.cz: 3-e.cz"
add wgclient 48.0.0.0/5 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.3 comment "3e.cz: 3-e.cz"
add wgclient 62.64.0.0/11 comment "3e.cz: 3-e.cz"
add wgclient 62.108.0.0/16 comment "3e.cz: 3-e.cz"
add wgclient 62.109.151.56/31 comment "3e.cz: 3-e.cz"
add wgclient 0.2.0.0/15 comment "3e.cz: 3-e.cz"
add wgclient 62.96.0.0/13 comment "3e.cz: 3-e.cz"
add wgclient 62.0.0.0/10 comment "3e.cz: 3-e.cz"
add wgclient 0.0.4.0/22 comment "3e.cz: 3-e.cz"
add wgclient 0.8.0.0/13 comment "3e.cz: 3-e.cz"
add wgclient 2.0.0.0/7 comment "3e.cz: 3-e.cz"
add wgclient 62.109.128.0/20 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.128/25 comment "3e.cz: 3-e.cz"
add wgclient 0.0.0.4/30 comment "3e.cz: 3-e.cz"
add wgclient 62.109.148.0/23 comment "3e.cz: 3-e.cz"
add wgclient 16.0.0.0/4 comment "3e.cz: 3-e.cz"
add wgclient 62.109.150.0/24 comment "3e.cz: 3-e.cz"
add wgclient 0.64.0.0/10 comment "3e.cz: 3-e.cz"
add wgclient 0.128.0.0/9 comment "3e.cz: 3-e.cz"
add wgclient 0.0.2.0/23 comment "3e.cz: 3-e.cz"
add wgclient 8.0.0.0/5 comment "3e.cz: 3-e.cz"
````

where expected outcome from working domain is similar to below, 1 line:
````
add wgclient 195.47.89.192 comment "DPD: dpdkuryr.cz"
````

stangri, 2020-12-17:
> running: /etc/init.d/vpn-policy-routing support -p
> give me a error:

Oh, thanks for reporting this. It's fixed in 0.3.0-3 available from my repo.

Regarding the `ipset` not processing arguments with dash in them, the 19.07.5 (at least on ramips) comes with ipset 7.3. I justed tested referred domain with the ipset 7.5 on Ubuntu 20.04 and while it doesn't report an error, it does add a large number of clearly incorrect entries to the ipset. 

```sh
$ nslookup 3-e.cz
Non-authoritative answer:
Name:   3-e.cz
Address: 37.46.83.207
$ sudo ipset create test hash:net
$ sudo ipset add test "3-e.cz"
add test 0.0.2.0/23
add test 56.0.0.0/6
add test 0.64.0.0/10
add test 0.16.0.0/12
add test 0.0.0.8/29
add test 8.0.0.0/5
add test 48.0.0.0/5
add test 2.0.0.0/7
add test 62.0.0.0/10
add test 0.32.0.0/11
add test 62.109.151.48/29
add test 0.0.8.0/21
add test 0.0.0.3
add test 62.108.0.0/16
add test 62.109.144.0/22
add test 0.0.4.0/22
add test 62.104.0.0/14
add test 0.0.1.0/24
add test 0.8.0.0/13
add test 1.0.0.0/8
add test 0.0.0.32/27
add test 4.0.0.0/6
add test 62.109.151.32/28
add test 16.0.0.0/4
add test 62.109.0.0/17
add test 0.0.0.16/28
```

Either I'm not using `ipset` correctly, or it's an upstream issue.

valkodev, 2020-12-18:
OK, understood. Thanks for you good and quick support. Appreciated.

stangri, 2021-04-20:
This can be closed. @aparcar @jow- as I don't have merge rights for packages, I can't close this.

closed 2021-04-20 by aparcar

stephankoelle, 2023-01-05:
>PS. I can reproduce on 19.07.5, seems to be an issue with ipset, nothing I can do about it.

https://ipset.netfilter.org/ipset.man.html

If host names or service names with dash in the name are used instead of IP addresses or service numbers, then the host name or service name must be enclosed in square brackets. Example:

stangri, 2023-01-05:
@stephankoelle thanks, I'll fix it in `pbr`/`pbr-iptables`, feel free to send the PR to fix the `vpn-policy-routing` in OpenWrt 21.02 and 19.07 if you want.

KurtKeesee, 2023-09-23:
It's crucial to address this issue in VPN policy routing, as the incorrect interpretation of remote domains with hyphens ("-") in their names when using the IPset command can lead to potential security and connectivity issues. Ensuring accurate parsing of domain names is essential for maintaining a secure and reliable [Windows VPN](https://fastestvpn.com/download/windows-vpn) infrastructure. This bug fix is much needed for a seamless and secure online experience.

stangri, 2023-09-24:
@KurtKeesee it's been fixed in `pbr`/`pbr-iptables` which are available in OpenWrt 22.03. If you insist on using an outdated version of OpenWrt, feel free to check the linked fix in `pbr` and send the PR to fix the `vpn-policy-routing` in OpenWrt 21.02 and 19.07.