openwrt/packages #14258: python3 does not compile with clang

neheb, 2020-12-18:
Maintainer: @commodo @jefferyto 
Environment: master HOSTCC=clang , CC=clang

```
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for python3.9... python3.9
checking for --enable-universalsdk... no
checking for --with-universal-archs... no
checking MACHDEP... "linux"
checking for x86_64-pc-linux-gnu-gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether clang accepts -g... yes
checking for clang option to accept ISO C89... none needed
checking how to run the C preprocessor... clang -E
checking for grep that handles long lines and -e... /home/mangix/devstuff/openwrt/staging_dir/host/bin/grep
checking for a sed that does not truncate output... /home/mangix/devstuff/openwrt/staging_dir/host/bin/sed
checking for --with-cxx-main=<compiler>... no
checking for the platform triplet based on compiler characteristics... x86_64-linux-gnu
checking for -Wl,--no-as-needed... yes
checking for egrep... /home/mangix/devstuff/openwrt/staging_dir/host/bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for the Android API level... not Android
checking for --with-suffix... 
checking for case-insensitive build directory... no
checking LIBRARY... libpython$(VERSION)$(ABIFLAGS).a
checking LINKCC... $(PURIFY) $(MAINCC)
checking EXPORTSYMS... 
checking for GNU ld... yes
checking for --enable-shared... no
checking for --enable-profiling... no
checking LDLIBRARY... libpython$(VERSION)$(ABIFLAGS).a
checking for x86_64-pc-linux-gnu-ar... no
checking for x86_64-pc-linux-gnu-aal... no
checking for ar... ar
checking for x86_64-pc-linux-gnu-readelf... no
checking for readelf... readelf
checking for a BSD-compatible install... /usr/bin/install -c
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for --with-pydebug... no
checking for --with-trace-refs... no
checking for --with-assertions... no
checking for --enable-optimizations... yes
checking PROFILE_TASK... -m test --pgo
checking for --with-lto... no
checking for x86_64-pc-linux-gnu-llvm-profdata... no
checking for llvm-profdata... no
configure: error: llvm-profdata is required for a --enable-optimizations build but could not be found.
```

commodo, 2020-12-21:
I found this on Python's bug list:  https://bugs.python.org/issue36002

It sounds like `llvm-profdata` should be installed on the host system.
Maybe do `apt-get install -y llvm` ? After installing this, I could run  `llvm-profdata` 

I could not reproduce this locally (even without installing `llvm`).
But we could maybe disable the `--enable-optimizations` flag for host-Python.
I tried a build with host-Python without `--enable-optimizations (and with a SOURCE_DATE_EPOCH)` and it seemed to be roughly the same output.

neheb, 2020-12-22:
My solution instead of hacking around it: https://github.com/neheb/packages/commit/ee2531cab406e27441913bb1c5b46ab4ae03a67a

jefferyto, 2020-12-22:
What is the host OS/version?

I would prefer to keep optimizations enabled if possible, even for host - it should provide a not-insignificant improvement when building Python packages (especially for buildbots).

neheb, 2020-12-22:
I've tested on both Fedora 33 and CentOS7.

If you want to replicate, run
```
export CC=clang
export CXX=clang++
```

and edit HOSTCC and HOSTCXX in rules.mk

jefferyto, 2020-12-22:
Will try to replicate this soon :pray:

jefferyto, 2021-02-15:
I have an idea for how to address this, but I have a question first: Why do we need to support clang?

AFAIK [gcc is required](https://github.com/openwrt/openwrt/blob/ac7d45b5e783fdd4ca35fe53a0921d8972f3f53b/include/prereq-build.mk#L27) (except on macOS where gcc is actually clang, but there llvm-profdata is available). The wiki [doesn't have any pages](https://openwrt.org/start?q=clang&do=search) that talk about building OpenWrt with clang.

Is there is a reason why we should spend time fixing something for a configuration that isn't supported?

closed 2021-02-15 by neheb

jefferyto, 2021-02-18:
I wasn't expecting this issue to be closed so suddenly :joy: perhaps my last message was more aggressive than it should have been (my apologies). I was a bit curious if there was a use-case for clang that I missed.

neheb, 2021-02-18:
Probably not.