openwrt/packages #14287: openconnect: Token Script Overwrites First Password in /etc/

FrederickGeek8, 2020-12-21:
Maintainer: @nmav 
Environment: OpenWRT SNAPSHOT r15237-fca0eb2d92

Description: When using a token script for a second password, the original password gets overwritten in `/var/etc/openconnect-$config.passwd`. If this is indeed a bug the fix is simple and below.

Relevant line: https://github.com/openwrt/packages/blob/master/net/openconnect/files/openconnect.sh#L117 should be changed to `$token_script >> "$pwfile" 2> /dev/null || {`. For context:
```bash
[ -n "$password" ] || [ "$token_mode" = "script" ] && {
    umask 077
    mkdir -p /var/etc
    pwfile="/var/etc/openconnect-$config.passwd"
    [ -n "$password" ] && {
        echo "$password" > "$pwfile"
        [ -n "$password2" ] && echo "$password2" >> "$pwfile"
    }
    [ "$token_mode" = "script" ] && {
        $token_script > "$pwfile" 2> /dev/null || {
            logger -t openconenct "Cannot get password from script '$token_script'"
            proto_setup_failed "$config"
        }
    }
    append_args --passwd-on-stdin
}
```
should be changed to
```bash
[ -n "$password" ] || [ "$token_mode" = "script" ] && {
    umask 077
    mkdir -p /var/etc
    pwfile="/var/etc/openconnect-$config.passwd"
    [ -n "$password" ] && {
        echo "$password" > "$pwfile"
        [ -n "$password2" ] && echo "$password2" >> "$pwfile"
    }
    [ "$token_mode" = "script" ] && {
        $token_script >> "$pwfile" 2> /dev/null || {
            logger -t openconenct "Cannot get password from script '$token_script'"
            proto_setup_failed "$config"
        }
    }
    append_args --passwd-on-stdin
}
```

nmav, 2020-12-21:
Thank you. Would you like to send a pull request?

FrederickGeek8, 2020-12-23:
@nmav I don't have the ability to test my fix right now, so I can't truthfully fill out the PR template. 

In my environment, I am currently using a workaround to this which just involves first outputting my primary password in my secondary password generation script.

nmav, 2020-12-28:
Please open a pull request when you can test it.

PR #14400 (2021-01-01): openconnect: Fix secondary password script overwriting primary
Maintainer: @nmav 
Compile tested: OpenWrt 19.07.5 x86_64
Run tested: OpenWrt 19.07.5 x86_64 / tested connecting to VPN with a secondary password

Description:
When specifying a secondary password script, the output should be appended to the temporary password file and shouldn't overwrite it. If you refer to the case where there is a static secondary password, you can see that the secondary password is appended. Without this fix, *only* the secondary password is passed to the `openconnect` session.

This pull request fixes #14287.
closed 2021-01-01 by nmav