openwrt/packages #14319: mwan3: Error: argument "2591999sec" is wrong: "expires" value is invalid

jamesmacwhite, 2020-12-23:
Maintainer: @feckert @aaronjg 
Environment: 19.07.5 mwan3 2.8.15

On starting mwan3, I am getting the following errors output in console:

```
Error: argument "2591998sec" is wrong: "expires" value is invalid

Error: argument "2591998sec" is wrong: "expires" value is invalid

Error: argument "2591998sec" is wrong: "expires" value is invalid

Error: argument "2591994sec" is wrong: "expires" value is invalid

Error: argument "2591994sec" is wrong: "expires" value is invalid

Error: argument "2591994sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid

Error: argument "2591999sec" is wrong: "expires" value is invalid
```

It doesn't appear to be breaking mwan3 as interfaces are up and routing rules are working, however it appears somewhere in the script, it is choking on the argument referenced.

jamesmacwhite, 2020-12-24:
I don't think any recent change in the 2.8.x branch is the cause of this error, it looks like a potential edge case.

aaronjg, 2020-12-24:
This should be resolved in the 2.10.x branch. With those errors, it is likely that some of your IPv6 routes are not making it into the interface specific tables.

We could backport this by taking part of the sed expression from the master branch:
https://github.com/openwrt/packages/blob/5691ff247f24d6784af8fd2f5b7ed17879dbcc08/net/mwan3/files/lib/mwan3/common.sh#L133

and applying it to the routing table parsing in the 2.8.x branch:
https://github.com/openwrt/packages/blob/49459505e7a1c93c81aa58190a5d3d9015892204/net/mwan3/files/lib/mwan3/mwan3.sh#L93
https://github.com/openwrt/packages/blob/49459505e7a1c93c81aa58190a5d3d9015892204/net/mwan3/files/lib/mwan3/mwan3.sh#L107

P/Rs are welcome

jamesmacwhite, 2020-12-24:
No worries, thanks for the explanation. I think this is more of an edge case, so probably not worth backporting.

closed 2020-12-24 by jamesmacwhite