openwrt/packages #14326: nfdump: recompile with -fPIC

BKPepe, 2020-12-24:
Maintainer: @flyn-org 
Environment: Turris Omnia, mvebu - cortex-a9+vfpv3-d16_musl_eabi, OpenWrt master

Description:

```
OpenWrt-libtool: link: ccache_cc -g -O3 -std=gnu11 -Wall -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wmissing-noreturn -fno-strict-aliasing -fpic -specs=/foo/build/include/hardened-ld-pie.specs -znow -zrelro -Wl,--as-needed -o .libs/nfcapd nfcapd-nfcapd.o nfcapd-nfstatfile.o nfcapd-launch.o nfcapd-nfnet.o nfcapd-collector.o nfcapd-netflow_v1.o nfcapd-netflow_v5_v7.o nfcapd-netflow_v9.o nfcapd-ipfix.o nfcapd-bookkeeper.o nfcapd-expire.o  -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/usr/lib -L/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nfdump-1.6.22/bin/.libs/libnfdump.so -lfts -lresolv -lbz2
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-nfcapd.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-nfstatfile.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-launch.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-nfnet.o: relocation R_ARM_MOVW_ABS_NC against `stderr' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-collector.o: relocation R_ARM_MOVW_ABS_NC against `stderr' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-netflow_v1.o: relocation R_ARM_MOVW_ABS_NC against `extension_descriptor' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-netflow_v5_v7.o: relocation R_ARM_MOVW_ABS_NC against `extension_descriptor' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-netflow_v9.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-ipfix.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-bookkeeper.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-expire.o: relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/foo/build/staging_dir/toolchain-arm_cortex-a9+vfpv3-d16_gcc-8.4.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: nfcapd-nfcapd.o(.text+0xc0): unresolvable R_ARM_CALL relocation against symbol `LogInfo'
collect2: error: ld returned 1 exit status
make[6]: *** [Makefile:938: nfcapd] Error 1
make[6]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nfdump-1.6.22/bin'
make[5]: *** [Makefile:795: all] Error 2
make[5]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nfdump-1.6.22/bin'
make[4]: *** [Makefile:411: all-recursive] Error 1
make[4]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nfdump-1.6.22'
make[3]: *** [Makefile:343: all] Error 2
make[3]: Leaving directory '/foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/nfdump-1.6.22'
```

Building it with enabled:
```
CONFIG_PKG_FORTIFY_SOURCE_2=y
CONFIG_PKG_ASLR_PIE_ALL=y
CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
CONFIG_PKG_RELRO_FULL=y
```

flyn-org, 2020-12-24:
I am running a build to investigate this now.

flyn-org, 2020-12-25:
I have tried a number of combinations of adding -fPIC, -fpic, or $(FPIC) to TARGET_CFLAGS and TARGET_LDFLAGS, but I continue to get the errors above. I will continue to try to figure this out, but any time someone might have would be appreciated. I confirmed zsh compiles, as zsh uses $(FPIC).

BKPepe, 2021-06-01:
Maybe @neheb might help.

neheb, 2021-06-04:
It's an issue somewhere in the build system where CFLAGS are not being propagated.