openwrt/packages #14327: snort: while compiling daq library can not be found

BKPepe, 2020-12-24:
Maintainer: @flyn-org 
Environment: Turris Omnia, mvebu - arm_cortex-a9_vfpv3-d16, OpenWrt master

Description:

Using my build server I am not able to compile snort and also, there is the same issue on buildbots. See: https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/snort/compile.txt

Here is following output:

```
checking dumbnet.h usability... no
checking dumbnet.h presence... no
checking for dumbnet.h... no
checking for eth_set in -ldnet... yes
checking for eth_set in -ldumbnet... no
checking for dlsym in -ldl... yes
checking for daq_load_modules in -ldaq... no

   ERROR!  daq library not found, go get it from
   http://www.snort.org/.
make[2]: *** [Makefile:162: /foo/build/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/snort/snort-2.9.17/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
```

flyn-org, 2020-12-24:
I have noticed that. The odd thing is that snort builds find from within the OpenWrt SDK. Do you know what might be different?

BKPepe, 2020-12-24:
I didn't try OpenWrt SDK. Maybe @neheb has some clue?

neheb, 2020-12-24:
The issue is the two libdaq. The presence of the second makes it fail. Not too sure how to fix.

neheb, 2020-12-26:
This is why https://github.com/openwrt/packages/blob/master/libs/libdaq/Makefile#L51
https://github.com/openwrt/packages/blob/master/libs/libdaq3/Makefile#L51

Both of these packages provide libdaq.so . snort is not compatible with version 3.

dl12345, 2021-03-12:
Change the Makefiles of both packages to use the configure option `--with-daq-libraries=DIR` to point to something like `/usr/lib/snort-<version>`

flyn-org, 2021-03-12:
This is partially addressed by https://github.com/openwrt/packages/pull/14891. I have to clean a few related things in the snort package, including fixing https://github.com/openwrt/packages/pull/14616.

dl12345, 2021-03-12:
Yeah I saw #14891 after I posted this comment and tried to post another reply but could not. Github is doing odd things to me today...

flyn-org, 2021-12-14:
Is this fixed? @BKPepe, would you please confirm and close?

BKPepe, 2021-12-14:
Yes, it is. Thank you, @flyn-org

closed 2021-12-14 by BKPepe