openwrt/packages #14332: mwan3: 2.10.x branch and IPv6 inconsistencies

jamesmacwhite, 2020-12-24:
Maintainer: @aaronjg @feckert 
Environment: OpenWrt 19.07.5 mwan3 2.10.5

Description:

Under the 2.10.x branch my IPv6 setup seems to be experiencing IPv6 issues. One symptom I can see is pings to IPv6 addresses i.e. `ipv6.google.com` from clients are timing out 50% of the time. All interfaces are reporting an up status within mwan3

This is a sample IPv6 ping from a Windows client:

```
Pinging ipv6.l.google.com [2404:6800:4005:808::200e] with 32 bytes of data:
Reply from 2404:6800:4005:808::200e: time=282ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=277ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=278ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=279ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=282ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=279ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=282ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=283ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=282ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=287ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=278ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=284ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=281ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=313ms
Request timed out.
Reply from 2404:6800:4005:808::200e: time=299ms
Request timed out.

Ping statistics for 2404:6800:4005:808::200e:
    Packets: Sent = 30, Received = 15, Lost = 15 (50% loss),
Approximate round trip times in milli-seconds:
    Minimum = 277ms, Maximum = 313ms, Average = 284ms
```

Doing the same from the router itself with mwan3 enable, no ping loss.

```
root@linksys-wrt3200acm:~# ping ipv6.google.com
PING ipv6.google.com (2404:6800:4005:808::200e): 56 data bytes
64 bytes from 2404:6800:4005:808::200e: seq=0 ttl=113 time=279.628 ms
64 bytes from 2404:6800:4005:808::200e: seq=1 ttl=113 time=282.981 ms
64 bytes from 2404:6800:4005:808::200e: seq=2 ttl=113 time=274.847 ms
64 bytes from 2404:6800:4005:808::200e: seq=3 ttl=113 time=278.284 ms
64 bytes from 2404:6800:4005:808::200e: seq=4 ttl=113 time=275.490 ms
64 bytes from 2404:6800:4005:808::200e: seq=5 ttl=113 time=290.525 ms
64 bytes from 2404:6800:4005:808::200e: seq=6 ttl=113 time=275.543 ms
64 bytes from 2404:6800:4005:808::200e: seq=7 ttl=113 time=275.152 ms
64 bytes from 2404:6800:4005:808::200e: seq=8 ttl=113 time=279.917 ms
64 bytes from 2404:6800:4005:808::200e: seq=9 ttl=113 time=275.959 ms
64 bytes from 2404:6800:4005:808::200e: seq=10 ttl=113 time=277.035 ms
64 bytes from 2404:6800:4005:808::200e: seq=11 ttl=113 time=276.082 ms
64 bytes from 2404:6800:4005:808::200e: seq=12 ttl=113 time=276.150 ms
64 bytes from 2404:6800:4005:808::200e: seq=13 ttl=113 time=275.293 ms
64 bytes from 2404:6800:4005:808::200e: seq=14 ttl=113 time=301.525 ms
64 bytes from 2404:6800:4005:808::200e: seq=15 ttl=113 time=302.579 ms
64 bytes from 2404:6800:4005:808::200e: seq=16 ttl=113 time=274.776 ms
64 bytes from 2404:6800:4005:808::200e: seq=17 ttl=113 time=284.501 ms
64 bytes from 2404:6800:4005:808::200e: seq=18 ttl=113 time=303.208 ms
64 bytes from 2404:6800:4005:808::200e: seq=19 ttl=113 time=284.229 ms
64 bytes from 2404:6800:4005:808::200e: seq=20 ttl=113 time=287.108 ms
64 bytes from 2404:6800:4005:808::200e: seq=21 ttl=113 time=277.517 ms
64 bytes from 2404:6800:4005:808::200e: seq=22 ttl=113 time=285.347 ms
64 bytes from 2404:6800:4005:808::200e: seq=23 ttl=113 time=276.671 ms
64 bytes from 2404:6800:4005:808::200e: seq=24 ttl=113 time=277.192 ms
^C
--- ipv6.google.com ping statistics ---
25 packets transmitted, 25 packets received, 0% packet loss
round-trip min/avg/max = 274.776/281.901/303.208 ms
```

The track_ips for the ping test seems to be OK as well:

[image]

However it appears LAN clients seem to have a 50% loss on IPv6 traffic, so this is causing long load times and something not loading anything over IPv6.

This doesn't happen with the same setup on the 2.8.x branch, so one or more changes between the branches seems to be having a negative impact with my IPv6 configuration.

Let me know what further information I can provide to try and debug the issue.

aaronjg, 2020-12-25:
Thanks for the diagnostics. The pings to ipv6.google.com seem really slow, especially compared to the tracking IPs. Can you try pinging the tracking IPs the same way you are pinging ipv6.google.com and see if they have the same latency?

Have you set this up as failover or load balancing? If load balancing, can you change to failover and see if the issue persists. If failover, can you try changing the policy so there is only one ipv6 member? 

Also, with the 2.8.x version, can you use tcpdump on the router to make sure that it is actually working as expected - that all ping echo ICMP packets are leaving the expected IPV6 interface and none are going out any other interface? 

Is the latency as bad with the 2.8.x version as 2.10?

jamesmacwhite, 2020-12-25:
The latency did seem high but it was run right after an upgrade of the mwan3 package, so the system load might have been higher than average, I think after a few minutes it did start to normalise.

Most of my configuration is failover with 3 physical WAN connections. I do use load balancing on a couple of Wireguard interfaces but they aren't used by default. The primary IPv6 interface was being used to test with the IPv6 pings and is configured with failover WAN -> WANB -> WANC

Latency isn't as bad on 2.8.x, but the latency being high might have been system load related. However, the 50% loss seemed to be consistent, so that was the main thing to flag.

jamesmacwhite, 2020-12-27:
Using tcpdump just targeting ICMP6 traffic. This was the result of the IPv6 ping test to `ipv6.google.com` which `lhr48s20-in-x0e.1e100.net` is a CNAME alias of. I've confirmed it's the same for any IPv6 address, so tracking IP etc. The 50% loss is consistent on clients, it doesn't appear to happen for pings from the router directly though, so maybe that's a starting point.

tcpdump when pinging `ipv6.google.com` from a LAN client.

```
# tcpdump -p icmp6 (filtered to lhr48s20-in-x0e.1e100.net)
10:06:51.604401 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 261, length 40
10:06:51.622531 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 261, length 40
10:06:52.618948 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 262, length 40
10:06:57.607266 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 263, length 40
10:06:57.625225 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 263, length 40
10:06:58.619986 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 264, length 40
10:07:03.594628 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 265, length 40
10:07:03.612588 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 265, length 40
10:07:04.607468 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 266, length 40
10:07:09.596411 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 267, length 40
10:07:09.614100 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 267, length 40
10:07:10.608564 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 268, length 40
10:07:15.597060 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 269, length 40
10:07:15.616658 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 269, length 40
10:07:16.612171 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 270, length 40
10:07:21.596419 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 271, length 40
10:07:21.614464 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 271, length 40
10:07:22.610983 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 272, length 40
10:07:27.606563 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 273, length 40
10:07:27.625479 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 273, length 40
10:07:28.621581 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 274, length 40
10:07:33.597230 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 275, length 40
10:07:33.615464 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 275, length 40
10:07:34.610049 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 276, length 40
10:07:39.592712 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 277, length 40
10:07:39.610250 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 277, length 40
10:07:40.607015 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 278, length 40
10:07:45.589684 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 279, length 40
10:07:45.607893 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 279, length 40
10:07:46.602136 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 280, length 40
10:07:51.591118 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 281, length 40
10:07:51.609105 IP6 lhr48s20-in-x0e.1e100.net > 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx: ICMP6, echo reply, seq 281, length 40
10:07:52.605198 IP6 2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx > lhr48s20-in-x0e.1e100.net: ICMP6, echo request, seq 282, length 40
```

I couldn't see any ICMP6 traffic going over the wrong interface, which is why I just ran tcpdump without `-i`. From the seq values, you can see an ECHO reply is not present after the previous packet with a reply, so it looks like 50% of the time the ICMP6 packets are disappearing into a blackhole or not being routed correctly.

The latency seen previously seemed to be a temporary issue and not related, as it was normal when testing it again, the 50% loss is the big issue. No idea why it behaves the way on 2.10.x.

aaronjg, 2020-12-27:
Just want to check - that tcpdump is from the 2.10.x router? I was actually curious if you could check the tcpdump on the outgoing interface on the 2.8.x interface. I had noticed some situations where the IPv6 packets could go out of the wrong interface on 2.8.x, and made some changes in the 2.10.x release to fix this. So I'm trying to determine if this is a regression in 2.10.x, or if it is just broken in a slightly different way.

Regarding the tcpdump you sared. I'm guessing this defaulted to the `br-lan` interface without the `-i` present. And the '2001:470:xxxx:x:xxxx:xxxx:xxxx:xxxx' is the client IPv6 address, not the NAT'd source IP address.

Does the same pattern appear when you look at the outgoing WAN interface? I suspect either the even echo requests are not making it out, or the even echo replies are coming in the WAN, but are not making it back to the client.

Also, you can actually do the filtering while capturing using the argument `'host <ipv6adr>`)

jamesmacwhite, 2020-12-27:
Yes that's correct, the tcpdump was from 2.10.x and you are right it will likely be the client IPv6 address from `br-lan`.

The same pattern does happen when looking at the specific WAN interface, however for the WAN interface tcpdump you can see a gap in the sequence value, so if the `seq` value started at 40, you'd next see 42, 44, 46, 48 etc for subsequent pings, there is always a missing gap, comparing that to `br-lan`, you seem to see the echo request but no echo reply. On the WAN you don't see either, it's missing entirely by the looks of it.

Perhaps it is a regression, if something in 2.8.x is relied on for my setup, unless something is broken with my configuration, that has gone undetected.

aaronjg, 2020-12-27:
I see. So every other ping request seems to not make it out of the interface. The thing I am wondering about with 2.8.x is if it was really doing the failover properly, or if it was somehow sending things out of the incorrect interface, and it just appeared to be working.

I also wonder with 2.10.x if the packets are leaving through another interface or are being swallowed by the router. You said that you didn't see any icmp6 traffic over the wrong interface. Did you check each one with with tcpdump -i?  Are they being correctly NAT'd so they all have the source IP of the router? If you haven't done that, could you run that diagnostic and report back?

If it does appear that the packets are being swallowed by the router, we need to figure what is going on with the firewall mark. Could you add these rules to the firewall table so we can monitor the marks and then report back?

```
IP=$(resolveip ipv6.google.com)
ip6tables --table mangle  -I FORWARD 1 -d $IP -j LOG --log-prefix "_forward start "
ip6tables --table mangle  -I FORWARD 1 -s $IP -j LOG --log-prefix "_forward start "
ip6tables --table mangle -A FORWARD  -d $IP -j LOG --log-prefix "_forward end "
ip6tables --table mangle -A FORWARD  -s $IP -j LOG --log-prefix "_forward end "
ip6tables --table mangle -I POSTROUTING 1  -d $IP -j LOG --log-prefix "postroute start  "
ip6tables --table mangle -I POSTROUTING  1 -s $IP -j LOG --log-prefix "postroute start  "
ip6tables --table mangle -A POSTROUTING  -d $IP -j LOG --log-prefix "postroute end  "
ip6tables --table mangle -A POSTROUTING  -s $IP -j LOG --log-prefix "postroute end  "
```

jamesmacwhite, 2020-12-27:
I did check every other IPv6 WAN interface when doing a continuous ping from a LAN client and didn't see any IPv6 ping packets for `ipv6.google.com` on them, so they were at least going to the primary IPv6 interface I believe, just every other packet dropped. The NAT6 address looked correct from what I could see on the primary WAN.

I tried removing all members except a single WAN and WAN6 member on the default rules which the ICMP rule would hit and the 50% loss issue still happens.

I can add those rules to my custom firewall. How would you like me to report back the data from those added rules when on 2.10.x and pinging ipv6.google.com?

Might take a bit of time when no one's using the network, I essentially have to keep upgrading and downgrading the mwan3 package with opkg, because I can't keep running 2.10.x for too long given the IPv6 issue I have.

aaronjg, 2020-12-27:
> Might take a bit of time when no one's using the network, I essentially have to keep upgrading and downgrading the mwan3 package with opkg, because I can't keep running 2.10.x for too long given the IPv6 issue I have.

Fair enough. I'll see if I can reproduce on my end with the details you provided.  Your IPV6 setup is still identical to what you have in your medium post? While you are on the 2.8.x version, could you check that the packets are actually all leaving out of the correct interface though?

I suspect there is something weird going on with the NAT66 and mwan3 that is causing this behavior, but I need to investigate more to figure exactly what the issue is.

> I can add those rules to my custom firewall. How would you like me to report back the data from those added rules when on 2.10.x and pinging ipv6.google.com?

Once you add the rules and ping, iptables will make an entry in syslog, so you can just read them out with logread and report back. 

If packets are going out of the wrong interface in 2.8.x, we can likely debug this on 2.8.x and make the fix on both versions.

jamesmacwhite, 2020-12-27:
> Fair enough. I'll see if I can reproduce on my end with the details you provided. Your IPV6 setup is still identical to what you have in your medium post? While you are on the 2.8.x version, could you check that the packets are actually all leaving out of the correct interface though?

They seem to be. Doing a ping from a LAN client and running tcpdump on the primary interface I see the echo request and replies in sequence without any gaps, so not going across any other interface other than the primary.

This is probably better for the most up to date setup I am using: https://gist.github.com/jamesmacwhite/58757c67cf6566c3d6cff46ece2fea32

> Once you add the rules and ping, iptables will make an entry in syslog, so you can just read them out with logread and report back.

I'm not seeing any log entries, doing `ip6tables -vnL -t mangle` I can see the rules in the FORWARD and POSTROUTING chain aren't matching as the pkts value is zero.

aaronjg, 2020-12-28:
> This is probably better for the most up to date setup I am using

> Most of my configuration is failover with 3 physical WAN connections

Sorry, a little confused. That link looks like the IPv6 interfaces are virtual interfaces for the first two for ipv6. For wan_wanb_wanc (aaisp6, henet). Am I reading that correctly? 

A bit surprised that those rules are not being hit. The `-I POSTROUTING 1` rule should go in at the top of the postrouting chain, so should be evaluated for all the traffic leaving the router.

https://i.stack.imgur.com/68Cvx.png

Maybe there is some round robin DNS and the IP addresses aren't matching?

jamesmacwhite, 2020-12-28:
Yeah, you are absolutely right. It is likely due to ipv6.google.com having high availability, I'll use one of the specific IP addresses directly from a DNS lookup to avoid the DNS round robin issue and lock that in the firewall rules. That's worked I can now see the logging happening.

`2607:f8b0:4002:c00::66` is ipv6.google.com in the log output.

What could be an issue is I noticed that briefly, the SRC IP of the Windows client was reporting being from the HE.net prefix which is not the primary, but going out from the L2TP interface which is wrong:

`2001:0470:xxxx` - HE.net /48 prefix which should be `6in4-henet`. However we see `l2tp-aaisp` and the IN value blank. So that looks suspect.

At the moment my DHCPv6 server is serving both /48 prefixes across the LAN, I wonder if Windows selecting the source address is also related?

```
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.287021] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3643 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.311569] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3643 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.335944] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3643 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.356123] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3643 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.398537] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3644 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.423082] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3644 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.447451] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3644 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.467627] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3644 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.508933] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3645 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.533473] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3645 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.557841] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3645 MARK=0x800
Mon Dec 28 08:36:55 2020 kern.warn kernel: [1629750.578017] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=1 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3645 MARK=0x800
Mon Dec 28 08:36:56 2020 kern.warn kernel: [1629751.528333] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=2 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3646 MARK=0x800
Mon Dec 28 08:36:56 2020 kern.warn kernel: [1629751.552884] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=2 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3646 MARK=0x800
Mon Dec 28 08:36:56 2020 kern.warn kernel: [1629751.577269] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:e52c:8236:47fe:f0a1 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=2 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=3646 MARK=0x800
[... cut ...]
Mon Dec 28 08:49:40 2020 kern.warn kernel: [1630515.702148] postroute start  IN= OUT=br-lan SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:e52c:8236:47fe:f0a1 LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=4297 MARK=0x3f00
Mon Dec 28 08:49:40 2020 kern.warn kernel: [1630515.722151] postroute end  IN= OUT=br-lan SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:e52c:8236:47fe:f0a1 LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=4297 MARK=0x3f00
```

aaronjg, 2020-12-29:
> What could be an issue is I noticed that briefly, the SRC IP of the Windows client was reporting being from the HE.net prefix which is not the primary, but going out from the L2TP interface which is wrong:

Thanks for the diagnostics. Yes, this is the issue, but it's not an issue with your setup. It shouldn't matter what the client prefix is, since the router will just be masquerading it anyway.

It was, however, due to a regression that was introduced in the 2.10.x factor. Sorry for the inconvenience, and thank you for the testing and locating the bug!

This branch should fix the issue for you. https://github.com/aaronjg/openwrt-packages/tree/bugfix/mwan3-ipv6-regression

jamesmacwhite, 2020-12-29:
Thank you! I will compile the mwan3 package from your branch to test it out and report back!

jamesmacwhite, 2020-12-29:
I'm afraid I'm still seeing the same 50% loss issue under your branch with the same problem. This is from 2.10.x from your branch, the outgoing interface seems to be jumping around

```
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.008960] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7358 MARK=0x800
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.033591] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7358 MARK=0x800
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.058054] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7358 MARK=0x800
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.078317] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7358 MARK=0x800
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.205764] _forward start IN=l2tp-aaisp OUT=br-lan MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7358 MARK=0x3f00
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.226908] _forward end IN=l2tp-aaisp OUT=br-lan MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7358 MARK=0x3f00
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.247883] postroute start  IN= OUT=br-lan SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7358 MARK=0x3f00
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.267889] postroute end  IN= OUT=br-lan SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7358 MARK=0x3f00
Tue Dec 29 07:53:57 2020 kern.warn kernel: [1713601.025831] _forward start IN=br-lan OUT=6in4-henet MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7359 MARK=0x3f00
Tue Dec 29 07:53:57 2020 kern.warn kernel: [1713601.050550] _forward end IN=br-lan OUT=6in4-henet MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7359 MARK=0x3f00
Tue Dec 29 07:53:57 2020 kern.warn kernel: [1713601.075098] postroute start  IN= OUT=6in4-henet SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7359 MARK=0x3f00
Tue Dec 29 07:53:57 2020 kern.warn kernel: [1713601.095448] postroute end  IN= OUT=6in4-henet SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7359 MARK=0x3f00
Tue Dec 29 07:54:02 2020 kern.warn kernel: [1713605.913505] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7360 MARK=0x800
Tue Dec 29 07:54:02 2020 kern.warn kernel: [1713605.938138] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7360 MARK=0x800
Tue Dec 29 07:54:02 2020 kern.warn kernel: [1713605.962599] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7360 MARK=0x800
Tue Dec 29 07:54:02 2020 kern.warn kernel: [1713605.982861] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7360 MARK=0x800
Tue Dec 29 07:54:02 2020 kern.warn kernel: [1713606.109549] _forward start IN=l2tp-aaisp OUT=br-lan MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7360 MARK=0x3f00
Tue Dec 29 07:54:02 2020 kern.warn kernel: [1713606.130688] _forward end IN=l2tp-aaisp OUT=br-lan MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a LEN=80 TC=0 HOPLIMIT=107 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7360 MARK=0x3f00
[... cut ...]
Tue Dec 29 07:54:33 2020 kern.warn kernel: [1713636.980466] postroute start  IN= OUT=6in4-henet SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7371 MARK=0x3f00
Tue Dec 29 07:54:33 2020 kern.warn kernel: [1713637.000820] postroute end  IN= OUT=6in4-henet SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7371 MARK=0x3f00
```

aaronjg, 2020-12-29:
Thanks. No need to recompile for this since it is just script updates and no changes to the compiled library.

There must have been multiple issues then. This log at least more clearly shows the issue. For example the two packets below have identical IN interface, SRC address, and DST address, yet they are being marked differently by iptables. 

```
Tue Dec 29 07:53:56 2020 kern.warn kernel: [1713600.008960] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7358 MARK=0x800
...
Tue Dec 29 07:53:57 2020 kern.warn kernel: [1713601.025831] _forward start IN=br-lan OUT=6in4-henet MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=127 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7359 MARK=0x3f00
```

You had mentioned that the Windows machine was sending packets with different src ips, but I'm not seeing this in the log you shared. Was this in a different log? It does seem like that could be related to the issue...

I'm going to see if I can reproduce your set up to recreate this issue. Seems like we are getting closer at least. Thanks for the help with the logs.

jamesmacwhite, 2020-12-29:
I just compiled instead of a copying the files in case I missed them in the paths just to be sure.

The Windows src address theory was because I have multiple /48 prefixes handed out, essentially the Windows machine will have one or more IPv6 addresses configured:

* A ULA address from the defined ULA prefix
* A global IPv6 address from `l2tp-aaisp` from the /48
* A global IPv6 address from `6in4-henet` from the /48

The ULA won't be a problem, because by default it will go for a global IPv6 address. For NAT6 to work technically you need a global prefix to be announced otherwise clients would prefer IPv4 anyway.

What I did wonder is, under 2.8.x it seems that it was using the HE.net prefix address, but going out over the `l2tp-aaisp` interface, which that prefix doesn't belong to. For example, right now if do a traceroute to an IPv6 address from a Windows client.

```
C:\Users\james>tracert -6 2607:f8b0:4002:c00::66

Tracing route to 2607:f8b0:4002:c00::66 over a maximum of 30 hops

  1     2 ms     3 ms     2 ms  linksys-wrt3200acm.internal.jmwhite.co.uk [2001:470:6839::1]
  2   118 ms   109 ms   113 ms  careless.aa.net.uk [2001:8b0:0:53::5a9b:3509]
  3   110 ms   111 ms   114 ms  o.aimless.tch.aa.net.uk [2001:8b0:0:53::105]
  4   109 ms   110 ms   109 ms  2001:4860:1:1::1592
  5   109 ms     *        *     2001:4860:0:1101::f
  6   207 ms   199 ms   198 ms  2607:f8b0:e000:8000::6
  7   201 ms   204 ms   199 ms  2001:4860::c:4001:9faf
  8   199 ms   198 ms   200 ms  2001:4860::c:4001:5890
  9   200 ms   207 ms   200 ms  2001:4860::c:4001:557b
 10   198 ms   199 ms   200 ms  2001:4860::cc:4002:d6
 11     *        *        *     Request timed out.
 12     *        *        *     Request timed out.
 13     *        *        *     Request timed out.
 14     *        *        *     Request timed out.
 15     *        *        *     Request timed out.
```

We can see the traffic is clearly traversing the AAISP network, but the iptables logging shows that the SRC address was the 2001:470:6839::/48 prefix, which is HE.net now AAISP. So I wondered if this had something to do with. Technically with NAT6 it shouldn't, because the outgoing IPv6 shows as AAISP, but it's something else to consider.

```
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.046993] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7637 MARK=0x800
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.071536] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7637 MARK=0x800
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.095914] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7637 MARK=0x800
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.116089] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7637 MARK=0x800
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.250896] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7638 MARK=0x800
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.275441] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7638 MARK=0x800
Tue Dec 29 14:29:57 2020 kern.warn kernel: [1737369.299819] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7638 MARK=0x800
Tue Dec 29 14:29:58 2020 kern.warn kernel: [1737369.319996] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7638 MARK=0x800
Tue Dec 29 14:29:58 2020 kern.warn kernel: [1737369.451458] _forward start IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7639 MARK=0x800
Tue Dec 29 14:29:58 2020 kern.warn kernel: [1737369.476001] _forward end IN=br-lan OUT=l2tp-aaisp MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7639 MARK=0x800
Tue Dec 29 14:29:58 2020 kern.warn kernel: [1737369.500369] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=9 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=763 [... cut ...]
Tue Dec 29 14:30:08 2020 kern.warn kernel: [1737379.726642] postroute start  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=10 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7642 MARK=0x800
Tue Dec 29 14:30:08 2020 kern.warn kernel: [1737379.746907] postroute end  IN= OUT=l2tp-aaisp SRC=2001:0470:6839:0000:11bf:4b6e:8ad2:fe6a DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=112 TC=0 HOPLIMIT=10 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7642 MARK=0x800
```

I did notice the mark values, so perhaps this is where the problem really is.

jamesmacwhite, 2020-12-29:
I guess there are a few issues based on the logs and comparing 2.8.x to 2.10.x.

1. There's potentially an issue with the mark values as shown in the 2.10.x log shown here: https://github.com/openwrt/packages/issues/14332#issuecomment-751986335
2. There are multiple outgoing interfaces involved when the failover configuration policy should be `l2tp-aaisp`, then `6in4-hetnet` if `l2tp-aaisp` is down, so there is no reason to see `6in4-henet` at all I don't think, but it does explain why there's a 50% ping loss, when it seems the ICMP6 packets are alternating between two interfaces.
3. Is there potentially problems with multiple prefixes from different WANs and clients such as Windows just using whatever it wants?

jamesmacwhite, 2021-01-01:
@aaronjg Would it help if I went back through the package versions and tested from around 2.9.0 onwards, maybe this can help isolate the specific commit that is perhaps causing the regression?

From memory, I believe early 2.9.x versions worked, then around 2.10.x onwards IPv6 was breaking for me.

aaronjg, 2021-01-01:
The next diagnostic step is to insert iptables logging rules to what is going on that causes the even packets to be marked with 0x3f00.

I was going to set up something to see if I could replicate, but if you can add the logging rules, that would be great!

jamesmacwhite, 2021-01-01:
Happy to assist debugging, let me know what iptables logging rules you want me to run to help with that.

I did build a few old versions of 2.10.x but they seemed to have other issues, so probably not helpful for testing this, I guess we'll have to narrow it down through more logs.

`0x3F00` being the default firewall mask of mwan3, although under 2.8.x the ICMP packets are all `0x800` but for some reason on 2.10.x they seem to be alternating between `0x800` and `0x3F00` which somewhat explains why the 50% loss issue is present, but why they are marked that way, I guess is the main issue to resolve.

Let me know what iptables logging rules would be beneficial and I'll temporarily upgrade the mwan3 package and run a ICMP ping test.

aaronjg, 2021-01-01:
Try these rules - these should track the packet through the mwan3 hook rules. After that, may need to dig into the NAT table or some of the mwan3 chains, depending on what these results show.

```
IP=<IPV6>
iptables --table mangle -A mwan3_hook -d $IP  -j LOG --log-prefix "rule end "
iptables --table mangle -A mwan3_hook -s $IP  -j LOG --log-prefix "rule end "
iptables --table mangle -I mwan3_hook 6 -d $IP  -j LOG --log-prefix "after conmark save "
iptables --table mangle -I mwan3_hook 6 -s $IP  -j LOG --log-prefix "after conmark save "
iptables --table mangle -I mwan3_hook 5 -d $IP  -j LOG --log-prefix "after mwan3_rules "
iptables --table mangle -I mwan3_hook 5 -s $IP  -j LOG --log-prefix "after mwan3_rules "
iptables --table mangle -I mwan3_hook 4 -d $IP  -j LOG --log-prefix "after mwan3_connected "
iptables --table mangle -I mwan3_hook 4 -s $IP  -j LOG --log-prefix "after mwan3_connected "
iptables --table mangle -I mwan3_hook 3 -d $IP  -j LOG --log-prefix "after mwan3_ifaces_in "
iptables --table mangle -I mwan3_hook 3 -s $IP  -j LOG --log-prefix "after mwan3_ifaces_in "
iptables --table mangle -I mwan3_hook 2 -d $IP  -j LOG --log-prefix "after after conmark restore "
iptables --table mangle -I mwan3_hook 2 -s $IP  -j LOG --log-prefix "after after conmark restore "
iptables --table mangle -I mwan3_hook 1 $RULE -d $IP  -j LOG --log-prefix "rule start "
iptables --table mangle -I mwan3_hook 1 $RULE -s $IP  -j LOG --log-prefix "rule start "

#check rules
iptables -t mangle -S mwan3_hook
```

jamesmacwhite, 2021-01-01:
Thanks, just to confirm though, shouldn't this be for `ip6tables`?

aaronjg, 2021-01-01:
> Thanks, just to confirm though, shouldn't this be for `ip6tables`?

Yes.

jamesmacwhite, 2021-01-01:
Thought so, what is the $RULE variable meant to be though?

aaronjg, 2021-01-01:
You can take that part out. It was from a previous debugging script I copied.

jamesmacwhite, 2021-01-01:
OK thanks for clarifying. This is what `ip6tables -t mangle -S mwan3_hook` shows with those custom rules added.

```
root@linksys-wrt3200acm:~# ip6tables -t mangle -S mwan3_hook
-N mwan3_hook
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 133 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 134 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 135 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 136 -j RETURN
-A mwan3_hook -p ipv6-icmp -m icmp6 --icmpv6-type 137 -j RETURN
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
```

aaronjg, 2021-01-01:
Ah shoot. I forgot about the icmp6 rules on the ipv6. I think you need to add 5 to the rule insert locations so they go in the correct place.

aaronjg, 2021-01-01:
Do you have any other rules that are marking the packets?

This line looks suspicious to me. It is a ping return, but at 'rule start' it should be before the first mwan3 rule is applied, but it already has a mark of 0x3f00.

``Fri Jan  1 21:00:36 2021 kern.warn kernel: [2020102.886920] rule start IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7949 MARK=0x3f00`
```

jamesmacwhite, 2021-01-01:
There shouldn't be any rules matching the destination IP being pinged other than the default v6 rule in mwan3. I do have ICMP IPv6 related rules in mwan3 but they would not be targeting this IP range and would be for a different WAN policy.

Here's the same from 2.8.x for a comparision:

```
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.353863] rule start IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.376315] after after conmark restore IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.400246] after mwan3_ifaces_in IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.423655] after mwan3_connected IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.447062] after mwan3_rules IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.470118] after conmark save IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.493289] rule end IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7998 MARK=0x800
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.624788] rule start IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x3f00
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.645053] after after conmark restore IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x3f00
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.666798] after mwan3_ifaces_in IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x3f00
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.688019] after mwan3_connected IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x3f00
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.709241] after mwan3_rules IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x3f00
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.730114] after conmark save IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x3f00
Fri Jan  1 22:19:17 2021 kern.warn kernel: [2024825.751082] rule end IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=7998 MARK=0x800
Fri Jan  1 22:19:18 2021 kern.warn kernel: [2024826.369851] rule start IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7999
Fri Jan  1 22:19:18 2021 kern.warn kernel: [2024826.392305] after after conmark restore IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7999
Fri Jan  1 22:19:18 2021 kern.warn kernel: [2024826.416235] after mwan3_ifaces_in IN=br-lan OUT= MAC=32:23:03:df:2c:80:04:d4:c4:4f:3b:bf:86:dd SRC=2001:0470:6839:0000:84ac:495e:f221:3619 DST=2607:f8b0:4002:0c00:0000:0000:0000:0066 LEN=80 TC=0 HOPLIMIT=128 FLOWLBL=0 PROTO=ICMPv6 TYPE=128 CODE=0 ID=1 SEQ=7999
[... cut ...]
Fri Jan  1 22:19:20 2021 kern.warn kernel: [2024828.773837] after conmark save IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=8001 MARK=0x3f00
Fri Jan  1 22:19:20 2021 kern.warn kernel: [2024828.794809] rule end IN=l2tp-aaisp OUT= MAC= SRC=2607:f8b0:4002:0c00:0000:0000:0000:0066 DST=2001:08b0:1111:1111:0000:ffff:51bb:edf2 LEN=80 TC=0 HOPLIMIT=109 FLOWLBL=0 PROTO=ICMPv6 TYPE=129 CODE=0 ID=1 SEQ=8001 MARK=0x800
```

aaronjg, 2021-01-01:
Okay, so there is something going on in your setup with connmark. Could you share your full mangle and nat tables with `ip6tables -t nat -S` and `ip6tables -t mangle -S'.

I think the problem is here:
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L296-L298

If you remove line 297 `-m mark --mark 0x0/$MMX_MASK \` it may fix your issue, but I'd still like to understand how the packets are getting marked in the first place.

jamesmacwhite, 2021-01-01:
Sure this is from 2.8.x, I can do it from 2.10.x if you want as well.

```
root@linksys-wrt3200acm:~# ip6tables -t nat -S
-P PREROUTING ACCEPT
-P INPUT ACCEPT
-P OUTPUT ACCEPT
-P POSTROUTING ACCEPT
-N postrouting_guest_rule
-N postrouting_lan_rule
-N postrouting_rule
-N postrouting_vpn_rule
-N postrouting_wan_rule
-N postrouting_wireguard_rule
-N prerouting_guest_rule
-N prerouting_lan_rule
-N prerouting_rule
-N prerouting_vpn_rule
-N prerouting_wan_rule
-N prerouting_wireguard_rule
-N zone_guest_postrouting
-N zone_guest_prerouting
-N zone_lan_postrouting
-N zone_lan_prerouting
-N zone_vpn_postrouting
-N zone_vpn_prerouting
-N zone_wan_postrouting
-N zone_wan_prerouting
-N zone_wireguard_postrouting
-N zone_wireguard_prerouting
-A PREROUTING -m comment --comment "!fw3: Custom prerouting rule chain" -j prerouting_rule
-A PREROUTING -i br-lan -m comment --comment "!fw3" -j zone_lan_prerouting
-A PREROUTING -i l2tp-aaisp -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i 6in4-henet -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth0.4 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth1.2 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i pppoe-wanb -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i eth0.3 -m comment --comment "!fw3" -j zone_wan_prerouting
-A PREROUTING -i wlan1-1 -m comment --comment "!fw3" -j zone_guest_prerouting
-A PREROUTING -i tun0 -m comment --comment "!fw3" -j zone_vpn_prerouting
-A PREROUTING -i wg -m comment --comment "!fw3" -j zone_wireguard_prerouting
-A PREROUTING -i wgb -m comment --comment "!fw3" -j zone_wireguard_prerouting
-A POSTROUTING -m comment --comment "!fw3: Custom postrouting rule chain" -j postrouting_rule
-A POSTROUTING -o br-lan -m comment --comment "!fw3" -j zone_lan_postrouting
-A POSTROUTING -o l2tp-aaisp -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o 6in4-henet -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth0.4 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth1.2 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o pppoe-wanb -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o eth0.3 -m comment --comment "!fw3" -j zone_wan_postrouting
-A POSTROUTING -o wlan1-1 -m comment --comment "!fw3" -j zone_guest_postrouting
-A POSTROUTING -o tun0 -m comment --comment "!fw3" -j zone_vpn_postrouting
-A POSTROUTING -o wg -m comment --comment "!fw3" -j zone_wireguard_postrouting
-A POSTROUTING -o wgb -m comment --comment "!fw3" -j zone_wireguard_postrouting
-A zone_guest_postrouting -m comment --comment "!fw3: Custom guest postrouting rule chain" -j postrouting_guest_rule
-A zone_guest_prerouting -m comment --comment "!fw3: Custom guest prerouting rule chain" -j prerouting_guest_rule
-A zone_lan_postrouting -m comment --comment "!fw3: Custom lan postrouting rule chain" -j postrouting_lan_rule
-A zone_lan_prerouting -m comment --comment "!fw3: Custom lan prerouting rule chain" -j prerouting_lan_rule
-A zone_vpn_postrouting -m comment --comment "!fw3: Custom vpn postrouting rule chain" -j postrouting_vpn_rule
-A zone_vpn_prerouting -m comment --comment "!fw3: Custom vpn prerouting rule chain" -j prerouting_vpn_rule
-A zone_wan_postrouting -m comment --comment "!fw3" -j MASQUERADE
-A zone_wan_postrouting -m comment --comment "!fw3: Custom wan postrouting rule chain" -j postrouting_wan_rule
-A zone_wan_prerouting -m comment --comment "!fw3: Custom wan prerouting rule chain" -j prerouting_wan_rule
-A zone_wireguard_postrouting -m comment --comment "!fw3" -j MASQUERADE
-A zone_wireguard_postrouting -m comment --comment "!fw3: Custom wireguard postrouting rule chain" -j postrouting_wireguard_rule
-A zone_wireguard_prerouting -m comment --comment "!fw3: Custom wireguard prerouting rule chain" -j prerouting_wireguard_rule
```

```
root@linksys-wrt3200acm:~# ip6tables -t mangle -S
[... 90 lines cut ...]
-A mwan3_policy_wan_wanb_wanc -m mark --mark 0x0/0x3f00 -m comment --comment "aaisp6 3 3" -j MARK --set-xmark 0x800/0x3f00
-A mwan3_policy_wanb_only -m mark --mark 0x0/0x3f00 -m comment --comment "henet 2 2" -j MARK --set-xmark 0x900/0x3f00
-A mwan3_policy_wanb_wan -m mark --mark 0x0/0x3f00 -m comment --comment "henet 2 2" -j MARK --set-xmark 0x900/0x3f00
-A mwan3_policy_wanc_only -m mark --mark 0x0/0x3f00 -m comment --comment "wanc6 2 2" -j MARK --set-xmark 0x600/0x3f00
-A mwan3_policy_wg_balanced -m mark --mark 0x0/0x3f00 -m statistic --mode random --probability 0.50000000000 -m comment --comment "wgb6 3 6" -j MARK --set-xmark 0xd00/0x3f00
-A mwan3_policy_wg_balanced -m mark --mark 0x0/0x3f00 -m comment --comment "wg6 3 3" -j MARK --set-xmark 0xb00/0x3f00
-A mwan3_policy_wg_only -m mark --mark 0x0/0x3f00 -m comment --comment "wg6 3 3" -j MARK --set-xmark 0xb00/0x3f00
-A mwan3_policy_wgb_only -m mark --mark 0x0/0x3f00 -m comment --comment "wgb6 3 3" -j MARK --set-xmark 0xd00/0x3f00
-A mwan3_rule_https -m mark --mark 0x0/0x3f00 -j mwan3_policy_wan_wanb_wanc
-A mwan3_rule_https -m mark ! --mark 0xfc00/0xfc00 -j SET --del-set mwan3_sticky_https src,src
-A mwan3_rule_https -m mark ! --mark 0xfc00/0xfc00 -j SET --add-set mwan3_sticky_https src,src
-A mwan3_rules -p tcp -m multiport --dports 5201 -m mark --mark 0x0/0x3f00 -j MARK --set-xmark 0x3f00/0x3f00
-A mwan3_rules -d 2a01:4c8:f000:1::1/128 -m mark --mark 0x0/0x3f00 -j mwan3_policy_wanc_only
[... cut ...]
-A mwan3_rules -m set --match-set wanb_set dst -m mark --mark 0x0/0x3f00 -j mwan3_policy_wanb_only
-A mwan3_rules -m set --match-set wanc_set dst -m mark --mark 0x0/0x3f00 -j mwan3_policy_wanc_only
-A mwan3_rules -m set --match-set aaisp_set dst -m mark --mark 0x0/0x3f00 -j mwan3_policy_aaisp_only
-A mwan3_rules -m set --match-set vpn_set dst -m mark --mark 0x0/0x3f00 -j mwan3_policy_wg_balanced
-A mwan3_rules -p tcp -m multiport --dports 443 -m mark --mark 0x0/0x3f00 -j mwan3_rule_https
-A mwan3_rules -m mark --mark 0x0/0x3f00 -j mwan3_policy_wan_wanb_wanc
```

aaronjg, 2021-01-01:
Thanks. I suspect the issue is the change from this:
```
-A mwan3_hook -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
```

to this

```
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
```

However, we can't just revert that line. We now need to be able to mark packets to be ignored by mwan3 with the LD_PRELOAD.

So the question is why are the packets being marked 0x3f00 before they enter the mwan3_hook? It must be something related to the NAT setup. Any idea what could be going on?

jamesmacwhite, 2021-01-01:
Getting closer!

So the NAT6 script I use is here: https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6, though I can't see why that would do anything related to fwmark. `0x3f00` is the firewall mask value of mwan3 though isn't it.

The IPv6 masquerading is partially broken in fw3, so that's why the NAT6 helper script is needed.

aaronjg, 2021-01-02:
The mark is updated after: 
```
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
```

by
```
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
```

It seems that NAT6 is saving and restoring this mark with its connection tracking, where NAT4 does not do this. I'm still not clear why though.

jamesmacwhite, 2021-01-02:
Interesting find. Nothing on the NAT6 side will have changed though so I assume it has been doing this under 2.8.x as well, it's just the change to `-A mwan3_hook -m mark --mark 0x0/0x3f00 -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00` now exposes the issue, as that's the only difference?

Not really sure of any alternatives. There is a simplified NAT6 configuration, compared to using the full helper library, but I'd assume it will do the same as it's still using `kmod-ipt-nat6`. There isn't any other IPv6 configuration method I can use, I can't use NETMAP as I don't have a prefix to map for every WAN interface.

aaronjg, 2021-01-02:
> it's just the change to `-A mwan3_hook -m mark --mark 0x0/0x3f00 -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00` now exposes the issue, as that's the only difference?

I suspect that is the case. Can you confirm by removing line 297 here:
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L296-L298
?

That fix will be fine on 19.07.x, but this will render useless the LD_PRELOAD mwan3track fix with the new kernel in 20.x, so not a great long term solution.

Maybe there is a bug upstream with kmod-ipt-nat6?

jamesmacwhite, 2021-01-02:
I'll see if I can test that theory out soon.

My next question was going to be I wonder if NAT6 behaves this way on the 5.4 kernel. I'm loathed to use a snapshot build for a production setting, but perhaps it's not broken like this on snapshot. It is not the first time something in the kernel has caused issues with mwan3, even though it's not directly related, the fwmark issue was a fun one.

There's one other person on the forums recently who was using 2.10.x with NAT6, they had a different IPv6 problem, so I'm not quite sure why I get the 50% loss issue, I removed all my rules except for the default, it does seem to be specific to NAT6 though.

wackejohn, 2021-01-03:
Hi, all
I'm using mwan3 `2.10.5-1` with IPv6 NAT on latest snapshot (kernel version 5.10) and my firewall package was patched to support IPv6 NAT directlly with the patch below:
```
diff -uNr firewall_orig/defaults.c firewall/defaults.c
--- firewall_orig/defaults.c	2020-07-25 08:54:12.000000000 +0800
+++ firewall/defaults.c	2020-08-08 20:26:45.678198422 +0800
@@ -29,8 +29,8 @@
 	C(ANY, FILTER, CUSTOM_CHAINS, "forwarding_rule"),
 	C(ANY, FILTER, SYN_FLOOD,     "syn_flood"),
 
-	C(V4,  NAT,    CUSTOM_CHAINS, "prerouting_rule"),
-	C(V4,  NAT,    CUSTOM_CHAINS, "postrouting_rule"),
+	C(ANY, NAT,    CUSTOM_CHAINS, "prerouting_rule"),
+	C(ANY, NAT,    CUSTOM_CHAINS, "postrouting_rule"),
 
 	{ }
 };
@@ -251,7 +251,8 @@
 			}
 		}
 
-		if (defs->flow_offloading)
+		// There seemd to be issues with offloading for nat6. Only use it on IPv4 for now.
+		if (defs->flow_offloading && handle->family == FW3_FAMILY_V4)
 		{
 			r = fw3_ipt_rule_new(handle);
 			fw3_ipt_rule_comment(r, "Traffic offloading");
diff -uNr firewall_orig/options.h firewall/options.h
--- firewall_orig/options.h	2020-07-25 08:54:12.000000000 +0800
+++ firewall/options.h	2020-08-08 17:56:11.852760994 +0800
@@ -263,6 +263,8 @@
 
 struct fw3_mark
 {
+	struct list_head list;
+
 	bool set;
 	bool invert;
 	uint32_t mark;
@@ -340,6 +342,9 @@
 	bool masq_allow_invalid;
 	struct list_head masq_src;
 	struct list_head masq_dest;
+    
+	bool masq6;
+
 
 	bool mtu_fix;
 
@@ -420,14 +425,14 @@
 
 	struct list_head proto;
 
-	struct fw3_address ip_src;
+	struct list_head ip_src;
 	struct list_head mac_src;
 	struct fw3_port port_src;
 
-	struct fw3_address ip_dest;
+	struct list_head ip_dest;
 	struct fw3_port port_dest;
 
-	struct fw3_address ip_redir;
+	struct list_head ip_redir;
 	struct fw3_port port_redir;
 
 	struct fw3_limit limit;
@@ -462,13 +467,13 @@
 
 	struct list_head proto;
[... 842 lines cut ...]
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=9 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=10 ttl=118 time=136 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=11 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=12 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=13 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=14 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=15 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=16 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=17 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=18 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=19 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=20 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=21 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=22 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=23 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=24 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=25 ttl=118 time=136 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=26 ttl=118 time=135 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=27 ttl=118 time=136 ms
64 bytes from lax17s44-in-x0e.1e100.net (2607:f8b0:4007:80f::200e): icmp_seq=28 ttl=118 time=136 ms
^C
--- ipv6.google.com ping statistics ---
28 packets transmitted, 28 received, 0% packet loss, time 27006ms
rtt min/avg/max/mdev = 134.869/135.855/136.683/0.569 ms
```

aaronjg, 2021-01-03:
I just followed the instructions from here: https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6 and tested it on a clean install of 19.07.5 and openwrt/openwrt@55e23f2, it worked fine for me on both. No marking on the packets on the "rule start" log line.

```
# uname -r
4.14.209
```

```
$ ping -c4  ipv6.google.com
PING ipv6.google.com(iad23s69-in-x0e.1e100.net (2607:f8b0:4004:815::200e)) 56 data bytes
64 bytes from iad23s69-in-x0e.1e100.net (2607:f8b0:4004:815::200e): icmp_seq=1 ttl=113 time=21.3 ms
64 bytes from iad23s69-in-x0e.1e100.net (2607:f8b0:4004:815::200e): icmp_seq=2 ttl=113 time=18.7 ms
64 bytes from iad23s69-in-x0e.1e100.net (2607:f8b0:4004:815::200e): icmp_seq=3 ttl=113 time=26.8 ms
64 bytes from iad23s69-in-x0e.1e100.net (2607:f8b0:4004:815::200e): icmp_seq=4 ttl=113 time=20.2 ms

--- 2607:f8b0:4004:806::200e ping statistics ---
4 packets transmitted, 4 received, 0% packet loss, time 3000ms
rtt min/avg/max/mdev = 667.986/674.297/681.094/4.641 ms
```

jamesmacwhite, 2021-01-03:
Well that's interesting. What is wrong with my setup then. I dropped all rules except the default. Is it the policy wan_wanb_wanc potentially wrong?

@wackejohn are you saying you also had the problem before adding NAT6 into fw3 directly?

wackejohn, 2021-01-03:
@jamesmacwhite 
> @wackejohn are you saying you also had the problem before adding NAT6 into fw3 directly?

Nope, but the NAT6 script on openwrt wiki was very buggy, mwan3 rules (ipv6) would become invalid when the ipv6 interfaces down and up, I had to reload the firewall manually, then the rules would work again.

jamesmacwhite, 2021-01-03:
@wackejohn Good to know. I think fw3 and NAT6 has been broken for a while, which is why that NAT6 helper script exists. If those patches were merged into master, it would remove the need for it entirely, but it looks like at least two of them have existed since 2015 and haven't been merged, hopefully NAT6 support can be added directly someday.

I'm now at a loss what specifically in my setup is causing the behaviour I see though.

aaronjg, 2021-01-03:
@jamesmacwhite - I'm not sure what is broken. My setup was a bit simpler, rather than three IPv6 interfaces, I just had two and neither were tunnel interfaces, so perhaps that is part of the problem. But I believe @wackejohn has been using tunnel interfaces and he seems to not have an issue.

Is there any way you can simplify your setup further to home in on the problem?

jamesmacwhite, 2021-01-03:
@aaronjg I think I'm going to have to simplify it at some point to dig into it. Given it looks like it's something specific in my configuration now, if you couldn't replicate the issue on a clean 19.07.5 image with NAT6.

If my providers themselves would hurry up and deploy IPv6, it would probably be less complicated, not having to use 6in4, L2TP etc, here's to 2021 being the year of IPv6! (probably not!).

jamesmacwhite, 2021-01-03:
It looks like it's something specific with L2TP. I applied the IPv6 routing table fixes and tested it again. If I disable the L2TP interface, 6in4 is the next policy and it works fine, no issues what so ever. So likely there's an issue with L2TP possibly upstream. I tested the simplified NAT6 script, but I don't think it made a difference as I could still get the 50% loss behaviour with L2TP with the simplified version in use, although I might just stick with that anyway.

aaronjg, 2021-01-03:
Interesting. It wouldn't be the first bug in L2TP that we've hit related to packet marking. I reached out to A&A to ask them to reactivate my developer account so I can continue to debug further and test if this is resolved on the snapshot kernel.

jamesmacwhite, 2021-01-03:
Thank you! That is very much appreciated, it would be interesting to know if the behaviour happens on snapshot, but it certainly isn't the first L2TP issue mwan3 has highlighted before.

jamesmacwhite, 2021-05-20:
@aaronjg Did you happen to get your L2TP tunnel from AAISP reactivated to see if this could be replicated at all? I'm curious to know if the latest kernel in 21.02 has this issue. I can't really test this without breaking my network. 21.02 isn't stable yet but I'd imagine it will be after a few more RC releases. Due to the DSA changes, I will have to reset the interfaces and network config in anyway, but don't really want to run it just yet.

jamesmacwhite, 2022-02-18:
Unfortunately, the same behaviour happens on 21.02. I've finally made the jump and done all the DSA conversion.

My L2TP IPv6 is timing out on a LAN client every other reply, but it is not happening when pinging with the router directly

```
Pinging ipv6.l.google.com [2a00:1450:4009:81d::200e] with 32 bytes of data:
Reply from 2a00:1450:4009:81d::200e: time=21ms
Request timed out.
Reply from 2a00:1450:4009:81d::200e: time=24ms
Request timed out.
Reply from 2a00:1450:4009:81d::200e: time=28ms
Request timed out.
Reply from 2a00:1450:4009:81d::200e: time=21ms
Request timed out.
Reply from 2a00:1450:4009:81d::200e: time=27ms
```

I'm now thinking there is something in 2.10.x compared to 2.8.x that is causing this, but it may be specific to L2TP.

Works fine on 19.07.x with mwan3 2.8.x
Doesn't work on 19.07.x with mwan3 2.10.x
Doesn't work on 21.02.2 with mwan3 2.10.x