openwrt/packages #14339: collectd-mod-dns segfaults after running a while

Strykar, 2020-12-26:
@hnyman / `19.07.4 r11208-ce6496d796 on x86/64
`
The `collectd-mod-dns` plugin runs fine for a while and then segfaults.
```
Sat Dec 26 14:14:48 2020 daemon.err collectd[17396]: dns plugin: pcap_loop exited with status -1.
Sat Dec 26 14:15:13 2020 kern.info kernel: [103791.357292] reader#0[17408]: segfault at 7f4142f1a758 ip 00007f414313519a sp 00007f4142f1a760 error 6 in dns.so[7f4143135000+2000]
Sat Dec 26 14:15:13 2020 daemon.info procd: Instance collectd::instance1 s in a crash loop 6 crashes, 271 seconds since last crash
```
I recall this happening even on `18.x`, I suppose the next step is to take debug logs, but I'm unsure if this is the correct place or if this should be reported upstream?

hnyman, 2020-12-26:
Feel free to debug it with the upstream. I tried a few years ago, but did not succeed much.

The DNS plugin has been problematic all the time, as it runs the a bit complicated data collection cycle using external pcap library.

See my debugging attempts a few years ago in upstream issues:
https://github.com/collectd/collectd/issues/1227

I inserted tons of debugging statements into collectd source, but never got into the root cause of the problem. The usage of pcap gets too complicated in multi-threaded environment.

My advice is to disable the DNS plugin, if it causes problems for you. Sorry.

https://forum.openwrt.org/t/collectd-dns-plugin-crashes/20386/2

Strykar, 2020-12-26:
Thanks hnyman, I have known of this issue for years on x64, good to know its not architecture specific.

After reviewing your attempts https://github.com/collectd/collectd/issues/1227 it's clear this is outside the scope of my attempts to assist with :smile: 
I will keep this open until someone responds to the issue created upstream and then close it.

Bummer, the data was quite useful.. when it did work.

Strykar, 2021-01-14:
Closing.. upstream issue.

closed 2021-01-14 by Strykar