openwrt/packages #14349: seafile-server not compiling under i386

neheb, 2020-12-28:
Maintainer: @jefferyto @commodo 

https://downloads.openwrt.org/snapshots/faillogs/i386_pentium-mmx/packages/seafile-server/compile.txt

neheb, 2020-12-29:
Meh. it's gone now.

closed 2020-12-29 by neheb

tjgonbor, 2020-12-29:
Goodnight.
I have tried several times unsuccessfully to compile the seafile-server package.
If it is within your power to help me with that it would be great.
The original seafile-server package that is in the official OpenWrt repository doesn't work either.
If this case is of interest to you, please write me back to send you all the information you need.
Final error message:

tulio@osboxes:~/openwrt$ make package/seafile-server/compile
 make[1] package/seafile-server/compile
 make[2] -C package/libs/ncurses host-compile
 make[2] -C package/libs/toolchain compile
 make[2] -C package/libs/zlib compile
 make[2] -C package/libs/ncurses compile
 make[2] -C package/libs/readline compile
 make[2] -C feeds/packages/utils/bash compile
 make[2] -C package/firmware/linux-firmware compile
 make[2] -C package/firmware/prism54-firmware compile
 make[2] -C package/kernel/linux compile
 make[2] -C feeds/packages/utils/fuse compile
 make[2] -C feeds/packages/utils/attr compile
 make[2] -C feeds/packages/libs/libffi compile
 make[2] -C feeds/packages/devel/ninja host-compile
 make[2] -C feeds/packages/devel/meson host-compile
 make[2] -C package/libs/gettext-full host-compile
 make[2] -C package/libs/gettext-full compile
 make[2] -C package/libs/libiconv host-compile
 make[2] -C package/libs/gettext compile
 make[2] -C package/libs/libiconv compile
 make[2] -C feeds/packages/libs/glib2 compile
 make[2] -C feeds/packages/libs/expat compile
 make[2] -C feeds/packages/utils/xz compile
 make[2] -C package/libs/openssl compile
 make[2] -C package/utils/bzip2 compile
 make[2] -C feeds/packages/libs/libarchive compile
 make[2] -C feeds/packages/libs/oniguruma compile
 make[2] -C package/libs/libevent2 compile
 make[2] -C feeds/packages/libs/libevhtp compile
    ERROR: package/feeds/packages/libevhtp failed to build.
make -r package/seafile-server/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/tulio/openwrt/include/toplevel.mk:242: package/seafile-server/compile] Error 1
tulio@osboxes:~/openwrt$

commodo, 2020-12-29:
I'm actually running into this libevhtp failure on my local machine.
So, I am investigating this already.
But with the holidays things are slower.

commodo, 2020-12-29:
ok; so this seems to have gone away to me as well;
which means it's a race condition with the build;

this is familiar though, so i dug a bit into it;
it's likely that this is a reboot/related to https://github.com/openwrt/packages/issues/12048 &  https://github.com/openwrt/packages/pull/12007

maybe we just need to add in the seafile-server Makefile this:
```
PKG_BUILD_DEPENDS:=vala/host libevent2 libevhtp
```

jefferyto, 2020-12-29:
IIRC seafile-server does not have a direct dependency on libevent2, and [libevhtp already depends on libevent2](https://github.com/openwrt/packages/blob/eb4a86a1bc113b7c427ff3ce0d911d3a54708608/libs/libevhtp/Makefile#L28). The purpose of #12007 was to not include libevent2 in seafile-server's `PKG_BUILD_DEPENDS`.

I'll try to test this when I have time, but unless there is error log to examine and/or a way to reproduce this issue, I don't want to make random attempts at fixes.

commodo, 2020-12-29:
> IIRC seafile-server does not have a direct dependency on libevent2, and [libevhtp already depends on [...]

it seems random, so you need some luck in getting to the right order of conditions;
but i guess it may be related to libevhtp getting built before libevent2-threads;

commodo, 2020-12-29:
it's back again:
```
touch /home/sandu/work/openwrt/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.fuse_installed
make[3]: Leaving directory '/home/sandu/work/openwrt/openwrt/feeds/packages/utils/fuse'
time: package/feeds/packages/fuse/compile#8.55#1.05#8.81
make[3]: Entering directory '/home/sandu/work/openwrt/openwrt/feeds/packages/libs/libevhtp'
mkdir -p /home/sandu/work/openwrt/openwrt/dl
SHELL= flock /home/sandu/work/openwrt/openwrt/tmp/.libevhtp-1.2.18.tar.gz.flock -c '  	/home/sandu/work/openwrt/openwrt/scripts/download.pl "/home/sandu/work/openwrt/openwrt/dl" "libevhtp-1.2.18.tar.gz" "316ede0d672be3ae6fe489d4ac1c8c53a1db7d4fe05edaff3c7c853933e02795" "" "https://codeload.github.com/criticalstack/libevhtp/tar.gz/1.2.18?"    '
touch /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18/.prepared_d30bdfec24646b04ce8359a97ebec407_6664517399ebbbc92a37c5bb081b5c53_check
. /home/sandu/work/openwrt/openwrt/include/shell.sh; gzip -dc /home/sandu/work/openwrt/openwrt/dl/libevhtp-1.2.18.tar.gz | tar -C /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18

Applying ./patches/010-openssl-thread.patch using plaintext: 
patching file evhtp.c
touch /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18/.prepared_d30bdfec24646b04ce8359a97ebec407_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18/.configured_*
rm -f /home/sandu/work/openwrt/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.libevhtp_installed
mkdir -p /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18
(cd /home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18; CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18=libevhtp-1.2.18 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/sandu/work/openwrt/openwrt/build_dir/target-i386_pentium4_musl/libevhtp-1.2.18=libevhtp-1.2.18 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/usr/lib -L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=i386 -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/bin/i486-openwrt-linux-musl-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/bin/i486-openwrt-linux-musl-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/bin/i486-openwrt-linux-musl-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/usr/lib -L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/usr/lib -L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/usr/lib -L/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/bin/i486-openwrt-linux-musl-gcc-ar" -DCMAKE_NM="/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/bin/i486-openwrt-linux-musl-gcc-nm" -DCMAKE_RANLIB="/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl/bin/i486-openwrt-linux-musl-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/sandu/work/openwrt/openwrt/staging_dir/target-i386_pentium4_musl/usr;/home/sandu/work/openwrt/openwrt/staging_dir/toolchain-i386_pentium4_gcc-8.4.0_musl" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/sandu/work/openwrt/openwrt/staging_dir/target-i386_pentium4_musl -DCMAKE_PREFIX_PATH=/home/sandu/work/openwrt/openwrt/staging_dir/target-i386_pentium4_musl -DCMAKE_SKIP_RPATH=TRUE -DCMAKE_EXPORT_PACKAGE_REGISTRY=FALSE -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=TRUE -DCMAKE_FIND_USE_SYSTEM_PACKAGE_REGISTRY=FALSE -DCMAKE_FIND_PACKAGE_NO_SYSTEM_PACKAGE_REGISTRY=TRUE  /home/sandu/work/ [... cut ...]
time: package/feeds/packages/libevhtp/compile#1.16#0.28#1.30
    ERROR: package/feeds/packages/libevhtp failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/libevhtp/compile] Error 1
make[2]: Leaving directory '/home/sandu/work/openwrt/openwrt'
make[1]: *** [package/Makefile:108: /home/sandu/work/openwrt/openwrt/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/sandu/work/openwrt/openwrt'
make: *** [/home/sandu/work/openwrt/openwrt/include/toplevel.mk:242: world] Error 2

```

commodo, 2020-12-29:
so, libevent2-pthreads isn't selected [which makes sense, nothing selects it]:
```
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libevent2-core=y
# CONFIG_PACKAGE_libevent2-extra is not set
CONFIG_PACKAGE_libevent2-openssl=y
# CONFIG_PACKAGE_libevent2-pthreads is not set
# CONFIG_PACKAGE_libexif is not set
```

selecting this helps;
will try to think of a solution

jefferyto, 2020-12-29:
seafile-server is suppose to [select](https://github.com/openwrt/packages/blob/48c60bf2a61ee067e82cf5bfac79d62c38496215/net/seafile-server/Makefile#L79) `LIBEVHTP_BUILD_DEPENDS`, which should cause `LIBEVHTP_BUILD_SELECT` to [select](https://github.com/openwrt/packages/blob/48c60bf2a61ee067e82cf5bfac79d62c38496215/libs/libevhtp/Makefile#L59) `PACKAGE_libevent2-pthreads`. Not sure what has changed to cause this to not happen.

jefferyto, 2020-12-29:
The `BUILDONLY:=1` added to libevhtp in 5a7148d112113544611ee358a7d062cec85c1629 broke it. Removing that line causes the symbols to be selected again. I also did not see any `package has no install section` warnings as a result.

@neheb Is is safe to revert the change for libevhtp?

neheb, 2020-12-29:
Strange. I'll look into this.