openwrt/packages #14357: PHP file_get_contents(): SSL operation failed

VladimirKalachikhin, 2020-12-28:
Let's start over.

I create OpenWRT-x86 virtual machine as described in https://openwrt.org/docs/guide-user/virtualization/virtualbox-vm
Image to machine - https://downloads.openwrt.org/releases/19.07.5/targets/x86/64/openwrt-19.07.5-x86-64-combined-ext4.img.gz

On the freshly started machine:
```
root@OpenWrt:/# opkg install php7-cli php7-mod-curl
Installing php7-cli (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7-cli_7.2.34-1_x86_64.ipk
Installing libpcre (8.43-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/libpcre_8.43-1_x86_64.ipk
Installing libxml2 (2.9.9-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/libxml2_2.9.9-3_x86_64.ipk
Installing zoneinfo-core (2020c-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/zoneinfo-core_2020c-1_x86_64.ipk
Installing php7 (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7_7.2.34-1_x86_64.ipk
Installing libstdcpp6 (7.5.0-2) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/targets/x86/64/packages/libstdcpp6_7.5.0-2_x86_64.ipk
Installing php7-mod-curl (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7-mod-curl_7.2.34-1_x86_64.ipk
Installing ca-bundle (20200601-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/base/ca-bundle_20200601-1_all.ipk
Installing libcurl4 (7.66.0-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/base/libcurl4_7.66.0-3_x86_64.ipk
Configuring libpcre.
Configuring libxml2.
Configuring zoneinfo-core.
Configuring php7.
Configuring ca-bundle.
Configuring libcurl4.
Configuring php7-mod-curl.
Configuring libstdcpp6.
Configuring php7-cli.
root@OpenWrt:/www# php-cli testSSL.php 
Get by file_get_contents:<br><br>

Warning: file_get_contents(): Unable to find the wrapper "https" - did you forget to enable it when you configured PHP? in /www/testSSL.php on line 8

Warning: file_get_contents(https://b.tile.opentopomap.org/14/8451/5349.png): failed to open stream: No such file or directory in /www/testSSL.php on line 8
http_response_header:<pre>
</pre><br><br>
Get by cURL:<br><br>
http_response_info:
<pre>Array
(
    [url] => https://b.tile.opentopomap.org/14/8451/5349.png
    [content_type] => image/png
    [http_code] => 200
    [header_size] => 279
    [request_size] => 77
    [filetime] => -1
    [ssl_verify_result] => 0
    [redirect_count] => 0
    [total_time] => 0.595267
    [namelookup_time] => 0.138891
    [connect_time] => 0.20573
    [pretransfer_time] => 0.416388
    [size_upload] => 0
    [size_download] => 15625
    [speed_download] => 26260
    [speed_upload] => 0
    [download_content_length] => 15625
    [upload_content_length] => -1
    [starttransfer_time] => 0.527365
    [redirect_time] => 0
    [redirect_url] => 
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.10.102
    [local_port] => 60018
)
</pre>

root@OpenWrt:/www# opkg install php7-mod-openssl
Installing php7-mod-openssl (7.2.34-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/packages/php7-mod-openssl_7.2.34-1_x86_64.ipk
Installing libopenssl1.1 (1.1.1i-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.5/packages/x86_64/base/libopenssl1.1_1.1.1i-1_x86_64.ipk
Configuring libopenssl1.1.
Configuring php7-mod-openssl.
root@OpenWrt:/www# php-cli testSSL.php 
Get by file_get_contents:<br><br>

Warning: file_get_contents(): SSL operation failed with code 1. OpenSSL Error messages:
error:1409442E:SSL routines:ssl3_read_bytes:tlsv1 alert protocol version in /www/testSSL.php on line 8

Warning: file_get_contents(): Failed to enable crypto in /www/testSSL.php on line 8

Warning: file_get_contents(https://b.tile.opentopomap.org/14/8451/5349.png): failed to open stream: operation failed in /www/testSSL.php on line 8
http_response_header:<pre>
</pre><br><br>
Get by cURL:<br><br>
http_response_info:
<pre>Array
(
    [url] => https://b.tile.opentopomap.org/14/8451/5349.png
    [content_type] => image/png
    [http_code] => 200
    [header_size] => 278
    [request_size] => 77
    [filetime] => -1
    [ssl_verify_result] => 0
    [redirect_count] => 0
    [total_time] => 0.39876
    [namelookup_time] => 0.000385
    [connect_time] => 0.070127
    [pretransfer_time] => 0.252536
    [size_upload] => 0
    [size_download] => 15625
    [speed_download] => 39258
    [speed_upload] => 0
    [download_content_length] => 15625
    [upload_content_length] => -1
    [starttransfer_time] => 0.324439
    [redirect_time] => 0
    [redirect_url] => 
    [primary_ip] => 131.188.76.144
    [certinfo] => Array
        (
        )

    [primary_port] => 443
    [local_ip] => 192.168.10.102
    [local_port] => 60026
)
</pre>

```
No other packages were installed.

_Originally posted by @VladimirKalachikhin in https://github.com/openwrt/packages/issues/12085#issuecomment-751490816_

mhei, 2021-01-01:
@VladimirKalachikhin: I think the issue is solved with dropping the patch, can you confirm?

VladimirKalachikhin, 2021-01-02:
No, I didn't succeed.
Please, tell me in detail, what should I do to apply patch?

mhei, 2021-01-02:
Since the buildbots only build master branch but not release branches when no new point release is scheduled, you need to compile yourself. Do you know how to to this?

VladimirKalachikhin, 2021-01-02:
Yes, I know how to compile, but in general. I don't know how to apply patch right.

mhei, 2021-01-02:
No need to apply the patch by hand, just update the package feed to latest branch head, then the dropped patch is automatically removed. You can verify by looking at the generated packages: -2 as package version tells you that you're fine.

VladimirKalachikhin, 2021-01-02:
Ok, I'll try.

VladimirKalachikhin, 2021-01-02:
Heh. How to update `lang` package in v19.07.5 branch by `lang` package from the latest branch? No right way, as I think.

VladimirKalachikhin, 2021-01-02:
Ok, I compiled x86 master branch and run it in VirtualBox.

Yes, bug fixed.

VladimirKalachikhin, 2021-01-02:
But, how to fix my copy of the v19.07.5 branch?
Master branch is buggy...

mhei, 2021-01-03:
Thanks for the feedback. Maybe you can ask for further support in https://forum.openwrt.org/

commit cafbae71 (2020-12-29): php7: drop patch for openssl deprecated API (fixes #14357)
closed 2021-01-03 by mhei