openwrt/packages #14360: strongswan: received netlink error: No such file or directory (2)

cwbsw, 2020-12-29:
Maintainer: @stintel 
Environment: x86, VirtualBox, OpenWrt snapshot

Description:

x86 target is easy to reproduce, some other targets are also affected, such as mt7621 and rockchip, but ipq40xx works well.
```
#opkg install strongswan-default

Tue Dec 27 12:48:55 2020 daemon.info : 05[CFG] selected proposal: ESP:AES_CBC_128/HMAC_SHA1_96/NO_EXT_SEQ
Tue Dec 27 12:48:55 2020 daemon.info : 05[KNL] received netlink error: No such file or directory (2)
Tue Dec 27 12:48:55 2020 daemon.info : 05[KNL] unable to add SAD entry with SPI c89d7b58 (FAILED)
Tue Dec 27 12:48:55 2020 daemon.info : 05[KNL] received netlink error: No such file or directory (2)
Tue Dec 27 12:48:55 2020 daemon.info : 05[KNL] unable to add SAD entry with SPI 0de293df (FAILED)
Tue Dec 27 12:48:55 2020 daemon.info : 05[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel

```

closed 2020-12-29 by cwbsw

joker32, 2021-09-11:
How did you fix the error?

cwbsw, 2021-09-12:
> How did you fix the error?

because some necessary packages don't be selected along with `opkg install strongswan-default`, try `opkg install kmod-crypto-gcm`
https://forum.openwrt.org/t/ipsec-has-been-broken-for-a-while/81120