openwrt/packages #14389: Disable direct push to packages.git

aparcar, 2020-12-31:
Hi all, with a CI performing compile and (basic) runtime tests for multiple architectures, we could consider blocking direct pushes to the repository and only allow merging of pull requests, after they successfully passed all CI tests.

This is especially interesting for backports. It's now possible to compile test backports with the matching SDK version.

Applying such requirement means that everyone has to use the GitHub web interface (or cli) to some degree.

champtar, 2020-12-31:
I'm ok with it but i don't want to drive anyone away with hard requirement, strong encouragements are fine with me ;)
That also means that CI MUST work if we do that (no false negative)

neheb, 2020-12-31:
No false negatives is a hard CI problem :). It's mostly the occasional internet failures and such.

champtar, 2021-01-01:
I think the CI is in a pretty good state right now, just need to keep it that way ;)