openwrt/packages #14391: collectd not compiling under glibc

neheb, 2020-12-31:
Maintainer: @jow- @hnyman @PolynomialDivision 
Environment: glibc

https://downloads.openwrt.org/snapshots/faillogs/arc_archs/packages/collectd/compile.txt

PolynomialDivision, 2021-01-01:
This is `CONFIG_TARGET_archs38=y`, or? I can have a look at this. But currently, my toolchain fails to compile on my buildserver for this target.

curtdept, 2021-01-01:
Just ran an x86/64, its ok there. Must be isolated.

neheb, 2021-01-01:
> This is `CONFIG_TARGET_archs38=y`, or? I can have a look at this. But currently, my toolchain fails to compile on my buildserver for this target.

Why is that?

PolynomialDivision, 2021-01-01:
> > This is `CONFIG_TARGET_archs38=y`, or? I can have a look at this. But currently, my toolchain fails to compile on my buildserver for this target.
> 
> Why is that?

gdb fails to compile. But the server runs with an old centos version. I have to free some diskspace on my laptop and then I can try again.

neheb, 2021-01-01:
Oohhh yeah glibc doesn't compile on CentOS7 as it wrongly uses OS make instead of tools/make.

PolynomialDivision, 2021-01-02:
I just compiled for `CONFIG_TARGET_archs38=y` and it worked?

neheb, 2021-01-02:
Select all mods and try again.

neheb, 2021-01-02:
```
configure:4484: checking whether the C compiler works
configure:4506: ccache_cc -Os -pipe -matomic -mcpu=arc700 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-arc_arc700_glibc/collectd-5.12.0=collectd-5.12.0 -Wformat -Werror=format-security -fstack-protector >
/home/mangix/devstuff/openwrt/staging_dir/toolchain-arc_arc700_gcc-8.4.0_glibc/lib/gcc/arc-openwrt-linux-gnu/8.4.0/../../../../arc-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-arc_arc700_glibc/usr/lib/liblua.so: undefined reference to `dlopen'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-arc_arc700_gcc-8.4.0_glibc/lib/gcc/arc-openwrt-linux-gnu/8.4.0/../../../../arc-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-arc_arc700_glibc/usr/lib/liblua.so: undefined reference to `dlclose'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-arc_arc700_gcc-8.4.0_glibc/lib/gcc/arc-openwrt-linux-gnu/8.4.0/../../../../arc-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-arc_arc700_glibc/usr/lib/liblua.so: undefined reference to `dlerror'
/home/mangix/devstuff/openwrt/staging_dir/toolchain-arc_arc700_gcc-8.4.0_glibc/lib/gcc/arc-openwrt-linux-gnu/8.4.0/../../../../arc-openwrt-linux-gnu/bin/ld: /home/mangix/devstuff/openwrt/staging_dir/target-arc_arc700_glibc/usr/lib/liblua.so: undefined reference to `dlsym'
```

PolynomialDivision, 2021-01-03:
> ```
> configure:4484: checking whether the C compiler works
> configure:4506: ccache_cc -Os -pipe -matomic -mcpu=arc700 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable [...]

Seems like we should change
```
LDFLAGS="$$$$LDFLAGS -lm -lz" \
```
to
```
LDFLAGS="$$$$LDFLAGS -lm -lz -ldl" \
```

neheb, 2021-01-03:
I think it's a lua problem actually. I sent a patch: https://patchwork.ozlabs.org/project/openwrt/patch/20210102235625.264305-1-rosenp@gmail.com/

PolynomialDivision, 2021-01-08:
Can be closed, or? @aparcar merged lua patch.

PolynomialDivision, 2021-01-08:
And no collectd anymore:
https://downloads.openwrt.org/snapshots/faillogs/arc_archs/packages/

closed 2021-01-08 by neheb