openwrt/packages #14408: ddns-scripts  cURL Error！！

snakwu, 2021-01-02:
ddns-scripts[23684]: rss_snakwu_com: cURL Error: '127'
cloudflare

dibdot, 2021-01-04:
RC 127 means usually "command not found" ... just install curl.

snakwu, 2021-01-04:
curl has been installed!

dibdot, 2021-01-04:
Than it might be a path issue. What's the output of `command -v curl`

snakwu, 2021-01-04:
/usr/bin/curl

dibdot, 2021-01-04:
Sorry I don't use ddns. Anyway, I'm quite confident that your curl installation is somewhat broken. Does a curl download work manually in the CLI?

snakwu, 2021-01-04:
Directly select crul and then compile

dibdot, 2021-01-04:
I don't know what you mean by that. Please test curl manually in the CLI, e.g. start `curl -Iv http://connectivity-check.ubuntu.com` in a ssh session.

snakwu, 2021-01-04:
ok.thx!

closed 2021-01-04 by snakwu

opz20, 2021-02-03:
It is actually very easy to solve, just install libwolfssl with opkg
[image]

davemunneke, 2024-09-19:
Also needed `opkg install libcurl4` for me to get curl to work

davemunneke, 2024-09-19:
Also needed `opkg install libcurl4` for me to  work.