openwrt/packages #14409: adblock redirect not applied for IPv6

bonanza123, 2021-01-02:
I am using OpenWrt 19.07.5 r11257-5090152ae3 with adblock 4.0.7-3. I have the 
`option adb_forcedns '1'` enabled which is also applied for ipv4:

```
# iptables -S -t nat  | grep -i Adblock
-A zone_lan_prerouting -p tcp -m tcp --dport 53 -m comment --comment "!fw3: Adblock DNS, port 53" -j REDIRECT --to-ports 53
-A zone_lan_prerouting -p udp -m udp --dport 53 -m comment --comment "!fw3: Adblock DNS, port 53" -j REDIRECT --to-ports 53
-A zone_lan_prerouting -p tcp -m tcp --dport 853 -m comment --comment "!fw3: Adblock DNS, port 853" -j REDIRECT --to-ports 853
-A zone_lan_prerouting -p udp -m udp --dport 853 -m comment --comment "!fw3: Adblock DNS, port 853" -j REDIRECT --to-ports 853
-A zone_lan_prerouting -p tcp -m tcp --dport 5353 -m comment --comment "!fw3: Adblock DNS, port 5353" -j REDIRECT --to-ports 5353
-A zone_lan_prerouting -p udp -m udp --dport 5353 -m comment --comment "!fw3: Adblock DNS, port 5353" -j REDIRECT --to-ports 5353
```

However, the IPv6 tables are not created
```
# ip6tables -t nat -S
-P PREROUTING ACCEPT
-P INPUT ACCEPT
-P OUTPUT ACCEPT
-P POSTROUTING ACCEPT
```

Please note also that I have the install the ipv6 nat kernel modules beforehand (`kmod-ipt-nat6` i guess)

neheb, 2021-01-04:
ping @dibdot

dibdot, 2021-01-04:
IPv6-NAT is not installed/configured by default in OpenWrt If you need that, you have to intervene manually anyway - I do not plan to support this programmatically in adblock. See https://openwrt.org/docs/guide-user/firewall/fw3_configurations/intercept_dns#nat6 for DNS/NATv6 details.

bonanza123, 2021-01-05:
@dibdot 
Would it be possible to link the user (in the LUCI adblock interface) to a wiki page that contains such information and mention that 
IPv6 is not supported out of the box?

treysis, 2021-04-01:
@bonanza123 I think it should be introduced as dependency for AdBlock. Whoever really insists on version without IPv6 support can build custom image.

rklasen, 2023-01-23:
Has there been any update on this? I've just noticed that ipv6 dns requests can bypass the adblocker.

furai, 2023-02-05:
Actually came here looking for a solution. iPv6 isn't filtered on my end as well. I'm using OpenWRT in DHCP relay mode as I've got only /64 prefix assigned.

Neustradamus, 2023-05-18:
Important missing features. More and more networks are IPv6 now...

dibdot, 2023-12-30:
obsolete

closed 2023-12-30 by dibdot

treysis, 2024-01-01:
So has this been fixed?