openwrt/packages #14419: opkg update error

ragnaroks, 2021-01-04:
### Environment
mips_kc24 / openwrt 19.07(git-20.341.57626-51f55b5)

### Description
opkg update will failed,like this
```shell
root@OpenWrt:~# opkg update
Downloading http://downloads.openwrt.org/releases/19.07.5/targets/ramips/mt7621/packages/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/19.07.5/targets/ramips/mt7621/packages/Packages.gz

Downloading http://downloads.openwrt.org/releases/19.07.5/targets/ramips/mt7621/kmods/4.14.209-1-2e88863ccdd594fb8e842df3c25842ee/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/19.07.5/targets/ramips/mt7621/kmods/4.14.209-1-2e88863ccdd594fb8e842df3c25842ee/Packages.gz

Downloading http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/Packages.gz

Downloading http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/freifunk/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/freifunk/Packages.gz

Downloading http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/luci/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/luci/Packages.gz

Downloading http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/packages/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/packages/Packages.gz
```

the actually reason is the source `http://downloads.openwrt.org` not support `HTTP`,it is will force redirect to `HTTPS`
```shell
ragnaroks@localhost:~# curl -v http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/
*   Trying 2a01:4f8:251:321::2...
* TCP_NODELAY set
*   Trying 168.119.138.211...
* TCP_NODELAY set
* Connected to downloads.openwrt.org (168.119.138.211) port 80 (#0)
> GET /releases/19.07.5/packages/mipsel_24kc/base/ HTTP/1.1
> Host: downloads.openwrt.org
> User-Agent: curl/7.55.1
> Accept: */*
>
< HTTP/1.1 301 Moved Permanently
< Server: nginx/1.14.2
< Date: Mon, 04 Jan 2021 12:40:53 GMT
< Content-Type: text/html
< Content-Length: 185
< Connection: keep-alive
< Location: https://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/
<
<html>
<head><title>301 Moved Permanently</title></head>
<body bgcolor="white">
<center><h1>301 Moved Permanently</h1></center>
<hr><center>nginx/1.14.2</center>
</body>
</html>
* Connection #0 to host downloads.openwrt.org left intact
```

jow-, 2021-01-04:
Something is messing with your HTTP requests. Works fine here:

```
root@OpenWrt:~# cd /tmp/
root@OpenWrt:/tmp# wget http://downloads.openwrt.org/releases/19.07.5/packages/mipsel
_24kc/packages/Packages.gz
--2021-01-04 13:18:18--  http://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/packages/Packages.gz
Resolving downloads.openwrt.org... 2a01:4f8:251:321::2, 168.119.138.211
Connecting to downloads.openwrt.org|2a01:4f8:251:321::2|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 414763 (405K) [application/octet-stream]
Saving to: 'Packages.gz'

Packages.gz         100%[===================>] 405.04K  1.49MB/s    in 0.3s    

2021-01-04 13:18:18 (1.49 MB/s) - 'Packages.gz' saved [414763/414763]

root@OpenWrt:/tmp# zcat Packages.gz | head -n 5
Package: acl
Version: 2.2.53-1
Depends: libc, libacl
License: LGPL-2.1 GPL-2.0
Section: utils
root@OpenWrt:/tmp# 
```

closed 2021-01-04 by jow-

ragnaroks, 2021-01-04:
it's very interesting,i tried some servers from some location,all responsed 301 redirect,but you are direct connect 80.

i write some thing here maybe help others
1. download and install **WinSCP**,connect to router with `scp` protocol
2. download `https://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/libustream-mbedtls20150806_2020-03-13-40b563b1-1_mipsel_24kc.ipk` then upload to router and install
3. download `https://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/ca-bundle_20200601-1_all.ipk` then upload to router and install
4. download `https://downloads.openwrt.org/releases/19.07.5/packages/mipsel_24kc/base/ca-certificates_20200601-1_all.ipk` then upload to router and install
5. now you should be `opkg update` success