openwrt/packages #14442: vpn-policy-routing: Does not work with busybox's ip command

rah2501, 2021-01-09:
Maintainer: @stangri
Environment: mvebu-cortexa9, linksys_wrt3200acm, "19.07-SNAPSHOT, r11261-3f5fecfd33"

Description:

I installed the vpn-policy-routing package but I get errors and the routing I configured doesn't work:

```
# /etc/init.d/vpn-policy-routing start
Creating table 'wan/pppoe-wan/51.xxx.xxx.xxx' ip: invalid argument 'ls' to 'ip'
ip: invalid argument '0x010000/0xff0000' to 'fwmark'
[✓]
Creating table 'vpn/tun0/172.yyy.yyy.yyy' ip: invalid argument 'ls' to 'ip'
ip: invalid argument '0x020000/0xff0000' to 'fwmark'
[✓]
Routing 'NoVPN' via wan [✓]
vpn-policy-routing 0.2.1-13 started with gateways:
wan/pppoe-wan/51.xxx.xxx.xxx
vpn/tun0/172.yyy.yyy.yyy [✓]
vpn-policy-routing 0.2.1-13 monitoring interfaces: wan vpn .
#
```

The critical errors are:

```
ip: invalid argument 'ls' to 'ip'
ip: invalid argument '0x010000/0xff0000' to 'fwmark'
```

I have to also install the ip-full package, and only then does it work:

```
# /etc/init.d/vpn-policy-routing stop
vpn-policy-routing 0.2.1-13 stopped [✓]
# opkg install ip-full
Installing ip-full (5.0.0-2.1) to root...
Downloading http://downloads.openwrt.org/releases/19.07-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/base/ip-full_5.0.0-2.1_arm_cortex-a9_vfpv3-d16.ipk
Configuring ip-full.
# /etc/init.d/vpn-policy-routing start
Creating table 'wan/pppoe-wan/51.xxx.xxx.xxx' [✓]
Creating table 'vpn/tun0/172.yyy.yyy.yyy' [✓]
Routing 'NoVPN' via wan [✓]
vpn-policy-routing 0.2.1-13 started with gateways:
wan/pppoe-wan/51.xxx.xxx.xxx
vpn/tun0/172.yyy.yyy.yyy [✓]
vpn-policy-routing 0.2.1-13 monitoring interfaces: wan vpn .
#
```

There seems to be [some effort](https://github.com/openwrt/packages/blob/f95f280a463651dfde53e610370c07893b7f41d4/net/vpn-policy-routing/Makefile#L18) to specify an appropriate dependency but this is clearly not working right.

stangri, 2021-01-09:
Thanks for the report, the change has already been merged in the source code: https://github.com/stangri/source.openwrt.melmac.net/pull/121, you can install updated IPK from [my repo](https://repo.openwrt.melmac.net/) or build it yourself.

Once it and the other changes are tested, I'll send PR for OpenWrt repo.

stangri, 2021-04-20:
This has been resolved a while ago, @aparcar @jow- as I don't have merge rights for packages, I can't close this.

closed 2021-04-20 by aparcar