openwrt/packages #14465: nginx-util not creating self-signed ssl certs

antoinedeschenes, 2021-01-11:
Maintainer: @peter-stadler 
Environment: openwrt-master + packages-0145513

Description:

luci-ssl-nginx crashes after factory reset caused by missing self-signed certificates.

`nginx-util init_lan` in `/etc/init.d/nginx` doesn't create the self-signed certs for `/etc/nginx/conf.d/_lan.conf`. Code seems to skip certificate creation if `.conf` file is present. (It's always there as it's copied in the ROM as per the nginx package Makefile)

I can get it to generate the cert by renaming `_lan.conf` temporarily, running `nginx-util init_lan`, then manually restarting nginx.

Note: Currently compiling a build that removes _lan.conf, as it looks like it should be created from uci config now.

antoinedeschenes, 2021-01-11:
Ok I confirm my build removing `net/nginx/files/_lan.conf` works properly. I suppose _redirect2ssl.conf should be removed as well

peter-stadler, 2021-01-11:
Thank you :-) I prepared a PR for Nginx #14471 ...

peter-stadler, 2021-01-14:
It should work now.

closed 2021-01-18 by antoinedeschenes