openwrt/packages #14467: mpd-full + upmpdcli: out of memory with certain audio files

skateskate, 2021-01-11:
Maintainer: @thess @ignisf 
Environment: Oolite v5.1 board (CPU: QCA9531, RAM: 128 MB), mips24_kc, SNAPSHOT r6712-7e32e86, ar71xx/generic, kernel 4.9.91, Upmpdcli 1.4.6 libupnpp 0.17.2, Music Player Daemon 0.21.25 (0.21.25)

Description:
The system goes in "out of memory" when I try to stream certain FLAC files (44,1 kHz 24bit) using upmpdcli + mpd in combination with BubbleUPnP (Android version). If the same audio files are reproduced locally (i.e., using mpc and reading them from an USB stick connected to the board, so not using upmpdcli) everything is fine. 

Mpd logs show these messages when the board reaches the out of memory condition:
```
alsa_output: Underrun on ALSA device "hw:CARD=x20,DEV=0"
alsa_output: Decoder is too slow; playing silence to avoid xrun
```

When the system is in the out of memory condition, I need to reboot the board since everything is unresponsive. I don't know if it is a problem of mpd or upmpdcli, the strange thing is that, from USB, the files are reproduced correctly.

I am a linux newbe, please forgive me if I have made some mistakes in this issue report.

ignisf, 2021-01-11:
Hi could you try to play them from a webserver. I think the command is `mpc add http://yourhttpserver.com:port/youfile.flac`? You can start a simple http server that shares the working directory by executing `python -m http.server 8000`.

neheb, 2021-01-12:
ping @medoc92

mpd is the one with memory problems it seems. In OpenWrt, it is patched to use libnpupnp.

skateskate, 2021-01-12:
> Hi could you try to play them from a webserver. I think the command is `mpc add http://yourhttpserver.com:port/youfile.flac`? You can start a simple http server that shares the working directory by executing `python -m http.server 8000`.

Using the python server everything works fine.

medoc92, 2021-01-12:
This is not related to the npupnp instance in mpd, it is only used when using an UPnP Media Server as a backend, not for normal operation like here. MPD is just using the Curl input plugin to fetch from the Media Server through HTTP. upmpdcli is doing all the UPnP stuff.

The output of "ps aux" before  and when the problem occurs would be useful, I don't know how the processes in the above trace are selected.

As someone who began working on a shared computer with 1MB of memory, I shudder to say this, but 128 MB is not a lot these days.

I had a look at the mpd resident size while playing from minimserver, and it's clearly linked to the track size. As far as I can see, it actually buffers the whole track (I see around 80 mb for a 14mn 44/16/2 track)

The strange thing is that you can't reproduce the problem when using mpc to play from a WEB server. The Curl input plugin is used in the same way in both cases (upmpdcli is just another mpd client like mpc, just with a different front-end interface). Maybe try to play a longer track from the WEB server and see what happens.

I would guess that there is a limit to how much mpd will prefetch, but I don't know how it is set.

medoc92, 2021-01-12:
Just ran a few more tests, and the mpd resident size seems to depend on the track size in a weird way: just tried to play a complete CD image, and it stays at 20MB. Go back to playing a 6mn track and it's at 55MB. It would weirdly appear that it has a track size limit beyond which it buffers a fixed size which is much smaller than the limit ?? Maybe there is an explanation in the mpd doc...

skateskate, 2021-01-12:
> The output of "ps aux" before and when the problem occurs would be useful, I don't know how the processes in the above trace are selected.

Unfortunately I don't have `ps aux` installed in my system, I hope that the output of `top` is equally valuable.

```
System in idle (i.e., nothing reproducing)
==========================================

Mem: 44076K used, 80804K free, 152K shrd, 6832K buff, 20064K cached
CPU:   0% usr   0% sys   0% nic  99% idle   0% io   0% irq   0% sirq
Load average: 0.01 0.27 0.20 1/59 1679
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
    4     2 root     SW       0   0%   0% [kworker/0:0]
 1679  1667 root     R     1200   1%   0% top
 1544     1 upmpdcli S     6188   5%   0% /usr/bin/upmpdcli -i br-lan
 1666  1514 root     S     1132   1%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
    8     2 root     SW       0   0%   0% [kworker/u2:1]
 1116     1 root     S<   14456  12%   0% /usr/bin/mpd --no-daemon /etc/mpd.conf
 1605     1 pulse    S<   10236   8%   0% /usr/bin/pulseaudio --system --disallow-exit --disallow-module-loading --disable-shm --exit-idle-time=-1 --realtime=false
 1352     1 root     S     1696   1%   0% /usr/sbin/hostapd -s -P /var/run/wifi-phy0.pid -B /var/run/hostapd-phy0.conf
 1013     1 root     S     1652   1%   0% /sbin/netifd
    1     0 root     S     1540   1%   0% /sbin/procd
 1050     1 root     S     1420   1%   0% /usr/sbin/odhcpd
 1407     1 dnsmasq  S     1312   1%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
  938     1 root     S     1224   1%   0% /sbin/logd -S 64
 1667  1666 root     S     1204   1%   0% -ash
 1580     1 root     S<    1200   1%   0% /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 2.openwrt.pool.ntp.org -p 3.openwr
  525     1 root     S     1184   1%   0% /sbin/ubusd
 1514     1 root     S     1064   1%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
  542     1 root     S      900   1%   0% /sbin/askfirst /usr/libexec/login.sh
  188     2 root     SW       0   0%   0% [spi0]
  122     2 root     SW       0   0%   0% [kworker/0:1]
    3     2 root     SW       0   0%   0% [ksoftirqd/0]
    6     2 root     SW       0   0%   0% [kworker/u2:0]
  423     2 root     SW       0   0%   0% [usb-storage]
  232     2 root     SW<      0   0%   0% [bioset]
  227     2 root     SW<      0   0%   0% [bioset]
  323     2 root     SW<      0   0%   0% [ipv6_addrconf]
  217     2 root     SW<      0   0%   0% [bioset]
  207     2 root     SW<      0   0%   0% [bioset]
  212     2 root     SW<      0   0%   0% [bioset]
  222     2 root     SW<      0   0%   0% [bioset]
  331     2 root     SW<      0   0%   0% [kworker/0:1H]
   92     2 root     SW<      0   0%   0% [kblockd]
   89     2 root     SW<      0   0%   0% [crypto]
   87     2 root     SW<      0   0%   0% [writeback]
   90     2 root     SW<      0   0%   0% [bioset]
    7     2 root     SW<      0   0%   0% [lru-add-drain]
  632     2 root     SW<      0   0%   0% [cfg80211]
  422     2 root     SW<      0   0%   0% [scsi_tmf_0]
 1521     2 root     SWN      0   0%   0% [jffs2_gcd_mtd3]
  428     2 root     SW<      0   0%   0% [bioset]
    2     0 root     SW       0   0%   0% [kthreadd]
    5     2 root     SW<      0   0%   0% [kworker/0:0H]
   36     2 root     SW       0   0%   0% [oom_reaper]
  129     2 root     SW       0   0%   0% [kswapd0]
  421     2 root     SW       0   0%   0% [scsi_eh_0]

FLAC song (6:52 minutes, 83.8 MB) streamed from python web server
================================================================

Mem: 50120K used, 74760K free, 156K shrd, 6832K buff, 20136K cached
CPU:  21% usr  10% sys   0% nic  53% idle   0% io   0% irq  13% sirq
Load average: 0.07 0.10 0.11 1/66 1905
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 1887     1 root     S<   19788  16%  31% /usr/bin/mpd --no-daemon /etc/mpd.conf
    3     2 root     SW       0   0%   5% [ksoftirqd/0]
 1544     1 upmpdcli S     6208   5%   1% /usr/bin/upmpdcli -i br-lan
    4     2 root     SW       0   0%   1% [kworker/0:0]
 1679  1667 root     R     1200   1%   1% top
[... 194 lines cut ...]
[ 2945.390807] Total swap = 0kB
[ 2945.393782] 32768 pages RAM
[ 2945.396671] 0 pages HighMem/MovableOnly
[ 2945.400642] 1548 pages reserved
[ 2945.403889] [ pid ]   uid  tgid total_vm      rss nr_ptes nr_pmds swapents oom_score_adj name
[ 2945.412734] [  525]     0   525      296       16       3       0        0             0 ubusd
[ 2945.421652] [  542]     0   542      302       12       3       0        0             0 ash
[ 2945.430391] [  938]     0   938      306       26       4       0        0             0 logd
[ 2945.439221] [ 1013]     0  1013      413       67       3       0        0             0 netifd
[ 2945.448227] [ 1050]     0  1050      355       49       3       0        0             0 odhcpd
[ 2945.457220] [ 1352]     0  1352      424       47       4       0        0             0 hostapd
[ 2945.466325] [ 1407]   453  1407      328       42       4       0        0             0 dnsmasq
[ 2945.475431] [ 1514]     0  1514      266       11       4       0        0             0 dropbear
[ 2945.484622] [ 1544]    89  1544     1563      168       5       0        0             0 upmpdcli
[ 2945.493810] [ 1580]     0  1580      300       19       3       0        0             0 ntpd
[... cut ...]
[ 2945.529932] [ 1680]     0  1680      283       25       4       0        0             0 dropbear
[ 2945.539126] [ 1681]     0  1681      301       12       3       0        0             0 ash
[ 2945.547866] [ 1959]     0  1959    25406    21784      27       0        0             0 mpd
[ 2945.556585] Out of memory: Kill process 1959 (mpd) score 677 or sacrifice child
[ 2945.564197] Killed process 1959 (mpd) total-vm:101624kB, anon-rss:86912kB, file-rss:224kB, shmem-rss:0kB
[ 2946.825536] oom_reaper: reaped process 1959 (mpd), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

medoc92, 2021-01-12:
I am not familiar with this version of top, and I'm not sure of what VSZ exactly measures. In any cases, mpd is changing from 20 to 99 MB, so it's not too hard to guess where the problem comes from. I'm not familiar enough with the mpd code to know how to cap the buffering. 

I've done some tests with mpc and a few HTTP servers.

On all of Apache, MinimServer, and the upmpdcli-uprcl Media Server, the mpd process size grows (to the track size it appears, at least up to some limit).

Only with the python http.server does mpd stay at 20MB. 

The only difference I can see between the servers is that the ones in the first group all support seeking (by setting Accept-Ranges: bytes), but the python server does not. Maybe this induces a different behaviour inside the mpd curl input module or around it.

I'd take the question to an mpd forum or mailing list, because the mystery clearly resides inside mpd.

vkostas, 2021-02-07:
Wouldn't it be better to upgrade to (i.e. build) the latest MPD version (0.22.4) so we can have better support from MPD owner?

neheb, 2021-02-07:
Nope. MPD requires C++17 which the build bots do not support.

ReimuNotMoe, 2024-04-04:
I have encountered this problem today as well. It appears that the `curl` plugin is problematic. When playing large FLAC files through network, it will quickly eat all the RAM up.

Temporary solution: Disable the `curl` plugin and use the `ffmpeg` plugin for the `http://` protocol.

neheb, 2024-04-04:
That's interesting that ffmpeg can be used for http://. In which case, what's the advantage to curl?

ReimuNotMoe, 2024-04-04:
> That's interesting that ffmpeg can be used for http://. In which case, what's the advantage to curl?

It doesn't OOM

neheb, 2024-04-04:
Just read the code. The same function "protocol_is_whitelisted" is used for both curl and ffmpeg. Which means they support the same protocols. The difference in OpenWrt seems to be mpd vs mpd-full.