openwrt/packages #14475: syncthing: Update to current version?

Fffrank, 2021-01-12:
# Issue template (remove lines from top till here)

Maintainer: @aparcar 
Environment: All require updates

Description:

Not sure why the version listed as the source is 1.4.0-2 but the current opkg version is 1.9.0-2?  Regardless, it is several months behind current release and should be updated.

aparcar, 2021-01-13:
I guess you're talking about the 19.07 release?

Fffrank, 2021-01-13:
Apologies -- referring to the syncthing version.  The opkg version is listed as 1.4.0-2.  The actual version in the rep is 1.9.0-2 and the most current stable release from syncthing is 1.12.1?

https://github.com/syncthing/syncthing/releases

Mrnofish, 2021-03-11:
Bumping: Syncthing stable has currently reached 1.14 and has RCs for 1.15.

The version in the packages repo is 1.4 which is, at the time of this writing, a year old and ten upgrades behind.

Is there some sort of blocker?
Can we contribute somehow?

aparcar, 2021-03-11:
Are you excited to become the maintainer?

Mrnofish, 2021-03-11:
Haha! Not at all, actually! I have begun using Syncthing not too long ago, and noticed in the last year it has received a slew of bug fixes, so I was wondering whether there might be a very good reason for the situation.

Because of the nick, it wasn't immediately obvious to me that you are the maintainer.

Peace!

aparcar, 2021-03-11:
I'll update it later today.

aparcar, 2021-03-11:
Updated

closed 2021-03-11 by aparcar

denisfrench, 2021-04-02:
Hi @aparcar ; thanks for updating the Syncthing package. I am having an issue that looks like it has been fixed in later versions, but I am having trouble upgrading.

LuCI wasn't showing Syncthing as being upgradable after updating the lists, so I tried via the terminal.

After opkg update still no upgradable version. I tried using --force-reinstall, but the same old version was installed, and is showing in the Syncthing web GUI about panel and logs.

I'm still running OpenWrt 19.07.4 (on a Netgear R7800 - arm_cortex-a15_neon-vfpv4); could that be an issue?

Any advice appreciated.

aparcar, 2021-04-02:
@denisfrench 19.07 uses an older Go version which is not compatible with recent Syncthing versions (iirc). So you best bet is to upgrade to 21.02 once it's out and enjoy the latest version.

mmokrejs, 2021-05-29:
So why is this bug closed if we cannot get newer syncthing on 19.07? The 1.4.0 version is so old it does not properly fallback from tcp6 to tcp4 connections so sync initiation takes ages. Nevertheless thank you for the bump. The deps for syncthing are horrible and requiring 7.9MB for just syncthing (without even counting the dependencies) is a huge space stress on tiny ARM routers. Hope syncthing-1.16.1 will be available on openwrt once we can upgrade to new stable.

aparcar, 2021-05-31:
The new OpenWrt release should be able to handle newer syncthing versions.

Regarding size, I'm not sure how to slim down Go binaries. If you have any ideas please elaborate (in a freshly opened issue).

Fffrank, 2021-07-14:
I was able to cross-compile a newer version of syncthing that runs fine on my OpenWRT rpi4.  Once there is a binary built it can run without the go dependency being updated.  It was very easy to do.  If anyone wants to try, here are the basic steps:

[Install Go](https://golang.org/doc/install)
[Build Syncthing from Source](https://docs.syncthing.net/dev/building.html) using this command: 
`go run build.go -goos linux -goarch arm64 tar`

Transfer new tar bundle to your OpenWRT install and run the binary (you can point it to your existing config directory and it should update the db appropriately.)  Be sure to make back-ups first.

Syncing starts almost immediately and hashing seems to be faster.

mmokrejs, 2021-08-17:
> The new OpenWrt release should be able to handle newer syncthing versions.
> 
> Regarding size, I'm not sure how to slim down Go binaries. If you have any ideas please elaborate (in a freshly opened issue).

So where are the new syncthing binaries? I had `OpenWrt 21.02.0-rc2` and now have even `OpenWrt 21.02.0-rc4` installed but there is still only `syncthing 1.14.0` available. What am I missing?

aparcar, 2021-08-18:
Let's see how https://github.com/openwrt/packages/pull/16392 goes. Golang was upgraded to a newer version therefore it might build.