openwrt/packages #14477: ddns-scripts_route53-v1: LUCI doesn't show fields for `password` and `domain`

hschmidt, 2021-01-12:
Maintainer: @maxberger
Environment: TP-Link Archer A7 v5, Qualcomm Atheros QCA956X ver 1 rev 0, OpenWrt 19.07.5 r11257-5090152ae3 / LuCI openwrt-19.07 branch git-20.341.57626-51f55b5

Description:

Fields for `domain` (in this case the Zone ID) and `password` (in this case the AWS secret key ID) are missing. You can set them via UCI but when applying changes via LUCI those to properties are removed again.

[image]

hectorese1, 2021-01-17:
> Maintainer: @maxberger
> Environment: TP-Link Archer A7 v5, Qualcomm Atheros QCA956X ver 1 rev 0, OpenWrt 19.07.5 r11257-5090152ae3 / LuCI openwrt-19.07 branch git-20.341.57626-51f55b5
>
> Description: [...]

``

maxberger, 2021-01-22:
Sorry about the late reply. I am actually only maintaing the backend and little idea about the frontend. I was unable to replicate the issue on my machine, but it runs on the current snapshot, so it probably behaves different.

@Ansuel: Do you have any idea why username/password would not show up in 19.07.5 ?

Ansuel, 2021-05-11:
Is this still present ?

hschmidt, 2021-05-15:
Yes.

Ansuel, 2021-05-19:
FUN FIX...

Ansuel, 2021-05-19:
You can also test this by manually edit the script and restart the router (or clean luci cache)

hschmidt, 2021-05-20:
I tested this by copying the script from the PR directly onto the router. The previously missing fields are displayed and applying changes does not remove any uci config entries anymore.

[image]

Thank you!

maxberger, 2021-05-20:
Thank you @Ansuel  for the fix and thank you @hschmidt for the testing! I was indeed unaware that the script was being parsed for the parameters.

Even though this is not an issue on current, should this still go there for consistency?

feckert, 2021-05-20:
As I see it, we also need this for the master and the next stable branch owrt-21.02. If this is ok for all involved, then I will add this first to the master and then as a cherry-pick to the stable branches?

PR #15663 (2021-07-02): [19-07] ddns-scripts: standardize required params declaration
The luci app scan the script and search for params in the form of $required_params.
This script use the form "${required_params}" and cause confusion with the luci app by hiding needed values. Fix this by using the standard way to declare required params following other ddns scripts.

Fixes: #14477