openwrt/packages #14479: libiwinfo20210106 Conflicts With libiwinfo20200105

JamesVorder, 2021-01-13:
# libiwinfo20210106 Conflicts With libiwinfo20200105

Maintainer: @jow- 
Environment: aarch64_cortex-a53, Raspberry Pi 4b, OpenWrt SNAPSHOT, r15204-b2d48c1dfe

Description:

I am experiencing the following error:

```
# opkg install libiwinfo20210106
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Installing libiwinfo20210106 (2021-01-06-23d2722b-1) to root...
Downloading https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2710/packages/libiwinfo20210106_2021-01-06-23d2722b-1_aarch64_cortex-a53.ipk
Collected errors:
 * check_data_file_clashes: Package libiwinfo20210106 wants to install file /usr/share/libiwinfo/hardware.txt
        But that file is already provided by package  * libiwinfo20200105
 * opkg_install_cmd: Cannot install package libiwinfo20210106.
```

jookk, 2021-07-10:
I wanted to install luci-app-statistics, but I cant:

Collected errors:
 * check_data_file_clashes: Package libiwinfo20210430 wants to install file /usr/share/libiwinfo/hardware.txt
	But that file is already provided by package  * libiwinfo20210106
 * opkg_install_cmd: Cannot install package luci-app-statistics.

ieure, 2021-07-10:
Same issue here, appears to be that luci-mod-status and rpcd-mod-iwinfo need to be rebuilt against libiwinfo20210430.

And maybe iwinfo should be updated so the version isn't in the package name, which would prevent this recurring the next time it's updated.

hnyman, 2023-01-14:
normal temporary mismatch of ABI versioned package versions after a version bump.
Unrelated to the packages feed.

closed 2023-01-14 by hnyman