openwrt/packages #14484: sane-backends: packages are hidden and not compiled

dangowrt, 2021-01-14:
Maintainer: @luizluca
Environment: OpenWrt snapshot

Description:
Due to Kconfig jugglery none of the backend packages are built and available for download. Please fix the package Makefile so that all backends are built for all available platforms and can hence be installed via opkg.
I don't even find a way in buildroot menuconfig to build with all backends :(

dangowrt, 2021-01-14:
See
https://downloads.openwrt.org/snapshots/packages/x86_64/packages/
There are `sane-frontsend` and `sane-daemon`, but all backends are missing.

luizluca, 2021-01-14:
I'll take a look. Thanks for reporting

luizluca, 2021-01-14:
I got it. It was hiding the sane-backends in menuconfig since 5a7148d112113544611ee358a7d062cec85c1629 when BUILDONLY was added. I didn't catch it in my tests as I do not use menuconfig for config.

@neheb, could you give me a hand on this? I would not like to undo your change before I get your opinion. 

sane-backends are both the name of the project but also only part of the code (drivers). This introduces some confusion.

I'm using a non-menu package that does not have an install as a placeholder for a submenu, declared in /config, for other backends (the drivers). I am manually generating each package menuconfig entry in that submenu, hiding the normal package menuconfig entry. I was getting something like this (before BUILDONLY):

- sane-backends
  - Select SANE Backends --->
     - select all backends
     - sane-aaa
     - sane-bbb
     - sane-ccc
     - sane-ddd
     - sane-eee
     - ...
     - sane-zzz
- sane-daemon
- sane-frontends

Normally, this type of setup would have a MENU:=1 and all subpackages would be there. However, sane-daemon and sane-frontends are not backends. As they do not depend on sane-backends, the menu breaks as soon as sane-daemon is processed:

- sane-backends --->
  - select all backends
  - sane-aaa
  - sane-bbb
  - sane-ccc
  - sane-ddd
- sane-daemon
- sane-frontends
- sane-eee
- ...
- sane-zzz

It looks like kconfig has two ways to define a submenu, with the explicitly menu/endmenu entry or depending on other entry. However, as soon it sees a endmenu or an entry that does not depend on the menu, it end it. I would need to reorder packages in menu to keep sane-daemon and sane-frontend after the last sane backend.

Is it a better way to fix the issue or do I simply remove BUILDONLY from sane-backends?

neheb, 2021-01-14:
Just remove BUILDONLY.

commit bf4340e1 (2021-01-15): sane-backends: revert BUILDONLY flag

BUILDONLY was disabling SANE backends (drivers) build.

Closes #14484
commit cae913fe (2022-04-22): sane-backends: revert BUILDONLY flag

BUILDONLY was disabling SANE backends (drivers) build.

Closes #14484

(cherry picked from commit bf4340e19ecd85e44d9b5a08719dd0e531d2c20a)
closed 2021-01-15 by luizluca