openwrt/packages #14485: [19.07]mwan3:user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)

openwrt-diy, 2021-01-15:
**Firmware Version: OpenWrt-19.07.5
Kernel Version: 4.14.209
Software Version: mwan3-2.8.15-1
There are multiple mwan3 problems in the system log**

Fri Jan 15 16:11:53 2021 daemon.info pppoe-server[7271]: Session 140 closed for client 24:4b:fe:99:71:66 (172.16.0.141) on br-lan
Fri Jan 15 16:13:29 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 46) from 172.16.0.171 port 60750 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:13:29 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 4 (fd 44) from 172.16.0.171 port 60744 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:14:54 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:14:56 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:15:41 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 42) from 172.16.0.171 port 60732 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:16:15 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:16:51 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:18:37 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 44) from 172.16.0.171 port 60759 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:19:33 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:20:14 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:20:49 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:21:02 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:21:55 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 44) from 172.16.0.171 port 60767 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:23:56 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:24:07 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 6 (fd 43) from 172.16.0.171 port 60653 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:24:09 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:24:24 2021 daemon.info pppoe-server[758]: Session 129 created for client 24:4b:fe:96:9e:d2 (172.16.0.130) on br-lan using Service-Name 'service-name'
Fri Jan 15 16:24:24 2021 daemon.info pppd[758]: Plugin rp-pppoe.so loaded.
Fri Jan 15 16:24:24 2021 daemon.info pppd[758]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Fri Jan 15 16:24:24 2021 daemon.notice pppd[758]: pppd 2.4.7 started by root, uid 0
Fri Jan 15 16:24:24 2021 daemon.warn pppd[758]: Connected to 24:4b:fe:96:9e:d2 via interface br-lan
Fri Jan 15 16:24:24 2021 daemon.info pppd[758]: Using interface ppp30
Fri Jan 15 16:24:24 2021 daemon.notice pppd[758]: Connect: ppp30 <--> br-lan
Fri Jan 15 16:24:25 2021 daemon.info netdata[8550]: RRDSET: chart name 'net.ppp30' on host '4WAN-PPPOE-LDC' already exists.
Fri Jan 15 16:24:26 2021 daemon.info pppd[758]: user 10086191129 logged in on tty br-lan intf ppp30
Fri Jan 15 16:24:26 2021 daemon.notice pppd[758]: peer from calling number 24:4B:FE:96:9E:D2 authorized
Fri Jan 15 16:24:26 2021 kern.warn kernel: [67676.688181] mppe_decomp_init[30]: unknown key length
Fri Jan 15 16:24:26 2021 daemon.notice pppd[758]: local  IP address 172.16.0.1
Fri Jan 15 16:24:26 2021 daemon.notice pppd[758]: remote IP address 172.16.0.129
Fri Jan 15 16:24:27 2021 daemon.info netdata[8550]: RRDSET: chart name 'net_packets.ppp30' on host '4WAN-PPPOE-LDC' already exists.
Fri Jan 15 16:25:36 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 44) from 172.16.0.171 port 60938 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:26:07 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:27:47 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 3 (fd 45) from 172.16.0.171 port 60748 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:28:17 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:29:15 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 44) from 172.16.0.171 port 60980 is idle for more than 60 seconds - closing it.
Fri Jan 15 16:30:10 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:30:12 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:30:24 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:30:54 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:31:07 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:33:05 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 16:33:18 2021 user.info mwan3track[14085]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 16:34:01 2021 daemon.info netdata[8550]: POLLFD: LISTENER: client slot 5 (fd 45) from 172.16.0.171 port 60989 is idle for more than 60 seconds - closing it.

openwrt-diy, 2021-01-15:
[image]
[image]
[image]
[image]
[image]

feckert, 2021-01-15:
Does it work with the other version 19.07.4?
As I see it, it can not reach some tracking IP targets.
I think this is a timing problem.
You can increase the timeout of the ping.
Maybe the tracking IP is answering too slow or is not reachable.
If it does not reach the tracking IP it takes another one.
The interface is recognized as up, isn´t it?
So it is only log outputs that he does not reach the tracking IP.
What is the output of ubus call mwan3 status?

openwrt-diy, 2021-01-15:
> Does it work with the other version 19.07.4?
> As I see it, it can not reach some tracking IP targets.
> I think this is a timing problem.
> You can increase the timeout of the ping.
> Maybe the tracking IP [...]

@feckert 

/$ ubus call mwan3 status
{
	"interfaces": {
		"wan": {
			"age": 2,
			"online": 79205,
			"offline": 0,
			"uptime": 79206,
			"score": 6,
			"lost": 0,
			"turn": 14387,
			"status": "online",
			"enabled": true,
			"running": true,
			"up": true,
			"track_ip": [
				{
					"ip": "119.29.29.29",
					"status": "skipped",
					"latency": 0,
					"packetloss": 0
				},
				{
					"ip": "180.76.76.76",
					"status": "skipped",
					"latency": 0,
					"packetloss": 0
				},
				{
					"ip": "223.5.5.5",
					"status": "up",
					"latency": 0,
					"packetloss": 0
				},
				{
					"ip": "223.6.6.6",
					"status": "up",
					"latency": 0,
					"packetloss": 0
				}
			]
		},
		"wanb": {
			"age": 4,
			"online": 15492,
			"offline": 0,
			"uptime": 79204,
			"score": 6,
			"lost": 0,
			"turn": 3050,
			"status": "online",
			"enabled": true,
			"running": true,
			"up": true,
			"track_ip": [
				{
					"ip": "119.29.29.29",
					"status": "skipped",
					"latency": 0,
					"packetloss": 0
				},
				{
[... 400 lines cut ...]
			],
			"wanc_wanb": [
				
			],
			"wanc_wand": [
				
			],
			"wand_only": [
				
			],
			"wand_wan": [
				
			],
			"wand_wanb": [
				
			],
			"wand_wanc": [
				
			],
			"wireless": [
				
			]
		}
	}
}

openwrt-diy, 2021-01-15:
I now adjust and save the Ping timeout seconds, and then restart the route to observe whether the situation improves.

uci set mwan3.wan.timeout=5
uci set mwan3.wanb.timeout=5
uci set mwan3.wanc.timeout=5
uci set mwan3.wand.timeout=5
uci commit mwan3
reboot

openwrt-diy, 2021-01-15:
Fri Jan 15 20:58:23 2021 daemon.notice pppd[20116]: local  IP address 172.16.0.1
Fri Jan 15 20:58:23 2021 daemon.notice pppd[20116]: remote IP address 172.16.0.139
Fri Jan 15 20:58:23 2021 daemon.warn pppoe-server[3216]: PADT for session 207 received from 64:6E:97:D0:9D:FA; should be from 00:00:00:00:00:00
Fri Jan 15 20:58:24 2021 daemon.warn pppoe-server[3216]: PADT for session 207 received from 64:6E:97:D0:9D:FA; should be from 00:00:00:00:00:00
Fri Jan 15 20:58:24 2021 daemon.info pppoe-server[20478]: Session 6 created for client 64:6e:97:d0:9d:fa (172.16.0.7) on br-lan using Service-Name ''
Fri Jan 15 20:58:24 2021 daemon.info pppd[20478]: Plugin rp-pppoe.so loaded.
Fri Jan 15 20:58:24 2021 daemon.info pppd[20478]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Fri Jan 15 20:58:24 2021 daemon.notice pppd[20478]: pppd 2.4.7 started by root, uid 0
Fri Jan 15 20:58:24 2021 daemon.warn pppd[20478]: Connected to 64:6e:97:d0:9d:fa via interface br-lan
Fri Jan 15 20:58:24 2021 daemon.info pppd[20478]: Using interface ppp31
Fri Jan 15 20:58:24 2021 daemon.notice pppd[20478]: Connect: ppp31 <--> br-lan
Fri Jan 15 20:58:27 2021 daemon.info pppd[20478]: user 00951092205 logged in on tty br-lan intf ppp31
Fri Jan 15 20:58:27 2021 daemon.notice pppd[20478]: peer from calling number 64:6E:97:D0:9D:FA authorized
Fri Jan 15 20:58:27 2021 daemon.notice pppd[20478]: local  IP address 172.16.0.1
Fri Jan 15 20:58:27 2021 daemon.notice pppd[20478]: remote IP address 172.16.0.205
Fri Jan 15 20:59:03 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:00:23 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:01:43 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:01:51 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:03:11 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:03:35 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:04:11 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:04:35 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:06:01 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:06:14 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:06:39 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:06:47 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:09:24 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:10:39 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:11:31 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:12:12 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:14:28 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:14:41 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:15:17 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:15:36 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:17:57 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:18:27 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:18:41 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:19:11 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:19:22 2021 user.info mwan3track[11908]: Check (ping) failed for target "223.5.5.5" on interface wanb (eth5)
Fri Jan 15 21:19:35 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:21:34 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:22:04 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:23:02 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)
Fri Jan 15 21:23:15 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.5.5.5" on interface wan (eth4)
Fri Jan 15 21:23:29 2021 user.info mwan3track[14312]: Check (ping) failed for target "223.6.6.6" on interface wan (eth4)

feckert, 2021-01-18:
@openwrt-diy Why did you close the issue? If it works now it would not be bad if you give us a feedback.

openwrt-diy, 2021-01-18:
@feckert 
At present, I have more than a dozen routers with MWAN3 function. The main usage scenarios are PPPOE server, VPN client routing strategy and wireless router. The network broadband used includes China Telecom, China Unicom, China Mobile, Telenor, etc. No solution has been found yet!

openwrt-diy, 2021-01-18:
Mon Jan 18 22:43:04 2021 user.info mwan3track[19174]: Check (ping) failed for target "8.8.8.8" on interface wand (pppoe-wand)
Mon Jan 18 22:43:07 2021 authpriv.info dropbear[18366]: Exit (root): Exited normally
Mon Jan 18 22:43:08 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:43:13 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:43:20 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:43:25 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:43:32 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:43:39 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:43:46 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:43:53 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:44:01 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:44:06 2021 user.info mwan3track[13960]: Lost 5 ping(s) on interface wan (eth4)
Mon Jan 18 22:44:13 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:44:18 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:44:30 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:44:37 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:44:42 2021 user.info mwan3track[13960]: Lost 2 ping(s) on interface wan (eth4)
Mon Jan 18 22:45:10 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:45:15 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:45:22 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:45:27 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:45:34 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:45:41 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:45:46 2021 user.info mwan3track[13960]: Lost 2 ping(s) on interface wan (eth4)
Mon Jan 18 22:45:53 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:45:58 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:46:05 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:46:12 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:46:14 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:46:19 2021 user.info mwan3track[13960]: Lost 3 ping(s) on interface wan (eth4)
Mon Jan 18 22:46:31 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:46:39 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:46:41 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:46:46 2021 user.info mwan3track[13960]: Lost 3 ping(s) on interface wan (eth4)
Mon Jan 18 22:46:53 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:47:00 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:47:07 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:47:14 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:47:19 2021 user.info mwan3track[13960]: Lost 4 ping(s) on interface wan (eth4)
Mon Jan 18 22:47:46 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:47:53 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:47:59 2021 user.info mwan3track[13960]: Lost 2 ping(s) on interface wan (eth4)
Mon Jan 18 22:48:06 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:48:11 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:48:18 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:48:23 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:48:40 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:48:45 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:48:57 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:49:02 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
[... cut ...]
Mon Jan 18 22:53:22 2021 user.info mwan3track[13960]: Check (ping) failed for target "1.1.1.1" on interface wan (eth4)
Mon Jan 18 22:53:27 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:53:34 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)
Mon Jan 18 22:53:40 2021 user.info mwan3track[13960]: Lost 1 ping(s) on interface wan (eth4)
Mon Jan 18 22:53:52 2021 user.info mwan3track[13960]: Check (ping) failed for target "8.8.8.8" on interface wan (eth4)

feckert, 2021-01-18:
Can you switch off the mwan3 and start a ping on the console.
I have also some problems with ping, especially with wireless connections.
The way I see it, it is a bad connection.

openwrt-diy, 2021-01-19:
Modifying the default configuration of the MWAN3 interface (PING count) can solve the system log: MWAN3 Check (ping) failed for target

/etc/config/mwan3
config interface 'wan'
	option enabled '1'
	option initial_state 'online'
	option family 'ipv4'
	list track_ip '223.5.5.5'
	list track_ip '223.6.6.6'
	list track_ip '180.76.76.76'
	list track_ip '119.29.29.29'
	option track_method 'ping'
	option reliability '2'
	**option count '2'**
	option size '56'
	option max_ttl '60'
	option check_quality '0'
	option timeout '2'
	option interval '5'
	option failure_interval '5'
	option recovery_interval '5'
	option down '3'
	option up '3'

closed 2021-01-21 by openwrt-diy

feckert, 2021-01-22:
@openwrt-diy maybe you could add them to openwrt [mwan3 Wiki](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3)
@jamesmacwhite ping

jamesmacwhite, 2021-01-22:
Just catching up. Was this issue because one of the public DNS resolvers used for ping tracking was being blocked by your provider?

https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#reliable_public_ip_addresses_to_ping

We can add a note around potential issues with certain resolvers, although in the opening paragraph before the table of info we do mention you could also use your ISP provided DNS resolver as an endpoint on that specific connection too.

openwrt-diy, 2021-01-23:
@jamesmacwhite Just catching up. Was this issue because one of the public DNS resolvers used for ping tracking was being blocked by your provider?

Both ISP broadband providers and public DNS resolvers will cause duplicate system logs! The router administrator can change the relevant host or IP settings in time.
In some areas, using the DNS resolver of the ISP broadband provider will cause pollution, but the network is relatively stable.

openwrt-diy, 2021-01-23:
@aaronjg @jamesmacwhite @feckert

Can you consider adding a new tracking status on the MWAN Interfaces status page?
Interface: wanc
Status: Online
Track: Good, fair, bad
Uptime: 26h:32m:31s
Track: Good, fair, bad
Uptime: 26h:32m:31s