openwrt/packages #14488: Update USBUTILS usb.ids file

CanadianJeff, 2021-01-16:
The latest version on http://www.linux-usb.org/usb.ids is `Version: 2020.08.26` on my APU2 it shows `Version: 2019.02.25` over a year old

```
root@APU2:/usr/share# opkg files usbutils
Package usbutils (007-10) is installed on root and has the following files:
/usr/share/usb.ids
/usr/bin/lsusb
```

BKPepe, 2021-01-24:
Package usbutils for OpenWrt 19.07 is in OpenWrt main repository.
See: https://github.com/openwrt/openwrt/blob/openwrt-19.07/package/utils/usbutils/Makefile

Master has the latest version.
See: https://github.com/openwrt/packages/commit/4881e4251feedb41edf4ca0e9bb2347770e06cd4

closed 2021-01-24 by BKPepe

CanadianJeff, 2022-03-25:
this seems to be broken in later versions of openwrt???

```
root@APU2:/usr/share/hwdata# opkg files usbutils
Package usbutils (013-2) is installed on root and has the following files:
/usr/bin/lsusb
/usr/bin/usbreset
root@APU2:/usr/share/hwdata# opkg files usbids
Package usbids (0.354-1) is installed on root and has the following files:
/usr/share/hwdata/usb.ids
root@APU2:/usr/share/hwdata# lsusb
Bus 002 Device 002: ID 2357:0120 Realtek  802.11ac WLAN Adapter
Bus 002 Device 001: ID 1d6b:0002 Linux 5.4.179 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 0438:7900
Bus 001 Device 001: ID 1d6b:0002 Linux 5.4.179 ehci_hcd EHCI Host Controller
Bus 003 Device 001: ID 1d6b:0003 Linux 5.4.179 xhci-hcd xHCI Host Controller
```
```
root@APU2:/usr/share/hwdata# cat /etc/banner
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.2, r16495-bf0c965af0
 -----------------------------------------------------
```