openwrt/packages #14490: prometheus-node-exporter-lua cannot bind ip localhost only

shandshellin, 2021-01-16:
Tried to set an ip in cat /etc/default/prometheus-node-exporter but it always defaults to local host. Is there somewhere else to configure this?

PolynomialDivision, 2021-02-03:
You have to set this under `/etc/config/prometheus-node-exporter-lua`.

dalins, 2021-02-05:
I also have the same problem and I set it under `/etc/config/prometheus-node-exporter-lua`

> root@OpenWrt:\~# cat /etc/config/prometheus-node-exporter-lua 
> config prometheus-node-exporter-lua 'main'
> 	option listen_address '192.178.1.1'
> 	option listen_port '9100'
>
> root@OpenWrt:~# netstat -l | grep 9100
> tcp        0      0 localhost:9100          0.0.0.0:*               LISTEN

dhewg, 2021-02-05:
There's no `listen_address` anymore, it's `listen_interface`: https://github.com/openwrt/packages/blob/master/utils/prometheus-node-exporter-lua/files/etc/config/prometheus-node-exporter-lua

See also https://github.com/openwrt/packages/commit/48568ad9f9dfca6cecdcf1eabda629d8ccb77f42

dalins, 2021-02-05:
Thank you ... works with '*' as expected.

For others:
> config prometheus-node-exporter-lua 'main'
>         option listen_interface '*'
>         option listen_ipv6 '0'
>         option listen_port '9100'

closed 2022-04-25 by champtar