openwrt/packages #14516: <ffmpeg>: does not compile in v19.07.6 arch mvebu

glucini, 2021-01-20:
Please make sure that the issue subject starts with `<package-name>: `

# Issue template (remove lines from top till here)

Maintainer: @\Ted Hess, Ian Leonard (find it by checking history of the package Makefile)
Environment: 
arm_cortex-a9_vfpv3, v19.07.6

Description:
`ffmpeg fails to compile with error: package/feeds/packages/gcc failed to build, the issue seems related to the componente flac failing to compile`

Formating code blocks by wrapping them with pairs of ```
```

BKPepe, 2021-01-24:
I am not able to reproduce your issue and even OpenWrt buildbots are compiling this package as it should. Can you please provide  more logs instead of just one row from whcih we don't know why it failed in your case? Also, it could be related to gcc.

glucini, 2021-01-24:
Thank you for looking into that. Pls find attached the compile log. Let me know which additional information are required to reproduce the error

<ffmpeg-full>
Package libffmpeg-full is missing dependencies for the following libraries:
libfdk-aac.so.2
make[3]: *** [Makefile:810: /home/viper/openwrt/openwrt/bin/packages/arm_cortex-a9_vfpv3-d16/packages/libffmpeg-full_3.4.7-3_arm_cortex-a9_vfpv3-d16.ipk] Error 1
time: package/feeds/packages/ffmpeg/full/compile#0.36#0.11#0.46

[compile ffmpeg full.txt](https://github.com/openwrt/packages/files/5862015/compile.ffmpeg.full.txt)
[compile audio-dec.txt](https://github.com/openwrt/packages/files/5862016/compile.audio-dec.txt)

neheb, 2021-01-30:
Sounds like you're compiling with BUILD_PATENTED.

glucini, 2021-01-31:
Yes, that is correct. I am building with BUILD_PATENTED; actually I was always doing so since many releases. It actually stopped working since release v19.07.6. Before I was not getting any error.