openwrt/packages #14518: Luci-app-sqm won’t automatically start after boot

rulet, 2021-01-20:
sqm rules doesn't work after reboot, I have to manually restart it, openwrt 19.07.5

tohojo, 2021-01-27:
Upstream issue: https://github.com/tohojo/sqm-scripts/issues/131

Potentially fixed by https://github.com/tohojo/sqm-scripts/commit/37ead1f6c6666cba202d10e934ee03da532aa84d but would be good to confirm before pushing that to the packages repo...

tapper82, 2021-02-10:
I think this is fixt. Can It be closed?

tohojo, 2021-02-10:
I was still waiting for someone to confirm that the upstream fixed linked above actually solved this; if it does, I'll push it to the repo and close the bug that way. Have you tested it?

tapper82, 2021-02-10:
It's fixt in master. I run it on my router and it's is fine. OpenWrt SNAPSHOT r15727-c382fe857d / LuCI Master git-21.040.83252-3c166c2
sqm-scripts
1.5.0-2

tohojo, 2021-02-10:
But are you running with a config like that in https://github.com/tohojo/sqm-scripts/issues/131 ?

tapper82, 2021-02-11:
No I am not. Sorry I did not see the strange config.

hilman-k8n, 2021-12-21:
does not work for me too
```
luci-app-sqm git-21.188.55209-f161b40
sqm-scripts 1.5.1-1
openwrt-21.02 branch (git-21.295.67054-13df80d) / OpenWrt 21.02.1 r16325-88151b8303
```

I need to manually add `/etc/init.d/sqm start` in `/etc/rc.local` to make it works

tohojo, 2021-12-21:
@hilman-k8n could you please post your /etc/config/sqm?

hilman-k8n, 2021-12-25:
Hi @tohojo, this is my `/etc/config/sqm`
```
config queue
	option enabled '1'
	option interface 'wlan0-2'
	option download '64'
	option upload '64'
	option debug_logging '0'
	option verbosity '0'
	option qdisc 'cake'
	option script 'piece_of_cake.qos'
	option linklayer 'none'

config queue
	option enabled '1'
	option interface 'wlan0-1'
	option download '512'
	option upload '512'
	option debug_logging '0'
	option qdisc 'cake'
	option script 'piece_of_cake.qos'
	option linklayer 'none'
	option verbosity '0'
```

tohojo, 2022-01-05:
@hilman-k8n okay first of all, which WiFi driver are you running, and why do you need to run SQM-scripts on the WiFi interface? Also, those are some very low bandwidth settings, what's your use case, exactly? :)

hilman-k8n, 2022-04-11:
Hi @tohojo, it's only for guest SSID

I've solved this issue. The root cause was caused by SQM trying to find interface that did not exist yet - so it'll fail. 

You need to update `service_triggers()` when wifi interface is up

tohojo, 2022-04-11:
> You need to update service_triggers() when wifi interface is up

Care to elaborate? Or maybe open a PR?