openwrt/packages #14564: joe: Missing symlinks to use emulation modes

dgasaway, 2021-01-25:
Maintainer: @aTanW (find it by checking history of the package Makefile)
Environment: OpenWrt 19.07.3 r11063-85e04e9f46, Linksys WRT 32X

Description:

The joe package includes at least the emacs and pico emulations built-in, I know because I've tested this myself.  However, the symlinks required to use them are not currently installed.  I created symlinks in /usr/bin named jmacs and jpico pointing to joe, and tested them both.  They work, including pulling in my jmacs settings from ~/.jmacsrc, which the jmacs from the jupp-data package could not do.

aTanW, 2021-01-26:
Symlinks are not missing.
Due to installed size package does not supply configuration for these modes, so there are no symlinks.

dgasaway, 2021-01-26:
Could you explain, please?  As I said, after adding the symlinks I mentioned, the emacs and joe emulations worked fine as best I could tell.  What configuration is missing?

aTanW, 2021-01-26:
The only joerc file installed as editor config, thus, only one, native mode fully configured.

dgasaway, 2021-01-26:
A user can provide their own file, as I did, ~/.jmacsrc.  The global rc files are not needed, but the symlinks are needed to enable the emulation modes.

aTanW, 2021-01-26:
You miss "fully configured". Base setup files for different modes are big enough. They exist to reconfigure editor to emulate desired mode. Without them you'll get standard editor which reads, for example, ~/.jmacsrc, but it will not emulate emacs. It is pointless state to have executable with name, but without named function.
If you need that emulation, you can drop joe's standard setup file to /etc/joe/ and make executable link yourself.
Or, someday, I'll try to add separate packets for all extended joe modes. Or for each mode separately.

dgasaway, 2021-01-26:
Correct me if I'm wrong.  Joe only reads a single initialization file.  It does not read /etc/joe/jmacsrc, then also read ~/.jmacsrc to apply user customizations.  The user would copy /etc/joe/jmacsrc to ~/.jmacsrc then make the desired changes (or copy from another system, as I did).  Thus, ~/.jmacsrc is a complete initialization file necessary to enable the emacs emulation.  It is "fully configured."  That is the way I've always seen it used, and that is the state of my jmacsrc files.  That's the point I was trying to make.

In any case, I thought it would be a good idea to meet the user half-way with this, but I'm no expert.

aTanW, 2021-01-26:
You're wrong. First of all joe reads it's default init file in /etc/joe/. And, in case it exist, user setup for desired mode from $HOME/.
You try to use software in your way. But here is "industrial" variant. Means, that it must have sence, it must be usable by many other setups, it must be optimal.

dgasaway, 2021-01-26:
I'm not trying to use the software in a way different than anyone else.  This was my understanding based on joe's documentation.  "^T options, the help screens and the key-sequence to editor command bindings are all defined in JOE´s initialization file. If you make a copy of this file (which normally resides in /etc/joe/joerc) to $HOME/.joerc, you can customize these setting to your liking."  If you have pointers on how else to use it, I'd appreciate the help.

aTanW, 2021-02-28:
New pull req:
joe: new package joe-extras #14943