openwrt/packages #14573: unbound: does not compile on macOS 11.1-x86_64 host

k-ronny, 2021-01-26:
Maintainer: @EricLuehrsen
Environment: all?
Description: cross compiling of unbound on a macOS 11.1-x86_64 host is failing in the configure phase:
```
...
checking whether CONNECT_RESUME_ON_READ_WRITE is declared... no
configure: error: TCP Fast Open is not available for client mode: please rerun without --enable-tfo-client
...
```
I think `configure` is using the wrong target because of the use of `uname` in configure.ac:
```
AC_ARG_ENABLE(tfo-client, AS_HELP_STRING([--enable-tfo-client],[Enable TCP Fast Open for client mode]))
case "$enable_tfo_client" in
	yes)
		case `uname` in
			Linux) AC_CHECK_DECL([MSG_FASTOPEN], [AC_MSG_WARN([Check the platform specific TFO kernel parameters are correctly configured to support client mode TFO])],
			                     [AC_MSG_ERROR([TCP Fast Open is not available for client mode: please rerun without --enable-tfo-client])], 
			                     [AC_INCLUDES_DEFAULT 
#include <netinet/tcp.h>
])
					AC_DEFINE_UNQUOTED([USE_MSG_FASTOPEN], [1], [Define this to enable client TCP Fast Open.])
			  ;;
			Darwin) AC_CHECK_DECL([CONNECT_RESUME_ON_READ_WRITE], [AC_MSG_WARN([Check the platform specific TFO kernel parameters are correctly configured to support client mode TFO])], 
			                      [AC_MSG_ERROR([TCP Fast Open is not available for client mode: please rerun without --enable-tfo-client])], 
			                      [AC_INCLUDES_DEFAULT
#include <sys/socket.h>
])
					AC_DEFINE_UNQUOTED([USE_OSX_MSG_FASTOPEN], [1], [Define this to enable client TCP Fast Open.])
			  ;;
		esac
	;;
	no|*)
		;;
esac
```

with this patch it compiles on macOS and Debian Buster:
```
--- a/configure.ac	2020-12-03 09:31:18.000000000 +0100
+++ b/configure.ac	2021-01-26 01:16:58.000000000 +0100
@@ -120,6 +120,9 @@
 AC_SUBST(LIBUNBOUND_REVISION)
 AC_SUBST(LIBUNBOUND_AGE)
 
+# AC_CANONICAL_HOST is needed to access the 'host_os' variable
+AC_CANONICAL_HOST
+
 
 cmdln="`echo $@ | sed -e 's/\\\\/\\\\\\\\/g' | sed -e 's/"/\\\\"/'g`"
 AC_DEFINE_UNQUOTED(CONFCMDLINE, ["$cmdln"], [Command line arguments used with configure])
@@ -1226,15 +1229,15 @@
 AC_ARG_ENABLE(tfo-client, AC_HELP_STRING([--enable-tfo-client], [Enable TCP Fast Open for client mode]))
 case "$enable_tfo_client" in
 	yes)
-		case `uname` in
-			Linux) AC_CHECK_DECL([MSG_FASTOPEN], [AC_MSG_WARN([Check the platform specific TFO kernel parameters are correctly configured to support client mode TFO])],
+		case "${host_os}" in
+			linux*) AC_CHECK_DECL([MSG_FASTOPEN], [AC_MSG_WARN([Check the platform specific TFO kernel parameters are correctly configured to support client mode TFO])],
 			                     [AC_MSG_ERROR([TCP Fast Open is not available for client mode: please rerun without --enable-tfo-client])], 
 			                     [AC_INCLUDES_DEFAULT 
 #include <netinet/tcp.h>
 ])
 					AC_DEFINE_UNQUOTED([USE_MSG_FASTOPEN], [1], [Define this to enable client TCP Fast Open.])
 			  ;;
-			Darwin) AC_CHECK_DECL([CONNECT_RESUME_ON_READ_WRITE], [AC_MSG_WARN([Check the platform specific TFO kernel parameters are correctly configured to support client mode TFO])], 
+			darwin*) AC_CHECK_DECL([CONNECT_RESUME_ON_READ_WRITE], [AC_MSG_WARN([Check the platform specific TFO kernel parameters are correctly configured to support client mode TFO])], 
 			                      [AC_MSG_ERROR([TCP Fast Open is not available for client mode: please rerun without --enable-tfo-client])], 
 			                      [AC_INCLUDES_DEFAULT
 #include <sys/socket.h>
```

sdl5086, 2021-01-28:
Just ran into this trying to build OpenWrt on 10.15.7 after resolving a few other "curiosities" when using a Mac OS build environment. I'm pretty new to OpenWrt, but it looks like I need to look into implementing this patch..?

k-ronny, 2021-01-29:
@sdl5086
just put the patch as e.g. `999-macos-build.patch` into `feeds/packages/net/unbound/patches` and start a new build

sdl5086, 2021-01-29:
Awesome thank you do much!!!

I think I had just barely figured it out on my own when I saw you replied, so I was quite happy to see that I was on the right track (even for what's likely a simple thing lol). 

While that worked and the build gets significantly further, unfortunately the worm-can lid has peeled back to expose `Undefined symbols for architecture x86_64` in `rpc_util` and this appears to be a known Mac OS thing, something about Apple changing something in `ld` or `clang` or something. I'm soooo far out of my n00b comfort zone as a lowly C# Winforms "enterprise" (read: ugly kludge) dev... and I don't want to clutter the comments here. I posted on the [openwrt dev forum](https://forum.openwrt.org/t/nanopi-r1-mac-os-10-15-5-build-unknown-type-name-u64-did-you-mean-s64/86921/5) but so far the only reply has been another Mac user who gave up! 😭

k-ronny, 2021-01-29:
oh, I do not know anything about `rpc_util` but I can tell you something about my build environment:

with `brew` I have installed:
```
brew install coreutils findutils gawk gnu-getopt gnu-tar grep make wget
```
my `PATH` is set to:
```
export PATH=/usr/local/bin:/usr/bin:/bin
```

with this environment I have successfully built my targets

sdl5086, 2021-01-29:
That looks like what I have, though if I remember correctly, grep gets installed as ggrep so I symlinked it to grep in one of the toolchain directories in openwrt and added that to the "env.sh" script I run in the console I'm using the build openwrt. (This will hopefully keep Xcode from throwing a temper tantrum when I try to build Mac OS and iOS apps in the future?)

There are a few openwrt packages that still grumble about various things but so far I've just been able to disable them and work around them. Otherwise, I've got openwrt building in Mac OS now thanks to you and some other Mac OS stalwarts who just didn't want to put up with some of the "features" of Ubuntu, like ghost drives not ejecting and requiring a reboot, SMB not working after waking from sleep, and so on. lol

EricLuehrsen, 2021-01-30:
This makes sense so I kicked it upstream https://github.com/NLnetLabs/unbound/issues/413

k-ronny, 2021-01-30:
@EricLuehrsen
Thank you for taking the time.

k-ronny, 2021-06-09:
fixed with https://github.com/NLnetLabs/unbound/commit/ad939e54d0cc00c32fe88f4c6c5859536874ff43
Thank you @EricLuehrsen @gthess

closed 2021-06-09 by k-ronny