openwrt/packages #14577: ttyd: --ssl not working?

doverride, 2021-01-26:
@tsl0922 

NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r15520-54bfebdca0"
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r15520-54bfebdca0"

root@OpenWrt:~# ttyd --version
ttyd version 1.6.3

Is --ssl supposed to work? I could not get it to work

root@OpenWrt:~# cat /etc/config/ttyd

config ttyd
	option interface '@wan'
#	option ssl 'true'
	option ssl_cert '/etc/acme/openwrt.defensec.nl/fullchain.cer'
#	option ssl_cert '/etc/acme/openwrt.defensec.nl/openwrt.defensec.nl.cer'
	option ssl_key '/etc/acme/openwrt.defensec.nl/openwrt.defensec.nl.key'
	option ssl_ca '/etc/acme/openwrt.defensec.nl/ca.cer'
	option ipv6 'true'
	option command 'runcon -t sandbox.net.subj -- /usr/libexec/login.sh'
	option credential 'test:abc123'

tsl0922, 2021-01-27:
Check this wiki: https://github.com/tsl0922/ttyd/wiki/SSL-Usage

if you do not use client certificate verification, remove the ssl-ca option.

doverride, 2021-01-27:
> Check this wiki: https://github.com/tsl0922/ttyd/wiki/SSL-Usage

I have

>
> if you do not use client certificate verification, remove the ssl-ca option.

Not sure if it is in any way related but it seems that if i ommit that
then it will failover to /etc/ssl/cert.pem which points to /etc/ssl/certs/ca-certificates.crt

Regardless, the verbose logging of debug 7 does not print any
information. I tried to make it more verbose but it does not tell me why
it fails. When i try to connect i just get a blank page and no ssl

tsl0922, 2021-01-27:
> the verbose logging of debug 7 does not print any information.

the debug level is passed to libwebsockets, you may need to build libwebsockets with `-DCMAKE_BUILD_TYPE=DEBUG` to make all the debug level work.

doverride, 2021-03-16:
this issue was finally addressed via: https://github.com/tsl0922/ttyd/commit/0b146b750c5c1e88a1bf52fe9e61e1256017c82e

can back port?

closed 2025-11-14 by GeorgeSapkin