openwrt/packages #14580: error with latest go and ccache

neheb, 2021-01-27:
Maintainer: @jefferyto 
Environment: CentOS 7

Example:

```
/home/mangix/devstuff/openwrt/build_dir/target-i386_pentium4_musl/go-1.15.7/pkg/tool/linux_amd64/link: running gcc failed: exit status 134
go-gcc-helper: running ccache_cc -m32 -s -o $WORK/b377/exe/a.out -rdynamic -Wl,--compress-debug-sections=zlib-gnu /home/mangix/devstuff/openwrt/tmp/go-link-534154293/go.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000000.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000001.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000002.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000003.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000004.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000005.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000006.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000007.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000008.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000009.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000010.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000011.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000012.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000013.o /home/mangix/devstuff/openwrt/tmp/go-link-534154293/000014.o -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium4_gcc-10.2.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium4_gcc-10.2.0_musl/lib -znow -zrelro -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium4_gcc-10.2.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium4_gcc-10.2.0_musl/lib -znow -zrelro -lpthread -no-pie -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium4_gcc-10.2.0_musl/usr/lib -L/home/mangix/devstuff/openwrt/staging_dir/toolchain-i386_pentium4_gcc-10.2.0_musl/lib -Wl,-z,now -Wl,-z,relro
ccache: Util.cpp:718: std::string Util::get_relative_path(nonstd::sv_lite::string_view, nonstd::sv_lite::string_view): failed assertion: Util::is_absolute_path(dir)
/home/mangix/devstuff/openwrt/staging_dir/hostpkg/lib/go-cross/openwrt/gcc: line 41: 20771 Aborted                 (core dumped) $cmd "$@
```

The crash comes from a failing assertion in the ccache code: https://github.com/ccache/ccache/blob/master/src/Util.cpp#L731

neheb, 2021-01-27:
ping @G-M0N3Y-2503 

runc does not respect GO_GCC_HELPER_CC.

jefferyto, 2021-02-22:
Does this still occur with Go 1.16?

Re: GO_GCC_HELPER_CC - IIRC I added this because `$(HOSTCC)` and `$(HOSTCXX)` expand to `ccache gcc` and `ccache g++`, respectively, while Go expects `CC` and `CXX` to be the name/path of the compiler only (no command line options). Unless runc compiles something for host, using GO_GCC_HELPER_CC isn't necessary.

neheb, 2021-10-03:
@jefferyto Just hit this on macOS.

closed 2022-09-16 by neheb