openwrt/packages #14599: p910nd default options fail to work

howels, 2021-01-28:
Maintainer: @danielfdickinson 
Environment: (MIPS, Archer C7, 19.07.4)

Description:

Default p910nd adds config file with `option runas_root '1'` but despite the `p910nd` user being added to /etc/passwd the service crashes when running with this option set. The service also tries to set the `option port: '0'` by default which prevented my printer from working, I needed to remove the port statement for my HP DeskJet J510 series 2050 to print.

danielfdickinson, 2021-02-06:
Sorry I shouldn't be listed as maintainer anymore for life reason I am no active with OpenWrt anymore. Are you using openwrt/packages **master**?

howels, 2021-02-07:
Was using 19.07.4 as mentioned in the report. Last commit in packages/net/p910nd looks to be 2 years ago and 19.07.4 was published September 2020, so I believe that the package is up to date.
Apart from adding p910nd, kmod-usb-print and luci-ssl and replacing wpad-basic with wpad I have a default install.