openwrt/packages #14606: kmod-swconfig_5.4.92-1_mipsel_24kc.ipk  Wrong dependencies

baiclamp, 2021-01-30:
Environment: (put here arch, model, OpenWrt version)
mips mt7621/A7000R OpenWrt 5.4.92 

Description:

When opk install this package with wrong depending messages.

```
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/kmods/5.4.92-1-6b0e6ccfc1a63ac8682d721effce8201/kmod-swconfig_5.4.92-1_mipsel_24kc.ipk
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-swconfig:
 * 	kernel (= 5.4.93-1-6b0e6ccfc1a63ac8682d721effce8201)
 * opkg_install_cmd: Cannot install package kmod-swconfig.
root@OpenWrt:/etc/config# uname -a
Linux OpenWrt 5.4.92 #0 SMP Thu Jan 28 11:01:14 2021 mips GNU/Linux
```

BKPepe, 2021-01-30:
You should update your kernel, but kernel and related kernel packages comes from [main repository of OpenWrt](https://github.com/openwrt/openwrt).

closed 2021-01-30 by BKPepe

baiclamp, 2021-01-31:
please tell me how . i am not a developer
you make a car that using H2 for power that now you don't have H2 power and you told the people to change the engine to use gas? you don't think that is a bug ?

BKPepe, 2021-01-31:
Dear @baiclamp,

You are using snapshots, it means that's the development branch [1]. Bugs are expected there, however, snapshots expect that have some experience with GNU/Linux. You don't need to be a developer for such things.

If you would check it carefully, you will notice that the packages repository does not have packages like kernel and kernel packages (kmod). These are in OpenWrt main repository [2], [3] as I said.  When you are not sure, there is an OpenWrt forum [4] and documentation for reporting bugs [5].

You might want to consider using OpenWrt stable releases.

TL;DR: This is not a bug, you should update your router to the latest version of snapshots.
You have installed kernel version 5.4.92 and once there is an updated version of the kernel, you should update it. So, you will be able to download the packages that you want. 

[1] https://openwrt.org/releases/snapshot
[2] https://github.com/openwrt/openwrt/blob/master/include/kernel-version.mk
[3] https://github.com/openwrt/openwrt/tree/master/package/kernel/linux/modules
[4] https://forum.openwrt.org/
[5] https://openwrt.org/bugs