openwrt/packages #14607: ddns-scripts_cloudflare.com-v4: Improvement suggestion, change authentication method

jcrsilva, 2021-01-30:
# Issue template (remove lines from top till here)

Maintainer: (not clear from history)
Environment: Xiaomi Mi R3P

Description:

Authentication to cloudflare APIs in this script is done via email + password (API key) combo:

Both are added via headers here:
```
__PRGBASE="$__PRGBASE --header 'X-Auth-Email: $username' "
__PRGBASE="$__PRGBASE --header 'X-Auth-Key: $password' "
```

This requires the user to use their global API key in their router. This is very insecure, as a compromise of said key will give an attacker full control over any domains owned by that account.

Cloudflare now [supports authentication via bearer token](https://api.cloudflare.com/#getting-started-endpoints) using the Authorization header:

```
"Authorization: Bearer 1234567893feefc5f0q5000bfo0c38d90bbeb"
```

Main advantages to this approach:
1. Scope of api access is limited, the router can be given a key to only update an existing DNS record;
2. Multiple keys can be created, and revoked, minimizing impact if one needs to be revoked
3. User email is not required for authentication

I'm happy to submit a PR adding support for this.

feckert, 2021-02-01:
The way I see it, it's already supported. https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/usr/lib/ddns/update_cloudflare_com_v4.sh#L130 in the master branch

jcrsilva, 2021-02-01:
You're right, I was looking at the 19.07 branch. Sorry for the confusion.

closed 2021-02-01 by jcrsilva