openwrt/packages #14625: prometheus_node_exporter_lua not sending wifi stats

matthewhelmke, 2021-02-01:
Maintainer: @champtar 
Environment: mvebu, Linksys WRT-1900ac v1, OpenWrt 19.07.6 r11278-8055e38794

Description:
Metrics for the prometheus_node_exporter_lua_wifi and _wifi_stations packages do not exist in `IP:9100/metrics` while all other expected metrics continue to be updated as expected.

The change occurred starting with the upgrade to OpenWrt 19.07.6.

Here is the current content of `IP:9100/metrics` with the following packages from the set installed.
Packages:
- prometheus-node-exporter-lua
- prometheus-node-exporter-lua-nat_traffic
- prometheus-node-exporter-lua-netstat
- prometheus-node-exporter-lua-openwrt
- prometheus-node-exporter-lua-wifi
- prometheus-node-exporter-lua-wifi_stations

```
# TYPE node_scrape_collector_duration_seconds gauge
# TYPE node_scrape_collector_success gauge
# TYPE node_nf_conntrack_entries gauge
node_nf_conntrack_entries 1259
# TYPE node_nf_conntrack_entries_limit gauge
node_nf_conntrack_entries_limit 16384
node_scrape_collector_duration_seconds{collector="conntrack"} 0.00044703483581543
node_scrape_collector_success{collector="conntrack"} 1
# TYPE node_boot_time_seconds gauge
node_boot_time_seconds 1611234660
# TYPE node_context_switches_total counter
node_context_switches_total 554732730
# TYPE node_cpu_seconds_total counter
node_cpu_seconds_total{cpu="cpu0",mode="user"} 2058
node_cpu_seconds_total{cpu="cpu0",mode="nice"} 0
node_cpu_seconds_total{cpu="cpu0",mode="system"} 4699.78
node_cpu_seconds_total{cpu="cpu0",mode="idle"} 943681.61
node_cpu_seconds_total{cpu="cpu0",mode="iowait"} 0.25
node_cpu_seconds_total{cpu="cpu0",mode="irq"} 0
node_cpu_seconds_total{cpu="cpu0",mode="softirq"} 10015.62
node_cpu_seconds_total{cpu="cpu0",mode="steal"} 0
node_cpu_seconds_total{cpu="cpu0",mode="guest"} 0
node_cpu_seconds_total{cpu="cpu0",mode="guest_nice"} 0
node_cpu_seconds_total{cpu="cpu1",mode="user"} 1291.66
node_cpu_seconds_total{cpu="cpu1",mode="nice"} 0
node_cpu_seconds_total{cpu="cpu1",mode="system"} 2541.25
node_cpu_seconds_total{cpu="cpu1",mode="idle"} 950643.49
node_cpu_seconds_total{cpu="cpu1",mode="iowait"} 0.08
node_cpu_seconds_total{cpu="cpu1",mode="irq"} 0
node_cpu_seconds_total{cpu="cpu1",mode="softirq"} 5978.87
node_cpu_seconds_total{cpu="cpu1",mode="steal"} 0
node_cpu_seconds_total{cpu="cpu1",mode="guest"} 0
node_cpu_seconds_total{cpu="cpu1",mode="guest_nice"} 0
# TYPE node_intr_total counter
node_intr_total 690234503
# TYPE node_forks_total counter
node_forks_total 36748
# TYPE node_procs_running_total gauge
node_procs_running_total 1
# TYPE node_procs_blocked_total gauge
node_procs_blocked_total 0
node_scrape_collector_duration_seconds{collector="cpu"} 0.0041790008544922
node_scrape_collector_success{collector="cpu"} 1
# TYPE node_filefd_allocated gauge
node_filefd_allocated 320
# TYPE node_filefd_maximum gauge
node_filefd_maximum 25078
node_scrape_collector_duration_seconds{collector="filefd"} 0.00034999847412109
node_scrape_collector_success{collector="filefd"} 1
# TYPE node_load1 gauge
node_load1 0.06
# TYPE node_load5 gauge
[... 232 lines cut ...]
node_network_transmit_carrier_total{device="eth1"} 0
# TYPE node_network_transmit_compressed_total counter
node_network_transmit_compressed_total{device="br-lan"} 0
node_network_transmit_compressed_total{device="ifb1"} 0
node_network_transmit_compressed_total{device="ifb0"} 0
node_network_transmit_compressed_total{device="lo"} 0
node_network_transmit_compressed_total{device="eth0"} 0
node_network_transmit_compressed_total{device="wlan0"} 0
node_network_transmit_compressed_total{device="ifb4eth0"} 0
node_network_transmit_compressed_total{device="wlan1"} 0
node_network_transmit_compressed_total{device="eth1"} 0
node_scrape_collector_duration_seconds{collector="netdev"} 0.012046098709106
node_scrape_collector_success{collector="netdev"} 1
# TYPE node_time_seconds counter
node_time_seconds 1612195115
node_scrape_collector_duration_seconds{collector="time"} 7.1048736572266e-05
node_scrape_collector_success{collector="time"} 1
# TYPE node_uname_info gauge
node_uname_info{version="#0 SMP Tue Jan 19 13:10:02 2021",machine="armv7l",release="4.14.215",domainname="(none)",nodename="OpenWrt",sysname="Linux"} 1
node_scrape_collector_duration_seconds{collector="uname"} 0.00022721290588379
node_scrape_collector_success{collector="uname"} 1

```

There should be a set of metrics that begin with `wifi_`.

PolynomialDivision, 2021-02-03:
Ah I saw that in fixed this in my own plugin. xD

PolynomialDivision, 2021-02-03:
@matthewhelmke Could you confirm that https://github.com/openwrt/packages/pull/14641 is fixing your bug?

matthewhelmke, 2021-02-08:
@PolynomialDivision no joy on my end. Same output as before.

champtar, 2021-02-08:
@matthewhelmke do you have hostapd/wpad full ?

matthewhelmke, 2021-02-08:
@champtar I have whichever versions of hostapd and wpad are current in opkg.

champtar, 2021-02-08:
`okpg list | grep wpad`

matthewhelmke, 2021-02-08:
```
root@OpenWrt:~# opkg list | grep wpad
wpad - 2019-08-08-ca8c2bd2-5 - This package contains a full featured IEEE 802.1x/WPA/EAP/RADIUS Authenticator and Supplicant
wpad-basic - 2019-08-08-ca8c2bd2-5 - This package contains a basic IEEE 802.1x/WPA Authenticator and Supplicant with WPA-PSK, 802.11r and 802.11w support.
wpad-basic-wolfssl - 2019-08-08-ca8c2bd2-5 - This package contains a basic IEEE 802.1x/WPA Authenticator and Supplicant with WPA-PSK, SAE (WPA3-Personal), 802.11r and 802.11w support.
wpad-mesh-openssl - 2019-08-08-ca8c2bd2-5 - This package contains a minimal IEEE 802.1x/WPA Authenticator and Supplicant (with 802.11s mesh and SAE support).
wpad-mesh-wolfssl - 2019-08-08-ca8c2bd2-5 - This package contains a minimal IEEE 802.1x/WPA Authenticator and Supplicant (with 802.11s mesh and SAE support).
wpad-mini - 2019-08-08-ca8c2bd2-5 - This package contains a minimal IEEE 802.1x/WPA Authenticator and Supplicant (WPA-PSK only).
wpad-openssl - 2019-08-08-ca8c2bd2-5 - This package contains a full featured IEEE 802.1x/WPA/EAP/RADIUS Authenticator and Supplicant
wpad-wolfssl - 2019-08-08-ca8c2bd2-5 - This package contains a full featured IEEE 802.1x/WPA/EAP/RADIUS Authenticator and Supplicant

```

champtar, 2021-02-08:
sorry, `opkg list-installed | grep wpad`

matthewhelmke, 2021-02-08:
That looks like it may be the issue.
```
root@OpenWrt:~# opkg list-installed | grep wpad
wpad-basic - 2019-08-08-ca8c2bd2-5
```
I have uninstalled `wpad-basic` and installed `wpad`.  Still no wifi metrics in /metrics.

champtar, 2021-02-08:
@blocktrron as you introduced the wifi stats collector, is it still working for you ?

champtar, 2021-02-08:
@matthewhelmke you rebooted right ?

matthewhelmke, 2021-02-08:
Nope. Slipped my mind (doing this in between work meetings). Just rebooted and the metrics are there.
Thank you!

PolynomialDivision, 2021-02-08:
So it is a dependency issue?

matthewhelmke, 2021-02-08:
@PolynomialDivision it appears so to me. Replacing `wpad-basic` with `wpad` (and remembering to reboot) is what took care of the issue for me.

champtar, 2021-02-09:
@matthewhelmke can you show me part of
```
ubus call network.wireless status
```
I'm interested where is `ifname` present with a full build

matthewhelmke, 2021-02-09:
Here you go, with a couple obvious changes for privacy.

```
root@OpenWrt:~# ubus call network.wireless status
{
	"radio0": {
		"up": false,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": true,
		"config": {
			"path": "soc/soc:pcie-controller/pci0000:00/0000:00:02.0/0000:02:00.0",
			"country": "US",
			"txpower": 30,
			"hwmode": "11a",
			"channel": "auto",
			"legacy_rates": true
		},
		"interfaces": [
			
		]
	},
	"radio1": {
		"up": false,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": true,
		"config": {
			"path": "soc/soc:pcie-controller/pci0000:00/0000:00:03.0/0000:03:00.0",
			"country": "US",
			"legacy_rates": false
		},
		"interfaces": [
			
		]
	},
	"radio2": {
		"up": true,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": false,
		"config": {
			"hwmode": "11g",
			"path": "soc/soc:pcie@82000000/pci0000:00/0000:00:02.0/0000:02:00.0",
			"htmode": "HT20",
			"country": "00",
			"channel": "9"
		},
		"interfaces": [
			{
				"section": "default_radio2",
				"ifname": "wlan0",
				"config": {
					"mode": "ap",
					"ssid": "matthewsmacssid",
					"encryption": "psk2",
					"key": "matthewskey",
					"macaddr": "matthewsmac",
					"network": [
						"lan"
					],
					"mode": "ap"
				}
			}
		]
	},
	"radio3": {
		"up": true,
		"pending": false,
		"autostart": true,
		"disabled": false,
		"retry_setup_failed": false,
		"config": {
			"hwmode": "11a",
			"path": "soc/soc:pcie@82000000/pci0000:00/0000:00:03.0/0000:03:00.0",
			"htmode": "VHT80",
			"channel": "36"
		},
		"interfaces": [
			{
				"section": "wifinet1",
				"ifname": "wlan1",
				"config": {
					"ssid": "matthewsssid",
					"encryption": "psk2",
					"mode": "ap",
					"key": "matthewskey",
					"macaddr": "matthewsmac",
					"mode": "ap",
					"network": [
						"lan"
					]
				}
			}
		]
	}
}
```

champtar, 2021-02-09:
Thanks
@PolynomialDivision your fix is still needed it seems

PolynomialDivision, 2021-02-09:
> Thanks
> @PolynomialDivision your fix is still needed it seems

Seems like this.

matthewhelmke, 2021-02-09:
FYI, thank you. I was able to get this out today. I held it back from publishing when I noticed my last screenshot (from Dec 2020) would no longer match was readers could achieve. Today, it will.
https://grafana.com/blog/2021/02/09/how-i-monitor-my-openwrt-router-with-grafana-cloud-and-prometheus/

champtar, 2022-04-25:
I believe everything was fixed

closed 2022-04-25 by champtar